Bumblebee-Project/bbswitch #103: bbwitch taints kernel all the time on ASUS X550JK

gnufreex, 2015-01-04:
dmesg | grep -E 'bbswitch'
[   71.644166] bbswitch: module verification failed: signature and/or  required key missing - tainting kernel
[   71.644335] bbswitch: version 0.8
[   71.644338] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   71.644342] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[   71.644438] bbswitch: detected an Optimus _DSM function
[   71.644486] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[   73.252381] bbswitch: disabling discrete graphics

BBswitch works but taints kenel. Laptop is ASUS X550JK. Secure Boot control is disabled in BIOS. I  previously used bumblebee without bbswitch,  kernel would not get tainted before I start game with optirun command. And after that it would stay tainted until i restart computer. But I suspected that NV GPU was working all the time. So I installed bbswitch expecting that all will remain the same except NV GPU will shutdown after 3d game and bbswitch would untaint kernel withour restart required. Guess not.

bbswitch taints kernel when computer is started, and it stays tainted all the time. It turns off GPU and nvidia kernel module gets unloaded when it is not used, but the kernel stays tainted. ALWAYS.

Lekensteyn, 2015-01-04:
bbswitch is an out-of-tree module. The kernel is signed with some key to which you cannot gain access. If this bothers you, rebuild the kernel with your own signing key.

Are you using the nouveau driver?

closed 2015-01-04 by Lekensteyn

gnufreex, 2015-01-04:
ohh... I see. 

No, I am not using noveau driver for this laptop, noveau doesnt  support GTX850m. I am using NVIDIA one with Bumblebee so I end up using Intel opensource driver most of the time. That is why it bothers me, I used tainted flag to know when NVIDIA driver is loaded. Also, posting bug reports might not work with tainted. I use Fedora, rebuilding every kernel release would be tiresome. What work need to be done to put bbswitch into mainline kernel? I might be interested to help.

Lekensteyn, 2015-01-05:
When you load nvidia, it will cause the proprietary taint to be set. bbswitch is GPL (G) and will also set the out-of-tree (O) taint.

nouveau can also provide the functionality of bbswitch, you then need to enable runtime PM for the nvidia PCI device. This will kick in later than bbswitch though (after 5 seconds idle), but has otherwise the same results.

gnufreex, 2015-01-05:
Is there a link with more specifics how to do that with noveau. And, does it matter that noveau doesn't support GTX850M yet?

Lekensteyn, 2015-01-06:
It does matter, if nouveau does not recognize the GTX850M, then it cannot bind to the device. You can enable runtime PM via the powertop program or create a udev rule: https://serverfault.com/questions/279527/powertop-reports-bad-runtime-pm-for-pci-device