Bumblebee-Project/bbswitch #104: ERROR: could not insert 'bbswitch': No such device

runya-v, 2015-01-17:
# sudo uname -a
-> Linux brain 3.2.0-4-amd64 #1 SMP Debian 3.2.65-1+deb7u1 x86_64 GNU/Linux
 
# lspci | grep -i vga
-> 00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
-> 01:00.0 VGA compatible controller: NVIDIA Corporation GF108 [GeForce GT 630M] (rev a1)

laptop is: Toshiba Satellite p875-bns

I tried:
# git clone git://github.com/Bumblebee-Project/bbswitch.git -b hack-lenovo
-> ...
# cd bbswitch
# sudo mkdir /usr/src/acpi-handle-hack-0.0.2
# sudo cp Makefile acpi-handle-hack.c /usr/src/acpi-handle-hack-0.0.2
# sudo cp dkms/acpi-handle-hack.conf /usr/src/acpi-handle-hack-0.0.2/dkms.conf
# sudo dkms install -m acpi-handle-hack -v 0.0.2
-> .....
-> DKMS: install completed.
# echo acpi-handle-hack | sudo tee -a /etc/modules
-> acpi-handle-hack
# sudo update-initramfs -u
-> update-initramfs: Generating /boot/initrd.img-3.2.0-4-amd64
# sudo modprobe -v bbswitch
-> ERROR: could not insert 'bbswitch': No such device

How can I fix this problem?

Lekensteyn, 2015-02-02:
Devices need to get whitelisted manually. Do you have an acpidump of your system somewhere? What does the kernel log (`dmesg | tail`) contain after trying to modprobe?

runya-v, 2015-02-03:
dmesg | grep bbswitch
[   12.871736] bbswitch: version 0.8
[   12.871745] bbswitch: Found integrated VGA device 0000:00:02.0:
\_SB_.PCI0.GFX0
[   12.871756] bbswitch: Found discrete VGA device 0000:01:00.0:
\_SB_.PCI0.PEG0.VGA_
[   12.871834] bbswitch: No suitable _DSM call found.
[   22.190138] bbswitch: version 0.8
[   22.190150] bbswitch: Found integrated VGA device 0000:00:02.0:
\_SB_.PCI0.GFX0
[   22.190163] bbswitch: Found discrete VGA device 0000:01:00.0:
\_SB_.PCI0.PEG0.VGA_
[   22.190251] bbswitch: No suitable _DSM call found.
[ 2566.348171] bbswitch: version 0.8
[ 2566.348176] bbswitch: Found integrated VGA device 0000:00:02.0:
\_SB_.PCI0.GFX0
[ 2566.348182] bbswitch: Found discrete VGA device 0000:01:00.0:
\_SB_.PCI0.PEG0.VGA_
[ 2566.348229] bbswitch: No suitable _DSM call found.

dmesg | grep acpi
[    0.000000] ACPI: LAPIC (acpi_id[0x01] lapic_id[0x00] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x02] lapic_id[0x01] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x03] lapic_id[0x02] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x04] lapic_id[0x03] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x05] lapic_id[0x04] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x06] lapic_id[0x05] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x07] lapic_id[0x06] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x08] lapic_id[0x07] enabled)
[    1.367011] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    1.367156] acpiphp_glue: can't evaluate _ADR (0x5)
[    7.246110] ACPI: Requesting acpi_cpufreq
[   10.153672] acpi device:38: registered as cooling_device9
[   10.154787] acpi device:44: registered as cooling_device10

sudo modprobe bbswitch
ERROR: could not insert 'bbswitch': No such device

cat /proc/bbswitch
cat: /proc/bbswitch: File not found

cat /proc/acpi/bbswitch
cat: /proc/acpi/bbswitch: File not found

ArchangeGabriel, 2015-02-16:
You must either upgrade to a newer kernel or provide us with the asked acpi dump (http://www.unix.com/man-page/freebsd/8/acpidump/).

runya-v, 2015-02-18:
In my system instold stable linux kernel for debian wheezy.

ArchangeGabriel, 2015-02-19:
I know, so that if you don’t want to upgrade, please provide us your acpi dump.

runya-v, 2015-02-19:
I dont understand you, 
I sent dump file "acpi.dump" in email, this file is large. 
Generate command: # sudo acpidump > acpi.dump.
This command is correct?

ArchangeGabriel, 2015-02-19:
Yes, but where did you send it?

runya-v, 2015-02-19:
in attached...

ArchangeGabriel, 2015-02-19:
Sorry, but that doesn’t work on GitHub. Can you send it by mail to bumblebee@lists.launchpad.net?

runya-v, 2015-02-19:
Ok, I sent.

runya-v, 2015-02-19:
It?

ArchangeGabriel, 2015-02-19:
Can’t see it there. Could you try sending to my address? You can find it here: https://launchpad.net/~b-pagani

runya-v, 2015-02-19:
I sent.

ArchangeGabriel, 2015-02-19:
Yes, received it. Will keep you updated here when this will be OK.

runya-v, 2015-02-19:
Thank you very much! I'll wait.

Lekensteyn, 2015-02-24:
The correct handle is `\_SB_.PCI0.PEG0.PEGP._DSM`. While the hack could be updated, are you able to upgrade your kernel? It has been fixed for over a year in newer Linux kernels.

Once you enable the backports repository, you can install [linux-image-3.16.0-0.bpo.4-amd64][1].

 [1]: https://packages.debian.org/wheezy-backports/linux-image-3.16.0-0.bpo.4-amd64

runya-v, 2015-02-25:
So this fix doesn`t exist in new version of kernel?
Will this fix be added in new version of kernel?
I cen`t install kernel header from back ports because dependencies crash.

ArchangeGabriel, 2015-02-25:
No, newer kernel (since 3.8.5) don’t need the acpi-handle-hack, because they have a better fix for this.

On older kernel, you need the acpi-handle-hack, modified for your machine. Lekensteyn can do this, but would prefer you to upgrade your kernel.

I see no reason to stay on debian stable today, Jessie will be released within a month or two, it’s already really stable, I’ve already upgraded my personal server, and where I work, the sysadmin is going to upgrade all the servers by April 6th, whether the final release happens before or not, because he is tired of solving integration bugs with newest softwares.

ArchangeGabriel, 2016-05-06:
Closing this one, wheezy is old-stable.

closed 2016-05-06 by ArchangeGabriel