Bumblebee-Project/bbswitch #108: Refused to change power state, /proc/acpi/bbswitch denies permission

SuperFluffy, 2015-03-06:
I have a rather new K620m on my Arch Linux system, kernel 3.19, and it appears like bbswitch is neither able to disable the discrete graphics, nor can I turn it off manually.

These issues seem related; while I have found both issues mentioned on here, I have not found them mentioned together.

````
% echo "OFF" > /proc/acpi/bbswitch
zsh: permission denied: /proc/acpi/bbswitch
````

The relevant output from dmesg after (succesfully) running `optirun glxspheres64`:
````
[   31.580217] [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:08:00.0 on minor 1
[   31.580222] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  346.47  Thu Feb 19 18:56:03 PST 2015
[   31.598180] vgaarb: this pci device is not a vga device
[   31.600203] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   31.600290] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   31.600355] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   31.600419] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   31.600482] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   31.600698] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   31.600884] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   31.601047] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   31.626459] ACPI Error: Field [TBF3] at 290816 exceeds Buffer [NULL] size 262144 (bits) (20141107/dsopcode-236)
[   31.626462] ACPI Error: Method parse/execution failed [\_SB_.PCI0.PEG_.VID_.GETB] (Node ffff8802454aee88), AE_AML_BUFFER_LIMIT (20141107/psparse-536)
[   31.626467] ACPI Error: Method parse/execution failed [\_SB_.PCI0.PEG_.VID_._ROM] (Node ffff8802454aee60), AE_AML_BUFFER_LIMIT (20141107/psparse-536)
[   31.628924] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   31.746135] vgaarb: this pci device is not a vga device
[   36.022118] [drm] Module unloaded
[   36.038922] bbswitch: disabling discrete graphics
[   36.038932] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   36.051329] pci 0000:08:00.0: Refused to change power state, currently in D0
````

Lekensteyn, 2015-03-07:
You need to be root to write to /proc/acpi/bbswitch. The refusal should not be a problem if the card gets disabled later. What is the contents of /proc/acpi/bbswitch after this?

SuperFluffy, 2015-03-07:
Hm, I thought having your user be part of the `bumblebee` group was sufficient? Either way, root access grants the permission, but it doesn't change the (reported) state:
```
% sudo sh -c 'echo "OFF" > /proc/acpi/bbswitch'
% cat /proc/acpi/bbswitch
0000:08:00.0 ON
```
The output of `dmesg` right after looks interesting:
```
[22507.839352] bbswitch: disabling discrete graphics
[22507.839365] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[22507.852372] ------------[ cut here ]------------
[22507.852379] WARNING: CPU: 1 PID: 17339 at drivers/pci/pci.c:1539 pci_disable_device+0x99/0xb0()
[22507.852380] pci 0000:08:00.0: disabling already-disabled device
[22507.852381] Modules linked in:
[22507.852382]  fuse ctr ccm bnep bluetooth uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common hid_generic videodev usbhid media hid snd_hda_codec_realtek mousedev snd_hda_codec_generic snd_hda_codec_hdmi msr arc4 coretemp intel_rapl iosf_mbi x86_pkg_temp_thermal intel_powerclamp iTCO_wdt iTCO_vendor_support iwlmvm kvm mac80211 crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel aes_x86_64 lrw gf128mul glue_helper iwlwifi ablk_helper cryptd serio_raw psmouse cfg80211 snd_hda_intel rtsx_pci_ms snd_hda_controller memstick snd_hda_codec thermal tpm_tis wmi ac battery e1000e tpm i2c_i801 snd_hwdep evdev snd_pcm mac_hid mei_me snd_timer mei ptp lpc_ich shpchp pps_core processor sch_fq_codel thinkpad_acpi nvram snd soundcore led_class rfkill hwmon nfs lockd grace
[22507.852426]  sunrpc fscache acpi_call(O) ext4 crc16 mbcache jbd2 sd_mod rtsx_pci_sdmmc mmc_core atkbd libps2 xhci_pci ahci libahci libata ehci_pci ehci_hcd xhci_hcd scsi_mod usbcore rtsx_pci usb_common i8042 serio bbswitch(O) i915 button intel_gtt i2c_algo_bit video drm_kms_helper drm i2c_core
[22507.852445] CPU: 1 PID: 17339 Comm: zsh Tainted: G           O   3.19.0-1-ARCH #1
[22507.852446] Hardware name: LENOVO 20E1S00N00/20E1S00N00, BIOS N11ET29W (1.05 ) 12/26/2014
[22507.852447]  0000000000000000 000000004c97d89e ffff8801b887bd88 ffffffff8155c81f
[22507.852450]  0000000000000000 ffff8801b887bde0 ffff8801b887bdc8 ffffffff81073aaa
[22507.852452]  ffff8801b887bda8 ffff8802441d9000 ffff88024418d5b0 00000000006b7300
[22507.852454] Call Trace:
[22507.852458]  [<ffffffff8155c81f>] dump_stack+0x4c/0x6e
[22507.852464]  [<ffffffff81073aaa>] warn_slowpath_common+0x8a/0xc0
[22507.852466]  [<ffffffff81073b35>] warn_slowpath_fmt+0x55/0x70
[22507.852469]  [<ffffffff812f528b>] ? __pci_set_master+0x3b/0x100
[22507.852471]  [<ffffffff812f81a9>] pci_disable_device+0x99/0xb0
[22507.852477]  [<ffffffffa000e235>] bbswitch_off+0xb5/0x260 [bbswitch]
[22507.852482]  [<ffffffffa000e635>] bbswitch_proc_write+0xa5/0xbc [bbswitch]
[22507.852485]  [<ffffffff81238b18>] proc_reg_write+0x48/0x70
[22507.852488]  [<ffffffff811d18a7>] vfs_write+0xb7/0x200
[22507.852490]  [<ffffffff811d2569>] SyS_write+0x59/0xd0
[22507.852493]  [<ffffffff815620a9>] system_call_fastpath+0x12/0x17
[22507.852494] ---[ end trace 02e925ea45f0a369 ]---
[22576.579898] bbswitch: disabling discrete graphics
[22576.579907] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
```

So is the card turned off but the report simply wrong? The other lines seem to be reminiscent of other bug reports found around here.

Lekensteyn, 2015-04-11:
The reported state reflects the actual state. Are you sure that nvidia (and nvidia-uvm if any) are unloaded?

petter3k, 2015-05-26:
Same problem her.

/proc/acpi/bbswitch shows correct info, "ON" when using for example nvidia-settings, and "OFF" when not. I can verify that the modules are indeed loaded and unloaded as they should. However, dmesg shows:

[  616.529891] bbswitch: disabling discrete graphics
[  616.529918] ACPI Warning: \_SB_.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150204/nsarguments-95)
[  616.542061] pci 0000:03:00.0: Refused to change power state, currently in D0

This is in Arch Linux with bbswitch 0.8, bumblebee 3.2.1 and nvidia-352.09

qezt, 2015-06-10:
same here, dmesg shows:

    [ 2316.938180] bbswitch: disabling discrete graphics
    [ 2316.938202] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)

mirtec001, 2015-06-11:
I'm having the same issue.  

[  576.847212] bbswitch: disabling discrete graphics
[  576.847249] ACPI Warning: \_SB_.PCI0.PEG0.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)

Running Ubuntu Mate 15.04

Here's the output of  lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
01:00.0 VGA compatible controller: NVIDIA Corporation GF119M [NVS 4200M] (rev ff)

palkakrzysiek, 2015-06-15:
try
```sudo chmod 666 /proc/acpi/bbswitch```

leoluk, 2015-06-15:
The `Argument #4 type mismatch` message is normal and not indicative of an error.

sharms, 2015-07-21:
Using Fedora 22 on a W541:

[root@soar  ~]$ cat /proc/acpi/bbswitch 
0000:01:00.0 ON
[root@soar  ~]$ tee /proc/acpi/bbswitch <<<OFF
OFF
[root@soar  ~]$ cat /proc/acpi/bbswitch 
0000:01:00.0 ON

Same error message: 
[  734.452018] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150204/nsarguments-95)

jdevelop, 2015-10-05:
Confirmed here on 4.2.2-1-ARCH #1 SMP PREEMPT Tue Sep 29 22:21:33 CEST 2015 x86_64 GNU/Linux on Lenovo T540P

robinhad, 2015-10-17:
Confirmed here on 4.2.0-16-generic #19-Ubuntu SMP Thu Oct 8 15:35:06 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

Lukypie, 2015-11-25:
always had this f****** problem, partially resolved by adding to the kernel the command 'acpi_osi=' and installing some acpi related packages (acpi, acpi_call), before formatting the pc I've also managed to get it fully working, but now the only thing I do is to rmmod manually the nvidia module everytime.
PS: I've got no problems before going standby, they came when I resume it.
But I think that are acpi problems

jdevelop, 2015-11-25:
```GRUB_CMDLINE_LINUX_DEFAULT="quiet acpi_osi=\"!Windows 2013\" acpi_osi=Linux"```

This one in ```/etc/default/grub``` solves the problem for me. Looks ugly.

robinhad, 2015-11-25:
optirun worked perfectly for my Nvidia Geforce 820M and Intel Ivy Bridge on Ubuntu 14.04.1

bmagyar, 2015-11-27:
Confirmed here too. Elementary OS Freya (based on Ubuntu 14.04) on Lenovo T440p with Geforce 730M

3.19.0-33-generic #38~14.04.1-Ubuntu SMP Fri Nov 6 18:17:28 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

None of the solutions above works for me.

`lsmod | grep nvidia` is empty

`sudo tee /proc/acpi/bbswitch <<<OFF`
results in
`cat /proc/acpi/bbswitch`
`0000:02:00.0 ON`

and what shows up in dmesg after this is
`[  558.744432] bbswitch: disabling discrete graphics`  
`[  558.744457] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)`

petter3k, 2015-11-27:
Can you please try `cat /proc/acpi/bbswitch` after bbswitch states that the card is off?

Also if you could `lsmod | grep nvidia` that would help too.

bmagyar, 2015-11-27:
Yeah, but actually I also realized that I forgot to do the `sudo update-grub` after modifying `grub.conf` to add `acpi_osi=Linux`.

Now I belong to the ever growing group of issue #78 

Can't these issues be related? Did anyone who has Lenovo T440p with Geforce 730M managed to fix this issue only by adding the acpi line?

klebed, 2015-12-22:
Same issue here with T440s. I guess it come after upgrading bios through lenovo crapware utility.

So bbswitch always say, that nvidia card is ON. 
If I try to sent OFF another time, I get this (bbswitch still reports ON):
[136839.918072] bbswitch: disabling discrete graphics
[136839.918083] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)

If I run glxgears, I see this:
[136903.547105] vgaarb: device changed decodes: PCI:0000:04:00.0,olddecodes=none,decodes=none:owns=none
[136903.547953] [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:04:00.0 on minor 1
[136903.547959] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  340.96  Sun Nov  8 22:33:28 PST 2015
[136903.565979] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136903.566061] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136903.566119] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136903.566175] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136903.566232] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136903.566436] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136903.566617] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136903.566675] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136903.604973] ACPI Error: Field [TBF3] at 524288 exceeds Buffer [NULL] size 262144 (bits) (20141107/dsopcode-236)
[136903.604978] ACPI Error: Method parse/execution failed [\_SB_.PCI0.PEG_.VID_.GETB] (Node ffff8803318b4be0), AE_AML_BUFFER_LIMIT (20141107/psparse-536)
[136903.604982] ACPI Error: Method parse/execution failed [\_SB_.PCI0.PEG_.VID_._ROM] (Node ffff8803318b4bb8), AE_AML_BUFFER_LIMIT (20141107/psparse-536)
[136903.605681] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136903.777764] thinkpad_acpi: asked for hotkey mask 0x0070ffbf, but firmware forced it to 0x0070ffbb
[136910.257821] [drm] Module unloaded
[136910.304604] bbswitch: disabling discrete graphics
[136910.304612] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[136910.319494] pci 0000:04:00.0: Refused to change power state, currently in D0

ArchangeGabriel, 2015-12-27:
People, please be more accurate when reporting things. ;)

So, what we have here:

* `/proc/acpi/bbswith` is not world-writable → that’s not an issue. If you want to manually write to it, run `sudo tee /proc/acpi/bbswitch <<< OFF` (or `ON`).

* `ACPI Warning: _SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires Package` → This is harmless, just ignore it.

* `Is there any non-Thinkpad laptop concerned?` → same thing.

Now, the real issues:

* ACPI ERRORs. Please open a dedicated thread if you see some, like @klebed.

* bbswitch not turning the card OFF. Please check if anything appears in `lsmod | grep nvidia`, if so look at https://github.com/Bumblebee-Project/Bumblebee/issues/699. Else, please open a new issue with accurate informations (like `cmdline`, laptop model and BIOS version…).

I’m closing this cluttered thread, if you don’t know what you should do you may still ask here so that I can tell you what to do given your situation.

closed 2015-12-27 by ArchangeGabriel

klebed, 2015-12-27:
ArchangeGabriel,

I'm sure that nothing is using discrete card, I've checked that and didn't noted it, assuming it's obvious. lsmod shows, that nvidia driver is unloaded.
It's just bbswitch issue, I guess. And it is very annoying. This issue causes card to drain my batteries so quickly! I want to get rid of it, but, right now, have no clue how. If you can guide me where to dig, and how to debug this, it will be helpful.

ArchangeGabriel, 2015-12-27:
@klebed Did you take a look at #112? You might be concerned. If not, please open a new issue with following information:

Title: [T440s] bbswitch can’t turn card OFF

Post your BIOS version (especially date if available), a dmesg log right after booting and one after running `optirun` (bash, glxinfo or whatever).

petter3k, 2016-01-21:
In Linux 4.3.3.3 (Arch x64):

```
[15943.514684] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  358.16  Mon Nov 16 18:58:09 PST 2015
[15979.104775] ACPI Warning: \_SB_.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150818/nsarguments-95)
[15979.112829] nvidia-modeset: Unloading
[15979.118159] nvidia-nvlink: Unregistered the Nvlink Core, major device number 246
[15979.118456] [drm] Module unloaded
[15979.125702] bbswitch: disabling discrete graphics
[15979.125711] ACPI Warning: \_SB_.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150818/nsarguments-95)
[15979.137791] pci 0000:03:00.0: Refused to change power state, currently in D0
[petterk@petterk-linux ~]$ cat /proc/acpi/bbswitch 
0000:03:00.0 OFF
```

petter3k, 2016-01-21:
Update:

lspci shows change to:

03:00.0 3D controller: NVIDIA Corporation GM108M [GeForce 840M] (rev ff)

and modules in unloading as they should. Is the "currently in D0" benign?

ArchangeGabriel, 2016-05-05:
@petter3k Sorry for the delay, and yes, this message is benign. ;)