Bumblebee-Project/bbswitch #109: powersaving centos 7.1 and type mismatch errors

HellboyRob, 2015-04-03:
Hi,

I am the maintainer for bumblebee and bbswitch for elrepo.
Since centos 7, powersavings are really bad, i used to be able to work for 2 hours on the same hardware, but after the upgrade, it is hardly an hour.. According to /proc/acpi/bbswitch the discrete videocard is off, but if i look at dmesg i see a lot of type mismatch errors, see the output below:

[    5.375077] bbswitch: version 0.8
[    5.375085] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[    5.375091] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[    5.375103] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20130517/nsarguments-95)
[    5.375226] bbswitch: detected an Optimus _DSM function
[    5.375240] pci 0000:01:00.0: enabling device (0006 -> 0007)
[    5.375292] bbswitch: disabling discrete graphics
[    5.375298] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20130517/nsarguments-95)
[    5.436973] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is off

I configured /etc/modprobe.d/bbswitch.conf with the following settings:

options bbswitch load_state=0 unload_state=1

For bumblebee i needed to create a custom selinux policy, to allow bumblebee to interact with bbswitch.

Please advice.

Thanks in advance.

Kind regards,

Rob

Lekensteyn, 2015-04-05:
Are you sure that it is related to your nvidia hardware? What write ON to bbswitch and leave it like that? It could also be a change in the CPU frequency driver or other components.

HellboyRob, 2015-04-05:
Hi,

The hardware hasn't changed. The laptop is 3 years old.
Do you need more informations about the hardware?

Lekensteyn, 2015-04-05:
While the hardware has not changed, the software did. Have you tried to isolate the issue to bbswitch or is it a guess?

HellboyRob, 2015-04-05:
I tried all kinds of settings. Rerun dracut etc.

Lekensteyn, 2015-04-05:
Ok, please try this and report the battery results:

1. Case 1: Disable the graphics card with bbswitch:

        echo OFF > /proc/acpi/bbswitch
2. Case 2: Enable the graphics card with bbswitch:

        echo ON > /proc/acpi/bbswitch

Also check your dmesg for messages related to loading the NVRM kernel module. This might also cause issues.

HellboyRob, 2015-04-07:
Thanks for the reply, i will do the tests and report back to you.

HellboyRob, 2015-04-08:
Hi,

I tested the following:

```
tee /proc/acpi/bbswitch <<<OFF
```

Nothing happened, then i turned the card on

```
tee /proc/acpi/bbswitch <<<ON
```

I see the following in dmesg:

```
[  954.139568] bbswitch: enabling discrete graphics
```

When i tried turning the card off again,  i see the following lines in dmesg:

```
[  964.082560] bbswitch: disabling discrete graphics
[  964.082591] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20130517/nsarguments-95)
```

I also noticed this line when i unplugged the powercord from the laptop:

```
[  892.434825] [Firmware Bug]: battery: (dis)charge rate invalid.
```

Kind regards,

Rob

Lekensteyn, 2015-04-11:
"Nothing happened" <-- what is the observed current draw? You noted that the battery life got degraded, but it is still not clear that this is caused by disfunctioning of bbswitch.

HellboyRob, 2015-04-13:
It looks like an issue with centos 7.1, i will report a bug.

closed 2015-04-13 by HellboyRob