Bumblebee-Project/bbswitch #110: Asus EB1037: bbswitch: No suitable _DSM call found

warpme, 2015-04-11:
Hi,
I'm trying to add support for Optimus in my Minimyth2 project. My testbed is ASUS EB1037 (BayTrail IGP + 820M dGPU). 
bbswitch seems to be non working on my test machine:
```
Apr  7 23:27:06 (none) user.info kernel: bbswitch: version 0.8
Apr  7 23:27:06 (none) user.info kernel: bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
Apr  7 23:27:06 (none) user.info kernel: bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.RP01.PXSX
Apr  7 23:27:06 (none) user.warn kernel: bbswitch: failed to evaluate \_SB_.PCI0.RP01.PXSX._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
Apr  7 23:27:06 (none) user.warn kernel: bbswitch: failed to evaluate \_SB_.PCI0.RP01.PXSX._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
Apr  7 23:27:06 (none) user.warn kernel: bbswitch: failed to evaluate \_SB_.PCI0.GFX0._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
Apr  7 23:27:06 (none) user.err kernel: bbswitch: No suitable _DSM call found.
```
I tested bbswitch 0.8 / bumblebee 3.2.1 with Nvidia 340.76 and 346.47 on 3.16.7, 3.19.3 and 4.0-rc6 kernels with all available BIOSes for this machine. All cases shows exactly the same problem.

I'm also experiencing problem with bumblebee: 
```
Apr  7 23:09:30 (none) user.warn kernel: NVRM: failed to copy vbios to system memory.
Apr  7 23:09:30 (none) user.warn kernel: NVRM: RmInitAdapter failed! (0x30:0xffff:800)
Apr  7 23:09:30 (none) user.warn kernel: NVRM: rm_init_adapter failed for device bearing minor number 0
Apr  7 23:09:30 (none) user.warn kernel: NVRM: nvidia_frontend_open: minor 0, module->open() failed, error -5
```
I'm not sure what additional info should I provide here for helping further debugging this issue.
pls advise

Lekensteyn, 2015-04-11:
See the README for required details. The acpidump output is particularly needed to perform an analysis.

warpme, 2015-04-12:
Pls find details about EB1037
Inserting bbswitch module:
```
modprobe: ERROR: could not insert 'bbswitch': No such device
root@FE-AsusEB1037:~ # dmesg | grep -C 10 bbswitch:
[ 2866.646193] bbswitch: version 0.8
[ 2866.646209] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[ 2866.646220] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.RP01.PXSX
[ 2866.646251] bbswitch: failed to evaluate \_SB_.PCI0.RP01.PXSX._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[ 2866.646311] bbswitch: failed to evaluate \_SB_.PCI0.RP01.PXSX._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[ 2866.646334] bbswitch: failed to evaluate \_SB_.PCI0.GFX0._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[ 2866.646339] bbswitch: No suitable _DSM call found.
root@FE-AsusEB1037:~ #
```
Kernel version:
```
# uname -a
Linux EB1037 3.17.4-301.fc21.x86_64 #1 SMP Thu Nov 27 19:09:10 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
```
ACPI tables
```
[root@EB1037 ~]# sh get-acpi-info.sh
Dumping ACPI tables...

Intel ACPI Component Architecture
ACPI Binary Table Extraction Utility version 20150408-64
Copyright (c) 2000 - 2015 Intel Corporation

Acpi table [DSDT] - 38358 bytes written to dsdt.dat
Acpi table [SSDT] - 1891 bytes written to ssdt1.dat
Acpi table [SSDT] - 656 bytes written to ssdt2.dat
Acpi table [SSDT] - 378 bytes written to ssdt3.dat
Acpi table [SSDT] - 2819 bytes written to ssdt4.dat
Acpi table [SSDT] - 956 bytes written to ssdt5.dat
Acpi table [SSDT] - 1075 bytes written to ssdt6.dat
Acpi table [SSDT] - 351 bytes written to ssdt7.dat
Acpi table [SSDT] - 141 bytes written to ssdt8.dat

Intel ACPI Component Architecture
ASL+ Optimizing Compiler version 20150408-64
Copyright (c) 2000 - 2015 Intel Corporation

Loading Acpi table from file   dsdt.dat - Length 00038358 (0095D6)
ACPI: DSDT 0x0000000000000000 0095D6 (v02 ALASKA A M I    01072009 INTL 20120913)
Acpi table [DSDT] successfully installed and loaded
Pass 1 parse of [DSDT]
Pass 2 parse of [DSDT]
Parsing Deferred Opcodes (Methods/Buffers/Packages/Regions)

Parsing completed

Found 1 external control methods, reparsing with new information
Pass 1 parse of [DSDT]
Pass 2 parse of [DSDT]
Parsing Deferred Opcodes (Methods/Buffers/Packages/Regions)

Parsing completed
Disassembly completed
ASL Output:    dsdt.dsl - 373954 bytes

iASL Warning: There were 1 external control methods found during
disassembly, but additional ACPI tables to resolve these externals
were not specified. The resulting disassembler output file may not
compile because the disassembler did not know how many arguments
to assign to these methods. To specify the tables needed to resolve
external control method references, the -e option can be used to
specify the filenames. Example iASL invocations:
    iasl -e ssdt1.aml ssdt2.aml ssdt3.aml -d dsdt.aml
    iasl -e dsdt.aml ssdt2.aml -d ssdt1.aml
    iasl -e ssdt*.aml -d dsdt.aml
[... 156 lines cut ...]
[root@EB1037 ~]#
```
ACPI dump
```
insmod acpi_dump_info.ko
[root@EB1037 acpi_dump_info]# cat /proc/acpi/dump_info
0000:00:00.0 060000 \_SB_.PCI0.VLVC
0000:00:02.0 030000 \_SB_.PCI0.GFX0
0000:00:13.0 010601 \_SB_.PCI0.SATA
0000:00:1a.0 108000 \_SB_.PCI0.SEC0
0000:00:1b.0 040300 \_SB_.PCI0.HDEF
0000:00:1c.0 060400 \_SB_.PCI0.RP01
0000:00:1c.1 060400 \_SB_.PCI0.RP02
0000:00:1c.2 060400 \_SB_.PCI0.RP03
0000:00:1c.3 060400 \_SB_.PCI0.RP04
0000:00:1d.0 0c0320 \_SB_.PCI0.EHC1
0000:00:1f.0 060100 \_SB_.PCI0.SBRG
0000:00:1f.3 0c0500 \_SB_.PCI0.SBUS
0000:01:00.0 030200 \_SB_.PCI0.RP01.PXSX
0000:02:00.0 020000 \_SB_.PCI0.RP02.PXSX
0000:03:00.0 0c0330 \_SB_.PCI0.RP03.PXSX
0000:04:00.0 028000 \_SB_.PCI0.RP04.PXSX
```

Please let me know if anything else is needed to help debuting problem

h220, 2015-08-06:
Any updates on this? I'm having the exact same problem on Asus EB1037.

warpme, 2015-08-06:
bbswitch seems to be dead project.
Better sell this EB1037 crap and buy something with real Nvidia (if You 
need Nvida)...

garsh, 2015-11-21:
Same issue with an EB1037.
I'd really like to get this thing working.
Is anyone willing to work on the problem?  I'll provide whatever additional information is required.

Lekensteyn, 2015-11-22:
The correct ACPI method is `\_SB.PCI0.RP01.PEGP._DSM`, not `\_SB_.PCI0.RP01.PXSX._DSM`. You could try the acpi-handle-hack branch after whitelisting your device, I can't dive deeper into this atm, sorry.

ArchangeGabriel, 2016-05-06:
@garsh Have you tried what @Lekensteyn suggested?

garsh, 2016-05-06:
I can't remember what all I tried.  I gave up, sold that machine, and replaced it with an Intel NUC NUC5PPYH.

Lekensteyn, 2016-06-05:
Just realized that this is a desktop machine that does not advertise hybrid graphics support. Closing

closed 2016-06-05 by Lekensteyn