Bumblebee-Project/bbswitch #111: bbswitch cannot switch OFF nvidia card after update to kernel  3.10.0-229.1.2.el7.x86_6

jeannotalpin, 2015-04-22:
Since last update of CentOS 7 (1503) and therefore kernel change to 3.10.0-229.1.2.el7.x86_64, bumblebee and bbswitch are not able to turn off nvidia card. Discrete card works fine but bug leads to high battery consumption. Everything was working fine on previous release. Method described here: http://elrepo.org/tiki/bumblebee  used to install bumblebee. 

Additional Information	[root@xxxxxxxxxxxxxx]# dmesg | grep -C 10 bbswitch
[ 4.189315] cfg80211: (5490000 KHz - 5710000 KHz @ 80000 KHz), (N/A, 2700 mBm), (0 s)
[ 4.189317] cfg80211: (57240000 KHz - 65880000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[ 4.461032] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[ 4.461036] Bluetooth: BNEP filters: protocol multicast
[ 4.461044] Bluetooth: BNEP socket layer initialized
[ 4.473521] SELinux: initialized (dev binfmt_misc, type binfmt_misc), uses genfs_contexts
[ 4.486009] Bluetooth: RFCOMM TTY layer initialized
[ 4.486021] Bluetooth: RFCOMM socket layer initialized
[ 4.486023] Bluetooth: RFCOMM ver 1.11
[ 4.585455] ip_tables: (C) 2000-2006 Netfilter Core Team
[ 4.631249] bbswitch: module verification failed: signature and/or required key missing - tainting kernel
[ 4.631530] bbswitch: version 0.8
[ 4.631537] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.VID_
[ 4.631544] bbswitch: Found discrete VGA device 0000:04:00.0: \_SB_.PCI0.PEG_.VID_
[ 4.631553] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20130517/nsarguments-95)
[ 4.632488] bbswitch: detected an Optimus _DSM function
[ 4.632505] pci 0000:04:00.0: enabling device (0000 -> 0003)
[ 4.632541] bbswitch: disabling discrete graphics
[ 4.632546] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20130517/nsarguments-95)
[ 4.644036] bbswitch: Succesfully loaded. Discrete card 0000:04:00.0 is on
[ 4.661872] bbswitch: disabling discrete graphics
[ 4.661885] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20130517/nsarguments-95)
[ 4.662706] ------------[ cut here ]------------
[ 4.662713] WARNING: at drivers/pci/pci.c:1523 pci_disable_device+0x84/0x90()
[ 4.662715] Device pci
disabling already-disabled device
[ 4.662716] Modules linked in:
[ 4.662718] iptable_mangle iptable_security bbswitch(OF) iptable_raw iptable_filter ip_tables rfcomm binfmt_misc bnep intel_powerclamp coretemp arc4 intel_rapl iTCO_wdt iTCO_vendor_support kvm crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel lrw gf128mul glue_helper ablk_helper cryptd iwlmvm mac80211 pcspkr cdc_mbim serio_raw cdc_wdm snd_hda_codec_hdmi i2c_i801 iwlwifi snd_hda_codec_realtek cdc_ncm cdc_acm snd_hda_codec_generic usbnet mii cfg80211 uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_core rtsx_pci_ms videodev memstick lpc_ich shpchp snd_hda_intel snd_seq snd_hda_controller snd_seq_device snd_hda_codec btusb snd_hwdep bluetooth snd_pcm mei_me mei snd_timer thinkpad_acpi snd soundcore rfkill nfsd auth_rpcgss nfs_acl lockd uinput sunrpc ext4 mbcache jbd2 sd_mod
[ 4.662770] crc_t10dif crct10dif_common rtsx_pci_sdmmc mmc_core i915 ahci libahci libata i2c_algo_bit drm_kms_helper drm rtsx_pci e1000e mfd_core ptp i2c_core pps_core wmi video dm_mirror dm_region_hash dm_log dm_mod
[ 4.662788] CPU: 1 PID: 790 Comm: bumblebeed Tainted: GF O-------------- 3.10.0-229.1.2.el7.x86_64 #1
[ 4.662790] Hardware name: LENOVO 20ARS44000/20ARS44000, BIOS GJET79WW (2.29 ) 09/03/2014
[ 4.662792] ffff88032a717e08 00000000afb992ee ffff88032a717dc0 ffffffff81604afa
[ 4.662796] ffff88032a717df8 ffffffff8106e34b ffff88032cce1000 ffff88032a717eb0
[ 4.662800] ffff88032a717f48 0000000000000028 0000000000000000 ffff88032a717e60
[ 4.662803] Call Trace:
[ 4.662810] [<ffffffff81604afa>] dump_stack+0x19/0x1b
[ 4.662815] [<ffffffff8106e34b>] warn_slowpath_common+0x6b/0xb0
[ 4.662818] [<ffffffff8106e3ec>] warn_slowpath_fmt+0x5c/0x80
[ 4.662822] [<ffffffff81303cf5>] ? __pci_set_master+0x35/0xf0
[ 4.662825] [<ffffffff81306ac4>] pci_disable_device+0x84/0x90
[ 4.662829] [<ffffffffa0876205>] bbswitch_off+0xc5/0x270 [bbswitch]
[ 4.662833] [<ffffffffa087664d>] bbswitch_proc_write+0x9d/0xac [bbswitch]
[ 4.662838] [<ffffffff8122e6ad>] proc_reg_write+0x3d/0x80
[ 4.662842] [<ffffffff811c6c5d>] vfs_write+0xbd/0x1e0
[ 4.662846] [<ffffffff811c76a8>] SyS_write+0x58/0xb0
[ 4.662850] [<ffffffff81614a29>] system_call_fastpath+0x16/0x1b
[ 4.662852] ---[ end trace 1faabfbedb67662e ]---
[ 4.664825] nf_conntrack version 0.5.0 (16384 buckets, 65536 max)
[ 4.687756] ip6_tables: (C) 2000-2006 Netfilter Core Team
[ 4.735927] Ebtables v2.0 registered
[ 4.749088] Bridge firewalling registered
[ 4.951697] IPv6: ADDRCONF(NETDEV_UP): enp0s25: link is not ready
--
[ 11.137017] iwlwifi 0000:03:00.0 wlp3s0: disabling HT/VHT due to WEP/TKIP use
[ 11.137663] wlp3s0: associate with 00:24:b2:96:74:a2 (try 1/3)
[ 11.140399] wlp3s0: RX AssocResp from 00:24:b2:96:74:a2 (capab=0x411 status=0 aid=2)
[ 11.150387] wlp3s0: associated
[ 11.150441] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
[ 12.451037] systemd-journald[514]: Received request to flush runtime journal from PID 1
[ 16.749367] Adjusting tsc more than 11% (7276734 vs 6913266)
[ 39.460320] fuse init (API version 7.22)
[ 39.463026] SELinux: initialized (dev fuse, type fuse), uses genfs_contexts
[ 39.480035] SELinux: initialized (dev fusectl, type fusectl), uses genfs_contexts
[ 85.095391] bbswitch: disabling discrete graphics
[ 85.095404] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20130517/nsarguments-95)

ArchangeGabriel, 2015-12-27:
Do you still have this issue?

jeannotalpin, 2016-01-23:
Problem is still present with kernel 3.10.0-327.4.4.el7.x86_64:
[  639.290565] bbswitch: disabling discrete graphics
[  639.290577] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20130517/nsarguments-95)

This branch https://github.com/smunaut/bbswitch/tree/hack-t440s seems to fix the problem though.

ArchangeGabriel, 2016-05-05:
OK, then it’s a duplicate of https://github.com/Bumblebee-Project/bbswitch/issues/112.

closed 2016-05-05 by ArchangeGabriel