Bumblebee-Project/bbswitch #113: T540p, Fedora 22, Kernel 4.1.X, dGPU refused to change power state

storrgie, 2015-08-01:
Was working well on Fedora 22 for a while, however when they updated to 4.1.X I noticed the dGPU was always ON. Please help. This machine is going to melt my legs or itself quite quickly (Lenovo T540p).

```
[agd@enyo ~]$ uname -a
Linux enyo 4.1.3-200.fc22.x86_64 #1 SMP Wed Jul 22 19:51:58 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```
```
[agd@enyo ~]$ cat /proc/cmdline 
BOOT_IMAGE=/vmlinuz-4.1.3-200.fc22.x86_64 root=UUID=1898cf22-bff7-4974-b836-61ef94a1d160 ro rootflags=subvol=root rd.luks.uuid=luks-9c941092-c8b7-497d-8bb6-2423567bbcd4 rd.luks.uuid=luks-2eac4914-e9ec-4510-9762-bc40bf36196c rd.luks.uuid=luks-574bf092-d0a6-4df7-b064-ee78f4b83008 nouveau.modeset=0 rd.driver.blacklist=nouveau "acpi_osi=!Windows 2013" rhgb quiet LANG=en_US.UTF-8
```
```
[agd@enyo ~]$ dmesg | grep -C 10 bbswitch
[   14.940550] usbcore: registered new interface driver uvcvideo
[   14.940552] USB Video Class driver (1.1.1)
[   15.043661] ieee80211 phy0: Selected rate control algorithm 'iwl-mvm-rs'
[   15.049443] iwlwifi 0000:03:00.0 wlp3s0: renamed from wlan0
[   15.051435] Bluetooth: Core ver 2.20
[   15.051446] NET: Registered protocol family 31
[   15.051446] Bluetooth: HCI device and connection manager initialized
[   15.051449] Bluetooth: HCI socket layer initialized
[   15.051450] Bluetooth: L2CAP socket layer initialized
[   15.051454] Bluetooth: SCO socket layer initialized
[   15.053679] bbswitch: version 0.8
[   15.053684] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.VID_
[   15.053691] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG_.VID_
[   15.053699] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[   15.054040] bbswitch: detected an Optimus _DSM function
[   15.054052] pci 0000:01:00.0: enabling device (0004 -> 0007)
[   15.054082] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[   15.056787] bbswitch: disabling discrete graphics
[   15.056797] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[   15.057369] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.057371] Bluetooth: BNEP filters: protocol multicast
[   15.057374] Bluetooth: BNEP socket layer initialized
[   15.071171] cfg80211: Calling CRDA for country: US
[   15.075489] cfg80211: Regulatory domain changed to country: US
[   15.075492] cfg80211:  DFS Master region: FCC
[   15.075493] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   15.075494] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[   15.075495] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 1700 mBm), (N/A)
--
[   26.265966] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
[   26.265967] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[   28.234359] nvidia: module license 'NVIDIA' taints kernel.
[   28.234362] Disabling lock debugging due to kernel taint
[   28.250824] vgaarb: device changed decodes: PCI:0000:01:00.0,olddecodes=io+mem,decodes=none:owns=none
[   28.251829] [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:01:00.0 on minor 1
[   28.251834] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  352.21  Tue Jun  9 21:53:31 PDT 2015
[   28.253141] nvidia_uvm: Loaded the UVM driver, major device number 244
[   28.255399] nvidia_uvm: Unregistered the UVM driver
[   35.406253] [drm] Module unloaded
[   35.421713] bbswitch: disabling discrete graphics
[   35.421722] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[   35.433395] pci_raw_set_power_state: 186 callbacks suppressed
[   35.433400] pci 0000:01:00.0: Refused to change power state, currently in D0
[  305.188573] vgaarb: device changed decodes: PCI:0000:01:00.0,olddecodes=none,decodes=none:owns=none
[  305.189589] [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:01:00.0 on minor 1
[  305.189593] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  352.21  Tue Jun  9 21:53:31 PDT 2015
[  305.217434] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  305.217508] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  305.217557] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  305.217604] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  305.217651] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  305.217820] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  305.217968] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  305.218016] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  305.369511] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  311.700821] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  311.706250] [drm] Module unloaded
[  311.719362] bbswitch: disabling discrete graphics
[  311.719376] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[  311.730359] pci 0000:01:00.0: Refused to change power state, currently in D0
```
```
[agd@enyo ~]$ sudo Xorg -version

X.Org X Server 1.17.2
Release Date: 2015-06-16
X Protocol Version 11, Revision 0
Build Operating System:  4.0.4-202.fc21.x86_64 
Current Operating System: Linux enyo 4.1.3-200.fc22.x86_64 #1 SMP Wed Jul 22 19:51:58 UTC 2015 x86_64
Kernel command line: BOOT_IMAGE=/vmlinuz-4.1.3-200.fc22.x86_64 root=UUID=1898cf22-bff7-4974-b836-61ef94a1d160 ro rootflags=subvol=root rd.luks.uuid=luks-9c941092-c8b7-497d-8bb6-2423567bbcd4 rd.luks.uuid=luks-2eac4914-e9ec-4510-9762-bc40bf36196c rd.luks.uuid=luks-574bf092-d0a6-4df7-b064-ee78f4b83008 nouveau.modeset=0 rd.driver.blacklist=nouveau "acpi_osi=!Windows 2013" rhgb quiet LANG=en_US.UTF-8
Build Date: 15 July 2015  08:16:41AM
Build ID: xorg-x11-server 1.17.2-2.fc22 
Current version of pixman: 0.32.6
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
```
```
[agd@enyo ~]$ sudo dnf info bbswitch.x86_64
Last metadata expiration check performed 0:23:12 ago on Sat Aug  1 10:11:27 2015.
Installed Packages
Name        : bbswitch
Arch        : x86_64
Epoch       : 0
Version     : 0.8.0
Release     : 1.fc22
Size        : 24 k
Repo        : @System
Summary     : Linux kernel module for Bumblebee
URL         : http://bumblebee-project.org/
License     : GPLv3
Description : bbswitch is a Linux kernel module which automatically detects the required
            : ACPI calls for two kinds of Optimus laptops. It has been verified to work
            : with "real" Optimus and "legacy" Optimus laptops.
            : 
            : See: https://github.com/Bumblebee-Project/bbswitch/
            : 
            : For further information.
            : 
            : For now, if you require nvidia module support with bumblebee you must
            : install bbswitch. In addition, even with nouveau, bbswitch will work
            : after a suspend whereas vga_switcharoo in the kernel might not. See:
            : https://github.com/Bumblebee-Project/Bumblebee/wiki/Comparison-of-PM-methods
            : for further information.
```

gsgatlin, 2015-08-01:
Hmnn. This is odd. I am typing this on t540p with the following results:

[gsgatlin@t540p ~]$ cat /proc/acpi/bbswitch 
0000:01:00.0 OFF
[gsgatlin@t540p ~]$ uname -a
Linux t540p.eos.ncsu.edu 4.1.2-200.fc22.x86_64 #1 SMP Wed Jul 15 20:12:12 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

If you boot into older kernel does it fix it?

storrgie, 2015-08-01:
@gsgatlin I tried running back to 4.1.2 and 4.0.8 but on 4.1.2 I get same behavior as above, and 4.0.8 there is no bbswitch in /proc/acpi

storrgie, 2015-08-01:
```
[agd@enyo ~]$ cat /proc/acpi/
bbswitch  button/   ibm/      wakeup    
[agd@enyo ~]$ cat /proc/acpi/bbswitch 
0000:01:00.0 ON
[agd@enyo ~]$ sudo tee /proc/acpi/bbswitch <<<OFF
[sudo] password for agd: 
OFF
[agd@enyo ~]$ cat /proc/acpi/bbswitch 
0000:01:00.0 ON
[agd@enyo ~]$ dmesg | grep -C 10 bbswitch
[   15.120618] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[   15.202958] Bluetooth: Core ver 2.20
[   15.203071] NET: Registered protocol family 31
[   15.203072] Bluetooth: HCI device and connection manager initialized
[   15.203076] Bluetooth: HCI socket layer initialized
[   15.203078] Bluetooth: L2CAP socket layer initialized
[   15.203082] Bluetooth: SCO socket layer initialized
[   15.209762] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.209764] Bluetooth: BNEP filters: protocol multicast
[   15.209767] Bluetooth: BNEP socket layer initialized
[   15.284503] bbswitch: version 0.8
[   15.284508] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.VID_
[   15.284512] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG_.VID_
[   15.284519] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[   15.284786] bbswitch: detected an Optimus _DSM function
[   15.284799] pci 0000:01:00.0: enabling device (0004 -> 0007)
[   15.284835] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[   15.286633] bbswitch: disabling discrete graphics
[   15.286639] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[   15.430138] nvidia: module license 'NVIDIA' taints kernel.
[   15.430140] Disabling lock debugging due to kernel taint
[   15.446317] vgaarb: device changed decodes: PCI:0000:01:00.0,olddecodes=io+mem,decodes=none:owns=none
[   15.446738] [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:01:00.0 on minor 1
[   15.446743] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  352.21  Tue Jun  9 21:53:31 PDT 2015
[   15.696884] nf_conntrack version 0.5.0 (65536 buckets, 262144 max)
[   15.717370] ip6_tables: (C) 2000-2006 Netfilter Core Team
[   15.756176] Ebtables v2.0 registered
[   15.768169] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
--
[   19.797354] cfg80211: Regulatory domain changed to country: US
[   19.797356] cfg80211:  DFS Master region: FCC
[   19.797356] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   19.797358] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[   19.797359] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 1700 mBm), (N/A)
[   19.797360] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2300 mBm), (0 s)
[   19.797361] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2300 mBm), (0 s)
[   19.797362] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
[   19.797362] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[   22.875396] fuse init (API version 7.23)
[   38.017261] bbswitch: device 0000:01:00.0 is in use by driver 'nvidia', refusing OFF
[agd@enyo ~]$ uname -a
Linux enyo 4.1.2-200.fc22.x86_64 #1 SMP Wed Jul 15 20:12:12 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

storrgie, 2015-08-01:
@gsgatlin I was a bit weirded out by the 'is in use by driver 'nvidia'' so I did a 'dnf reinstall bbswitch bumblebee-nvidia' and this is now the error I'm seeing:

```
[agd@enyo ~]$ cat /proc/acpi/bbswitch 
0000:01:00.0 ON
[agd@enyo ~]$ sudo tee /proc/acpi/bbswitch <<<OFF
[sudo] password for agd: 
OFF
[agd@enyo ~]$ dmesg | grep -C 10 bbswitch
[   17.900204] cfg80211: Calling CRDA for country: US
[   17.901770] cfg80211: Regulatory domain changed to country: US
[   17.901772] cfg80211:  DFS Master region: FCC
[   17.901773] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   17.901774] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[   17.901776] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 1700 mBm), (N/A)
[   17.901777] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2300 mBm), (0 s)
[   17.901778] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2300 mBm), (0 s)
[   17.901779] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
[   17.901780] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[   17.993188] bbswitch: version 0.8
[   17.993195] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.VID_
[   17.993200] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG_.VID_
[   17.993207] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[   17.993407] bbswitch: detected an Optimus _DSM function
[   17.993418] pci 0000:01:00.0: enabling device (0004 -> 0007)
[   17.993447] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[   17.996658] bbswitch: disabling discrete graphics
[   17.996668] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[   18.012733] Bluetooth: Core ver 2.20
[   18.012790] NET: Registered protocol family 31
[   18.012791] Bluetooth: HCI device and connection manager initialized
[   18.012794] Bluetooth: HCI socket layer initialized
[   18.012796] Bluetooth: L2CAP socket layer initialized
[   18.012800] Bluetooth: SCO socket layer initialized
[   18.018716] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   18.018719] Bluetooth: BNEP filters: protocol multicast
[   18.018723] Bluetooth: BNEP socket layer initialized
--
[   22.112366] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 3000 mBm), (N/A)
[   22.112367] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[   23.881390] [drm] Module unloaded
[   25.937688] fuse init (API version 7.23)
[   31.571575] vgaarb: device changed decodes: PCI:0000:01:00.0,olddecodes=none,decodes=none:owns=none
[   31.572469] [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:01:00.0 on minor 1
[   31.572474] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  352.21  Tue Jun  9 21:53:31 PDT 2015
[   31.573979] nvidia_uvm: Loaded the UVM driver, major device number 244
[   31.576179] nvidia_uvm: Unregistered the UVM driver
[   37.650007] [drm] Module unloaded
[   37.663361] bbswitch: disabling discrete graphics
[   37.663371] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
[   37.674337] pci_raw_set_power_state: 186 callbacks suppressed
[   37.674342] pci 0000:01:00.0: Refused to change power state, currently in D0
[   42.964848] bbswitch: disabling discrete graphics
[   42.964856] ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
```

gsgatlin, 2015-08-01:
This is probably not going to be very helpful but here is the output from a terminal window on this box...

http://fpaste.org/250557/41210143/

I see similarities with yours except on my system everything seems to work. The /proc/acpi/bbswitch file shows ON or OFF like its supposed to.

I will reboot and check something...

gsgatlin, 2015-08-01:
http://i.imgur.com/84kXVzN.jpg

Again, maybe not that helpful. But I can't really think of anything else.

Does it work ok except that bbswitch never shows off now? Like with primusrun?

storrgie, 2015-08-01:
I have primusrun installed, which might be mucking with it. I just use the dGPU so infrequently my goal is to ensure that it is absolutely turned off. You're saying that with primusrun it will always show ON but it will be in a lower powered state?

gsgatlin, 2015-08-01:
no. Sorry I was not clear.  I'm asking if 

primusrun glxgears -info

or

optirun -b primus glxgears -info

or 

optirun -b virtualgl glxgears -info

works. By works I mean does a window open with spinning gears in it? whichever. You don't have to do all three of them.  Just test one of them. the bumblebee daemon usually is able to turn the card on/off via bbswitch. Its working on this work provided system, a t540p. And I have the same kernel version as you do

[gsgatlin@t540p ~]$ systemctl status bumblebeed
● bumblebeed.service - Bumblebee C Daemon
   Loaded: loaded (/usr/lib/systemd/system/bumblebeed.service; enabled; vendor preset: disabled)
   Active: active (running) since Sat 2015-08-01 11:07:28 EDT; 14min ago
 Main PID: 839 (bumblebeed)
   CGroup: /system.slice/bumblebeed.service
           └─839 /usr/sbin/bumblebeed

Aug 01 11:07:28 t540p.eos.ncsu.edu systemd[1]: Started Bumblebee C Daemon.
Aug 01 11:07:28 t540p.eos.ncsu.edu systemd[1]: Starting Bumblebee C Daemon...
Aug 01 11:07:37 t540p.eos.ncsu.edu bumblebeed[839]: [   39.868825] [INFO]/usr...
Hint: Some lines were ellipsized, use -l to show in full.
[gsgatlin@t540p ~]$ cat /proc/acpi/bbswitch 
0000:01:00.0 OFF

storrgie, 2015-08-01:
@gsgatlin I can get the glxgears window to open, and I could paste the output but I don't think that would be helpful. The card however doesn't turn OFF, just stays ON always.

```
[agd@enyo ~]$ sudo systemctl status bumblebeed -l
● bumblebeed.service - Bumblebee C Daemon
   Loaded: loaded (/usr/lib/systemd/system/bumblebeed.service; enabled; vendor preset: disabled)
   Active: active (running) since Sat 2015-08-01 11:16:11 EDT; 15min ago
 Main PID: 5613 (bumblebeed)
   CGroup: /system.slice/bumblebeed.service
           └─5613 /usr/sbin/bumblebeed

Aug 01 11:17:28 enyo bumblebeed[5613]: [  124.653404] [ERROR][XORG] (EE) Failed to load module "mouse" (module does not exist, 0)
Aug 01 11:17:28 enyo bumblebeed[5613]: [  124.653409] [ERROR][XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied
Aug 01 11:30:30 enyo bumblebeed[5613]: [  137.354185] [WARN]Unable to disable discrete card.[  906.062672] [ERROR][XORG] (EE) systemd-logind: failed to get session: PID 10799 does not belong to any known session
Aug 01 11:30:30 enyo bumblebeed[5613]: [  906.062941] [ERROR][XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied
Aug 01 11:30:30 enyo bumblebeed[5613]: [  906.062944] [WARN][XORG] (WW) Warning, couldn't open module mouse
Aug 01 11:30:30 enyo bumblebeed[5613]: [  906.062946] [ERROR][XORG] (EE) Failed to load module "mouse" (module does not exist, 0)
Aug 01 11:30:30 enyo bumblebeed[5613]: [  906.062952] [WARN][XORG] (WW) NVIDIA(0): Unable to get display device for DPI computation.
Aug 01 11:30:30 enyo bumblebeed[5613]: [  906.062955] [WARN][XORG] (WW) Warning, couldn't open module mouse
Aug 01 11:30:30 enyo bumblebeed[5613]: [  906.062957] [ERROR][XORG] (EE) Failed to load module "mouse" (module does not exist, 0)
Aug 01 11:30:30 enyo bumblebeed[5613]: [  906.062961] [ERROR][XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied
```

That permission denied might be the issue?

gsgatlin, 2015-08-01:
I doubt it. I get that error also

[root@t540p ~]# tail -f /var/log/messages
Aug  1 11:36:22 t540p kernel: ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
Aug  1 11:36:22 t540p kernel: ACPI Warning: \_SB_.PCI0.PEG_.VID_._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150410/nsarguments-95)
Aug  1 11:36:22 t540p bumblebeed: [ 1764.944666] [ERROR][XORG] (EE) systemd-logind: failed to get session: PID 3222 does not belong to any known session
Aug  1 11:36:22 t540p bumblebeed: [ 1764.944925] [ERROR][XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied
Aug  1 11:36:22 t540p bumblebeed: [ 1764.944929] [WARN][XORG] (WW) Warning, couldn't open module mouse
Aug  1 11:36:22 t540p bumblebeed: [ 1764.944931] [ERROR][XORG] (EE) Failed to load module "mouse" (module does not exist, 0)
Aug  1 11:36:22 t540p bumblebeed: [ 1764.944938] [WARN][XORG] (WW) NVIDIA(0): Unable to get display device for DPI computation.
Aug  1 11:36:22 t540p bumblebeed: [ 1764.944941] [WARN][XORG] (WW) Warning, couldn't open module mouse
Aug  1 11:36:22 t540p bumblebeed: [ 1764.944944] [ERROR][XORG] (EE) Failed to load module "mouse" (module does not exist, 0)
Aug  1 11:36:22 t540p bumblebeed: [ 1764.944948] [ERROR][XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied

Maybe @Lekensteyn will have an idea of what to try next?

If you don't care that much about using the dGPU maybe bumblebee without bbswitch or nvidia would be a solution?

You have to install bumblebee-nouveau for that to work I think. So you'd have

bumblebee and bumblebee-nouveau

and you'd remove

bbswitch primus and bumblebee-nvidia

If you try that the interface to see if the card is on/off is at the path:
/sys/kernel/debug/vgaswitcheroo/switch

I have some notes on that here but they are buried in a long rambling notes.

testing the card. You must be root or use sudo:
 
(off)
[root@y470c ~]# cat /sys/kernel/debug/vgaswitcheroo/switch
0:IGD:+:Pwr:0000:00:02.0
1:DIS: :DynOff:0000:01:00.0
 
(on via "optirun" or "DRI_PRIME=1")
[root@y470c ~]# cat /sys/kernel/debug/vgaswitcheroo/switch
0:IGD:+:Pwr:0000:00:02.0
1:DIS: :DynPwr:0000:01:00.0
 
Card seems to stay on all the time. Or at very least is highly intermittent.
I thought it was working correctly in fc21 but after a re-install of fc21
it stays on all the time without bumblebee. Sadtrombone. Maybe its just my
laptops ACPI? With bumbleee installed however it did seem to work. But that was on a 
different laptop, not a t540p. It was on a ideapad y470. Its a bit older of a system.

http://fpaste.org/225738/64146143/

gsgatlin, 2015-08-01:
I guess the important part was this:

Card stays on all the time even if you don't use it.
 
[root@localhost ~]# cat /sys/kernel/debug/vgaswitcheroo/switch
0:IGD:+:Pwr:0000:00:02.0
1:DIS: :DynPwr:0000:01:00.0
 
Adding bumblebee (even without bbswitch) sems to "fix" this.
 
[root@localhost ~]# cat /sys/kernel/debug/vgaswitcheroo/switch
0:IGD:+:Pwr:0000:00:02.0
1:DIS: :DynOff:0000:01:00.0

gsgatlin, 2015-08-01:
I see there is a newer nvidia driver. Would you be interested is trying that?

storrgie, 2015-08-02:
I've installed from the bumblebee repo in Fedora 22, I am fine to try another nvidia driver (or use nouveau with vgaswitcheroo) just to ensure that the card is off. Main thing for me is to get the largest battery life I can.

gsgatlin, 2015-08-02:
Ok. So it sounds like to me the nvidia driver is not something you will want or care about. Its just a hassle.

So. Read everything but easy first way might be to try this:

dnf remove bumblebee-nvidia bbswitch bumblebee primus

(once thats done)

dnf install bumblebee-nouveau

(It will add bumblebee and I think virtualgl as a dependency) This comes from the bumblebee repository.

Then reboot. bumblebee-nouveau will stop the autoadd feature in the x startup (Option      "AutoAddGPU" "false") and also it changes some selinux settings so bumblebee has permissions.

Please note this disables prime. I would advise to use intel for everything. (Do not use optirun, it will not help honestly)
 
Then once your system boots as root check:

 /sys/kernel/debug/vgaswitcheroo/switch

and see if it says 

DynOff

in the output somewhere.

I'm not 100% sure how to turn it off "by hand" but bumblebee seem to be able to use that interface. I think maybe its 
echo OFF > /sys/kernel/debug/vgaswitcheroo/switch
But you will have to test that.

Please note I never tested this on a t540p but I would guess it should work. I did test bumblebee + nouveau on an older lenovo and it worked good in f22 and changed the state to DynOff until you used optirun.

Maybe you could even add "echo OFF > /sys/kernel/debug/vgaswitcheroo/switch" to a startup script (/etc/rc.d/rc.local) and not need bumblebee? The distro doesn't handle this sort of thing yet however by default. So both cards stay on without some kind of intervention. Let me know if that helped?

gsgatlin, 2015-08-02:
like this sample output from old lenovo:

[root@localhost ~]# cat /sys/kernel/debug/vgaswitcheroo/switch
0:IGD:+:Pwr:0000:00:02.0
1:DIS: :DynOff:0000:01:00.0

The above example is with the card turned off by bumblebee. This is what we want.

DIS is the nvidia chip. If it says

[root@localhost ~]# cat /sys/kernel/debug/vgaswitcheroo/switch
0:IGD:+:Pwr:0000:00:02.0
1:DIS: :DynPwr:0000:01:00.0

That means the nvidia chip is on and "wasting battery."

IGD is intel. It should always be on. The explinations of the various power states can be found here:

http://nouveau.freedesktop.org/wiki/Optimus/

Lekensteyn, 2015-08-02:
Can you try removing the `acpi_osi` option from your kernel command line? Is the problem visible in an older kernel? If neither nouveau (via dynpm, don't know if vgaswitcheroo still works) nor bbswitch can disable the card, then you should look in the ACPI direction.

smunaut, 2015-08-03:
I talked to @storrgie on IRC and decompiled his DSDT table.

It's exactly the same things as I have in issue #112 with the same "new" way of turning the card OFF.

Also that acpi_osi string does nothing ... the 4.1 kernel added the Windows 10 string and his bios version has windows 10 support, so the new behavior is enabled in the ACPI. (confirmed it in the DSDT table)

storrgie, 2015-08-05:
can confirm with @smunaut that his patch has my system working again.

ArchangeGabriel, 2015-12-27:
So, this is a duplicate of #112 too. Closing as so.

closed 2015-12-27 by ArchangeGabriel