Bumblebee-Project/bbswitch #115: GTX970m still on and overheating while suspended

BernardoGO, 2015-10-25:
I have a Clevo P650SE-A(Sager NP8651) with NVidia GeForce GTX970m and Intel HD5600, the laptop does have a LED that shows if the discrete GPU is being used or not.
I got bumblebee successfully working on both Ubuntu 14 and 15, I have noticed that while turning on, off or going to sleep the laptop turns on the dGPU LED.
The problem is that it leaves the GeForce on while sleeping. Which leads to overheat with the fans off and a waste of 15-20% per hour while sleeping. It is actually heating and using more battery while off than it does on.

Hibernating is not an option because it cannot resume after hibernation(black screen with cursor) and I don't really want it to hibernate since I'm used to suspend it many times per day.

This happens with Ubuntu 14 and 15. I'm using the 3.19 kernel because the 4.2 does not seems to support my video card.
The sleep problem does not happens on Windows using optimus.

I'm not using UEFI, does it have something to do with it?

```
[    6.003280] init: plymouth-upstart-bridge main process (266) terminated with status 1
[    6.003337] init: plymouth-upstart-bridge main process ended, respawning
[    6.004558] init: plymouth-upstart-bridge main process (267) terminated with status 1
[    6.004614] init: plymouth-upstart-bridge main process ended, respawning
[    6.005717] init: plymouth-upstart-bridge main process (269) terminated with status 1
[    6.005774] init: plymouth-upstart-bridge respawning too fast, stopped
[   13.951982] Adding 15904764k swap on /dev/sda5.  Priority:-1 extents:1 across:15904764k FS
[   14.077132] systemd-udevd[369]: starting version 204
[   14.167793] lp: driver loaded but no devices found
[   14.174461] [drm] Initialized drm 1.1.0 20060810
[   14.185978] bbswitch: module verification failed: signature and/or  required key missing - tainting kernel
[   14.186065] bbswitch: version 0.7
[   14.186068] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   14.186073] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[   14.186079] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   14.186122] bbswitch: detected an Optimus _DSM function
[   14.186132] pci 0000:01:00.0: enabling device (0000 -> 0003)
[   14.186164] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[   14.188705] ppdev: user-space parallel port driver
[   14.255353] wmi: Mapper loaded
[   14.299229] shpchp: Standard Hot Plug PCI Controller Driver version: 0.4
[   14.312440] Bluetooth: Core ver 2.20
[   14.312452] NET: Registered protocol family 31
[   14.312453] Bluetooth: HCI device and connection manager initialized
[   14.312455] Bluetooth: HCI socket layer initialized
[   14.312457] Bluetooth: L2CAP socket layer initialized
[   14.312462] Bluetooth: SCO socket layer initialized
[   14.314616] usbcore: registered new interface driver btusb
--
[   16.967846] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.967848] Bluetooth: BNEP filters: protocol multicast
[   16.967851] Bluetooth: BNEP socket layer initialized
[   17.097357] init: cups main process (934) killed by HUP signal
[   17.097362] init: cups main process ended, respawning
[   18.053497] r8169 0000:03:00.1 eth0: link down
[   18.053521] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   18.054570] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[   18.054757] iwlwifi 0000:04:00.0: L1 Enabled - LTR Enabled
[   18.068951] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   23.143307] bbswitch: disabling discrete graphics
[   23.143316] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[   29.139937] systemd-hostnamed[1623]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[   46.982117] audit_printk_skb: 168 callbacks suppressed
[   46.982119] audit: type=1400 audit(1445801247.879:68): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=1669 comm="apparmor_parser"
[   46.982123] audit: type=1400 audit(1445801247.879:69): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=1669 comm="apparmor_parser"
[   46.982343] audit: type=1400 audit(1445801247.879:70): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=1669 comm="apparmor_parser"
[  115.014906] wlan0: authenticate with 00:26:3e:52:11:02
[  115.018714] wlan0: direct probe to 00:26:3e:52:11:02 (try 1/3)
[  115.221157] wlan0: direct probe to 00:26:3e:52:11:02 (try 2/3)
[  115.425080] wlan0: direct probe to 00:26:3e:52:11:02 (try 3/3)
```

```
Linux bernardo-P650SE-A 3.19.0-31-generic #36~14.04.1-Ubuntu SMP Thu Oct 8 10:21:08 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

Ubuntu 14.10

https://bugs.launchpad.net/debian/+bug/752542/+attachment/4504886/+files/CLEVO-P65xSE-A.tar.gz

firetech, 2015-11-04:
I have the same issue on my Clevo P651SE (Schenker XMG P505), running Debian.

However, I'm quite sure it isn't bbswitch's fault per se. If I boot into a Live USB of Ubuntu, without bumblebee or nvidia drivers installed, the same overheating happens (nothing touches the dGPU, so nothing turns it off after BIOS).

I've "solved" my problem by patching bbswitch to turn off the dGPU before suspend instead of turning it on. This gets rid of the overheating, but I need to reboot after suspend if I want to use the dGPU.

My (somewhat uneducated) guess is that this is either an issue in the BIOS/EC of these computers or something that Linux does/doesn't do during suspend (it works without overheating in Windows, which also, according to the dGPU power LED, turns on the dGPU just before suspend).

BernardoGO, 2015-11-04:
As you can see, it is a normal behavior to turn on the dGPU before going to sleep or turning off.
I have tried it on Windows too and turns it on too right before suspension but there is no heating or battery drain after that. I have noticed some coil whine while suspended using Linux with bbswitch that does not happen using windows too.

I don't think it is related to the bios or to the laptop itself because it does not happen using Windows 8, 10 or FreeBSD.
Hibernation is a problem too since it never recovers from it.

How did you managed to make bbswitch turn off the video card before suspension? I have tried to do that without success.
It will not solve the problem but at least I'll be able to suspend.

firetech, 2015-11-04:
Here's my patch for bbswitch 0.8: https://gist.github.com/firetech/831ca1ea11c6fba4864f

It's, of course, by no means a patch intended for public use oror integration, it's just a quick hack to make suspend work properly on my laptop.

BenThompson22, 2015-11-04:
I have this problem too with my MSI GE62 with GTX970m on Arch Linux.
The fix that @firetech posted is definitely not a fix that should be used at all. Fixes like this only help to delay the real fix as it makes people to think it is ok to live with the problem.

If you bought a laptop with a high-end gpu you need to use it instead of disabling it to be able to suspend.
That ridiculous, I have seen people complaining of this since October of 2014 and still no fix.

firetech, 2015-11-04:
I actually agree.

However, my hack doesn't disable the dGPU, everything works perfectly until suspend. After you suspend and resume at least once, you'll need to reboot to be able to use the card again.

I wrote that hack about a year ago, mostly as a test to determine whether it was the dGPU heating up or not, then stuck with it. Back then, I researched the issue, but didn't find anyone with the same problem.

I do use my 970m, but I mostly do so from within Windows, using Linux for everything not game related. I also re-research the issue now and then, which brought me here.

Anyhow, since the overhearing issue happens without bbswitch as well, the problem either lies in hardware/BIOS or in the Linux kernel itself.

BernardoGO, 2015-11-04:
@firetech 
Do you know if there is a way to turn the dGPU completely off on linux?
I use it on Linux for some games but can live without it and use it only for gaming on windows.
Then it would be great to have it always off.

Lekensteyn, 2015-11-06:
Can you try the nouveau driver with runtime power management? bbswitch is currently ignorant of some flags that became documented in kernel commit 5addcf0a5f0fadceba6bd562d0616a1c5d4c1a4d. Those bits may actually be significant here.

firetech, 2015-11-06:
Does the nouveau driver actually support the 970m? The wiki doesn't even mention that GPU, and it's family contains a lot of WIPs. Is the wiki outdated?

BernardoGO, 2015-11-06:
In order to set the runtime power management on I just have to use the Nouveau, right?
If I don't install the nvidia driver it does not use the video card.

mmedum, 2015-11-06:
Hey, so i think im hitting the same problem with a lenovo w550s with the nvidia quadro 620M, even through bbswitch reports the gpu to off, im still not seeing any change to the power draw. Im running bumblebee with the nvidia driver on Arch Linux 4.2.5-1. 
As it stands i can actually do on and off both manually and with optirun and by doing that get reports that the nvidia card should be off, but well it's not really powering down and thereby i get terrible battery time.

sharms, 2015-11-06:
@mmedum - I agree, I have a W541 and I don't believe the Nvidia K2100 is shutting off due to the heat I am feeling.  I have a windows install to compare it to, and there is a significant difference in heat in the palm rest where the gpu is

BernardoGO, 2015-11-07:
Do you have this issues while the system is on or while suspended?
For me I don't know if it is on when it is running because the P650SE is really cool, but while suspended it gets hot because the fans are off.

BenThompson22, 2015-11-07:
I'm moving to Windows. No one seems to care for this problem, I'm not a programmer and can't fix this by myself.
I see many posts on NVIDIA forums since the start of 2014 and still it has problems.
I would pay for someone to solve it if I find someone who can solve this.

JonasVivi, 2015-11-07:
I have a P670SE and it has the exactly same problem.

Found those:

https://unix.stackexchange.com/questions/213962/huge-battery-drain-during-suspend-to-ram-12-hour-nvidia-card-doesnt-turn-o
https://devtalk.nvidia.com/default/topic/855456/gtx970m-power-consumption-during-standby/
https://bbs.archlinux.org/viewtopic.php?id=195931
https://unix.stackexchange.com/questions/214297/nvidia-card-doesnt-turn-off-during-suspend-to-ram?lq=1
https://devtalk.nvidia.com/default/topic/893126/gtx970m-powered-on-and-overheating-while-suspended-/?offset=3#4718636
http://forum.notebookreview.com/threads/geforce-does-not-turn-off-while-suspended-heating-cannot-hibernate.782781/

NVidia pushes the problem to Linux kernel, that pushes it to bumblebee or prime that pushes it to the firmware that pushes it to NVidia.
Problems related to suspend/hibernate using Optimus are known since 2007,

mmedum, 2015-11-07:
@BernardoGO Im not hearing anything when suspended, it looks like it suspending correctly without troubles. Well I'll try to look in to the code for this particular problem, havn't done much hardware programming. @BenThompson22 @sharms @BernardoGO, could i get specs with your graphics card and which linux distribution your running. Would it be possible to get the dmesg from bbswitch also?

BernardoGO, 2015-11-07:
@mmedum For P650SE-A(mine) it is:
Intel i7-5700HQ 
Geforce GTX970m 3GB
16gb RAM DDR3 

For P651SE and P670SE it is:
Intel i7-4720HQ
Geforce GTX970m 3GB

For MSI GE62:
Intel i7 4710HQ or 5700HQ
Geforce GTX970m 3GB

Do you need more details for the video card? If yes, which details?

mmedum, 2015-11-07:
@BernardoGO thanks! I try looking into it tomorrow. Have you tried both the nvidia and nouveau driver?

BernardoGO, 2015-11-07:
@mmedum 
Yes, I have tried it on Ubuntu and Fedora. Both comes with Nouveau after installation.
The video card does not work on Nouveau but still heats while suspended.

ArchangeGabriel, 2015-12-27:
Well indeed nouveau can’t help with 9xx cards (missing signed firmware that NVIDIA refused to ship for more than one year now).

Suspend is definitively not working well in Linux world, and especially for Optimus laptop. I’m adding this on the list of big issues that need a summary of the current state before being handed-on to capable people (i.e. not me).

Lekensteyn, 2016-01-21:
While nouveau cannot drive the display engine, runtime power management seems functional (it is powering off the GPU). Can confirm this for a GTX 965M (Clevo P651RA) on Linux 4.4 (which has other problems though, such as running in an infinite loop when powering on again).

Let's see how this works... if anyone knows how to do ACPI method tracing on Windows 8.1/10, please chime in. Some information I found is use of ETW / Tracelog, though it does not look too effective here.
https://msdn.microsoft.com/en-us/library/windows/hardware/ff549603%28v=vs.85%29.aspx
https://msdn.microsoft.com/en-us/library/windows/hardware/ff552994%28v=vs.85%29.aspx
http://www.magicsplat.com/book/event_tracing.html

BernardoGO, 2016-01-21:
@Lekensteyn P65xRA is one of the new batch of Clevos with Skylake, does the heating issues while suspended happens on this batch too?
So yours can't use the GTX965 at all, can suspend properly but cant resume?

Lekensteyn, 2016-01-21:
@BernardoGO I haven't really used the new laptop due to the ACPI issue, no idea if it gets heated or not. I did notice that the laptop couldn't last two days in suspended and unplugged state, maybe that is related?

I have also not tried to start Xorg at all on this thing, I think nouveau needs some patching too (dmesg is attached to https://lkml.kernel.org/g/20160119100503.GA13617@al).

Let me know if you want me to run some tests, there is no important data on this laptop, I'll wipe it later again. It currently has Win8.1 and Win10 installed next to Arch Linux. At the moment I am downloading a 3.6GiB checked Windows 10 build just for a single checked acpi.sys file... Then I will try the AMLi debugger mentioned in https://github.com/Bumblebee-Project/bbswitch/issues/78#issuecomment-31794575 and https://msdn.microsoft.com/en-us/library/windows/hardware/ff551079%28v=vs.85%29.aspx

BernardoGO, 2016-01-22:
Have you tried using the nvidia driver with prime? 
I found that my laptop has a very faint high frequency coil whine that I'm assuming it is coming from the GPU. Can't say if it happens while awake because the cooler noise may overlap it, but while suspended I can hear it if I lie my ear on the bottom of the laptop.
I have noticed that it only happens in the same situations that heats the laptop and waste battery, while suspended on windows or free BSD it emits no noise at all. It only occurs if suspended by Linux. 

This problem does not appear in the power management logs, the laptop itself also seems to be on S3 state.

BernardoGO, 2016-02-04:
@Lekensteyn 
I have tried the Linux Kernel 4.5 and it still have the same problem to sleep even if I use the nouveau driver.
Left it sleeping for a few minutes and it used 2% of battery. It also had the coil whine

BernardoGO, 2016-03-19:
Is there any news about it?

BernardoGO, 2016-03-22:
@Lekensteyn 
I had some improvements on getting the geforce to be powered down to sleep. My laptop would not last 6 hours suspended on Arch.
Now it is actually going to sleep with the card off and resuming with it still off. Sometimes I can even turn it on after resume, but, lets say, 20% of the times I simply can't. I can't even modprobe nvidia, i says "no such device".

Now, more than ever, I think that this is bbswitch related. Because to do so I only have to modprobe -r bbswitch and it suspends fine.
I know it is supposed to suspend with it on, and guess what, Windows also does that. But, unlike linux with bbswitch, Windows goes to a hybrid sleep where it can perform some actions while sleeping such as flushing all the memory to HD(hibernation after sleep if the battery goes low) or, the most important part, turn the video card off.

I have disassembled the laptop and checked the video card state after sleeping on my three test systems. Here I have Windows 10, Arch with default configurations for nvidia with bbswitch and Manjaro with some scripts that I've wrote for it to suspend properly.
With Arch it turns the video card on, suspends and leave it on. Battery lasts around six hours before going to 0%.
With Windows it also sleeps with the video card on but after 5 minutes(It always take exactly 5 minutes) the video card turns off while sleeping. It is not hibernating or flushing memory to disk, it is only turning the video card off. Maybe if I suspend it with less battery it may hibernate.
With Manjaro running my sleep script it never turns the videocard on since it removes bbswitch before performing sleep. But it also can't turn it back on sometimes after resume. Computer works fine but without the geforce.

Now I need your help with this. I really need to make this work in a reliable way, I've been trying to use hibernation but it also troublesome with this computer.(At this point i'm starting to wonder why did i bought this)
What I really want to know is that if is there a way to make bbswitch to work just like with it is in windows. Maybe perform this action after going to sleep or if it can force the nvidia to modprobe after resume.

Lekensteyn, 2016-04-06:
FYI, I started analyzing this some weeks ago, but will only be able to look at it again next week. My approach is to analyze the AMLi traces from Windows 10 Debug (obtained via the kernel debugger over UDP) and try events like suspending/resuming/activating Nvidia card/etc.

Unfortunately these do not contain timing information (and with time logging enabled, the logs flood and are too slow to be processed). Therefore I captured packets instead and will use the pcap timestamps to relate relevant log entries using a custom Wireshark dissector (https://github.com/Lekensteyn/kdnet). This is all work in progress.

BernardoGO, 2016-04-06:
@oanonymos0
I can confirm that having something running prior to suspend does decrease the amount of energy wasted while suspended somehow but it is still wasting a way too much and still overheating. By doing this it used 12% per hour and its heat were noticeable by my back while it was inside of my backpack. Also, it uses a lot of battery to initiate the opengl program which makes it unfeasible.

@Lekensteyn 
I have modified some parts of the code to make suspend reliable for P650SE. It is more like a nasty workaround but it works and since my hibernation is unreliable also I had to make something about it.
Also I cannot lose my GTX970m after suspend since I use it mostly for image processing on linux and the Intel is a key factor for me to have this laptop to last almost 8 hours.
So far I have 5 days of uptime with my modified bbswitch and a battery waste of 2% per hour while suspended which is pretty nice considering that it was wasting more than 40% per hour. Now I'm trying to make it reliable while suspended in D3Cold, right now i'm using D3hot. Maybe I can get 1% per hour with it.
After suspending like this the video card stop working. To turn it back on I have added a new state to /proc/acpi/bbswitch that when passed to this file bbswitch forces the computer to suspend with the video card in D0. Then, on resume it is possible to use it and even turn it on/off.
With the new code I was able to remove the bash scripts that I was using to resume and suspend.

I saw that you have some problems related to suspend on Lenovo once, do you think that it is possible to do the same here to make it use the videocard on cold resume without having to suspend it back again in P0?
Is possible to patch the linux kernel so that bbswitch can turn the videocard off after it changes the PCI table so that we don't lose it on cold suspend?

JeorgeMatt, 2016-04-06:
My issue was closed since it seems to be similar to this one. I'm not sure if it is the same problem, the laptop is also different. Ill copy it here anyway to keep it documented. 

I recently decided to use Ubuntu as my main OS in Eurocom M5 Pro, but I have noticed something strange with bbswitch.
After installing it is draining a way too much battery while suspended to run. In windows or without bbswitch it uses about 3% per hour.
Everything is working perfectly but the system is impossible to use as a laptop since the suspend to ram does not last very long and the hibernation is unreliable.

Robson002, 2016-04-08:
I have this problem too with MSI GE62. It only works if I disable bbswitch or disable the videocard completely in bios.
Is there any official fix for this? Or maybe a estimated time

jacobmischka, 2016-04-24:
I'm also having this problem, with a 940m on an Acer e5-574g. Is there a better way to tell bbswitch to not reenable the card before suspend than simple rmmodding it?

CykaBlyat22, 2016-04-24:
@jacobmischka 
If you rmod it bofere suspending it do suspend wheel?
What hapens if you dont rmod it before?

When it returns everything works wheel?

I have the same problem with a 940m also in a acer. I do not know the version thought

hundredyearslate, 2016-04-24:
@jacobmischka 
I think that they might have abandoned this project actually.
No updates since 2013, abandoned repo and issues. It obviously does not work anymore for this new batch of computers with the geforce 900 series.

No one even tried to help me with this issue https://github.com/Bumblebee-Project/bbswitch/issues/131 which is somehow related.

Maybe we need to start over a new bbswitch. We can call it bbswitcheroo.

jacobmischka, 2016-04-24:
If I don't stop bbswitch before suspending it enables the card like it's supposed to, and my laptop doesn't disable it like it's supposed to, which uses power.

Yeah, looks like bbswitch is pretty dead. I would have used PRIME actually except it looks like nouveau doesn't work with my card at all, it didn't detect it. I'll just continue doing this for now, I don't use the card much anyway.

Robson002, 2016-04-24:
Why isn't switcheroo fully supported and why is is supposed to be better than bbswitch?
Is there any official alternative from nvidia for us?

Nouveau does not work for the new video cards anymore.

@mmedum @firetech @BernardoGO @sharms 
Did you guys have any news about it? I seriously thinking on selling this computer and getting a macbook.

jacobmischka, 2016-04-24:
Er, sorry, I meant the drivers need to support it. And the proprietary ones don't.

BenThompson22, 2016-04-24:
@hundredyearslate @jacobmischka 
Do you guys think that it might be good to start developing a new alternative to bbswitch as you mentioned?
If so, I would like to participate.

jacobmischka, 2016-04-24:
I mean, it probably would, but that's a significant undertaking and one I'm not equipped to know anything about.

mmedum, 2016-04-24:
Hmm

Well honestly, I think that bbwitch is a good starting point for actually
supporting nvidia 900 series and above. We could try to poke nvidia, maybe
start a dialog about it? I don't get any overheating issues with my quadro
card, instead I experience problems related to the card always running,
which in total destroys my battery life.

BenThompson22, 2016-04-24:
@mmedum 

What kills bbswitch as starting point is that it seems to be abandoned and even when it wasn't its developers are known to ignore/reject pull requests and sometimes, even worse, they reject a pull request and copy the modifications to push as if they wrote it.

How can we poke nvidia? Don't you think that they probably know about it and just don't care? Optimus is a reality that we deal with since ~2009.

jacobmischka, 2016-04-24:
They don't really care about notebooks it seems. Frankly I'm glad enough that the drivers themselves are as good as they are, I'm not surprised about the lack of interest in Optimus.

Lekensteyn, 2016-04-25:
@hundredyearslate @jacobmischka
vgaswitcheroo and nouveau both use the same ACPI methods as bbswitch. Last time I checked vgaswitcheroo does not work across suspend/resume. nouveau does handle s/r and does some additional initialization but can also be turned on for PRIME.

@BenThompson22 You must be confusing projects, if you look at the commit logs you will see proper attributions were applicable. Now if you have some magic insight that fixes the current problems, please do contribute (to bbswitch, Linux/nouveau, etc.) :-)

As for efforts to improve PM for the new Maxwell laptops, nouveau devs got firmware last month and support should improve much starting from Linux 4.6. vgaswitcheroo has some changes (http://patchwork.ozlabs.org/patch/594848/) for newer systems, I don't know its current status though.

bbswitch will eventually be fixed... My current laptop from 2011 is dieing, but I have not found sufficient time to hack on my new laptop (from January this year, has a GTX 965M). Once I find some time between studies and work, hopefully then I will be able to further analyze the PM process on Windows 10.

If you have some spare time, you can try to do the same using the Windows Kernel Debugger. Because the power management messages are time-sensitive and WinDbg could not keep up with the flood of messages, I tried an approach based on network captures (https://github.com/Lekensteyn/kdnet). Earlier ACPI trace files got lost in a power disruption, so the experiments have to be redone. (time permitting...)

GregoryHupkins, 2016-04-27:
@Lekensteyn 
So is it happening for all the new batch of windows 10 laptops using linux?
I thought it was just for clevos and some MSI.

So maybe it is an attempt from microsoft to make linux gamers return to windows since they are losing many of them to ubuntu because of steam.

jacobmischka, 2016-04-27:
No they're not. Linux gaming is to this day a negligible market share. This has literally nothing to do with Microsoft.

oanonymos0, 2016-04-27:
question: does it suspend correctly with nvidia-settings (optirun -b none nvidia-settings -c :8 ) running in the background? 

Maybe bbswitch needs to load the nvidia driver on suspend (if it doesnt already) [ correct me if i'm wrong here], then unload it on resume and turn off the card if no apps are using it. With nvidia-settings open it suspends correctly on my GT 650M Samsung NP550P5C ubuntu 14.04 kernel 3.19 nvidia-352

oanonymos0, 2016-04-27:
@BernardoGO hmm instead of an OpenGL program, try having nvidia-settings open before suspend. For my GT 650M it works correctly this way. (optirun -b none nvidia-settings -c :8)

firetech, 2016-04-27:
@oanonymos0 That trick doesn't work for me (970M, Debian Jessie). The GPU was hotter after having the laptop suspended for a few minutes (~40°C) than it was before suspending (~35°C).

jacobmischka, 2016-04-27:
I haven't tried @oanonymous0's idea but your hardware and software is pretty significantly different than the ones mentioned here, you probably are having a different problem that that helps with.

BernardoGO, 2016-04-28:
@oanonymos0 

It doesn't work for me. Currently I'm using my modified version of bbswitch.

oanonymos0, 2016-04-28:
@jacobmischka my reasoning was that maybe the your card(s) will suspend correctly if the nvidia driver is loaded during sleep.

HumbleJosh, 2016-04-29:
@oanonymos0 
It makes sense but unfortunately  it doesn't quite work because bbswitch expects the computer to turn it off.

jacobmischka, 2016-04-29:
Yeah, it was a fine suggestion, I'm just explaining why it's not working for people here. Ultimately, this is a hardware/firmware issue and not really bbswitch's fault.

jacobmischka, 2016-05-05:
Just confirming that this is still an issue for me in Linux 4.6 rc6, I read 4.6 had a lot of graphics improvements so I hoped this might be something that was fixed, but unfortunately not.

BernardoGO, 2016-05-05:
Which graphic improvements did you have?

jacobmischka, 2016-05-05:
4.6 fixed an unrelated issue I was having with a flickering/hanging display: https://bugs.freedesktop.org/show_bug.cgi?id=94161.

BernardoGO, 2016-05-05:
Was it the black flickering in full screen Windows?
I had it fixed by using Wayland or KDE.

It used to happen mostly when things are changed in the screen

jacobmischka, 2016-05-05:
No, it was over aggressive power saving that would make the skylake integrated graphics just stop working for a fraction of a second at random times.

BernardoGO, 2016-05-05:
Do you have this black flickering as well?

jacobmischka, 2016-05-05:
I did until 4.6, yes.

BernardoGO, 2016-05-06:
I'm trying it right now. Just installed it. Version: 4.6.0-1-ARCH 
I can see that indeed the problem is not fixed also for Broadwell. 
But it seems like my flickering in Gnome is not happening anymore. I have to try it for a little longer before confirming it.

Have you tried it with Nouveau? I saw somewhere that the kernel 4.6 allied to the new nouveau will solve the problem related to the suspension.

[26 more replies]