Bumblebee-Project/bbswitch #118: HP ZBook 15 G2: bbswitch: No suitable _DSM call found

rkrell, 2015-12-14:
During booting my HP ZBook 15 G2 (nVidia GK208GLM [Quadro K610M], BIOS v31/03/2015, Hybrid Graphics setting in BIOS: Auto) I get three times the following output:
`bbswitch: No suitable _DSM call found.`

Hardware:
- HP ZBook 15 G2 Mobile Workstation
- Integrated: Intel HD Graphics 4600
- Discrete: NVIDIA® Quadro® K610M
- see http://h20195.www2.hp.com/v2/GetDocument.aspx?docname=c04434046

```
dmesg | grep -C 10 bbswitch
[    6.062086] psmouse serio3: synaptics: queried min coordinates: x [1324..], y [1248..]
[    6.076584] clocksource: Switched to clocksource tsc
[    6.120154] psmouse serio3: synaptics: Touchpad model: 1, fw: 8.1, id: 0x1e2b1, caps: 0xd40123/0x840300/0x26800/0x0, board id: 2706, fw id: 1486004
[    6.144044] systemd-journald[154]: Received SIGTERM from PID 1 (systemd).
[    6.156378] input: SynPS/2 Synaptics TouchPad as /devices/platform/i8042/serio3/input/input7
[    6.172605] usb 3-5: new full-speed USB device number 3 using xhci_hcd
[    6.242602] random: nonblocking pool is initialized
[    6.301987] usb 3-5: New USB device found, idVendor=138a, idProduct=003f
[    6.301990] usb 3-5: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[    6.301991] usb 3-5: SerialNumber: 0030b43df893
[    6.314213] bbswitch: version 0.8
[    6.314222] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEGP.DGFX
[    6.314234] ACPI Warning: \_SB_.PCI0.PEGP.DGFX._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150818/nsarguments-95)
[    6.315034] ACPI Warning: \_SB_.PCI0.PEGP.DGFX._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150818/nsarguments-95)
[    6.315223] bbswitch: No suitable _DSM call found.
[    6.320794] EXT4-fs (sda2): re-mounted. Opts: acl,discard
[    6.372741] usb 3-1.2: new low-speed USB device number 4 using xhci_hcd
[    6.485229] usb 3-1.2: New USB device found, idVendor=03f0, idProduct=0024
[    6.485232] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.485233] usb 3-1.2: Product: HP Basic USB Keyboard
[    6.485234] usb 3-1.2: Manufacturer: CHICONY
[    6.485314] usb 3-1.2: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[    6.504610] systemd-journald[443]: Received request to flush runtime journal from PID 1
[    6.542341] ACPI: AC Adapter [AC] (on-line)
[    6.543001] Initializing HPQ6001 module
```
\-\-
```
[    6.637836] snd_hda_intel 0000:01:00.1: Disabling MSI
[    6.637844] snd_hda_intel 0000:01:00.1: Handle VGA-switcheroo audio client
[    6.648951] systemd[1]: langset.service: Cannot add dependency job, ignoring: Input/output error
[    6.649246] input: HP WMI hotkeys as /devices/virtual/input/input15
[    6.652947] usb 3-7: new high-speed USB device number 5 using xhci_hcd
[    6.657269] iwlwifi 0000:5c:00.0: CONFIG_IWLWIFI_DEBUG enabled
[    6.657271] iwlwifi 0000:5c:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[    6.657272] iwlwifi 0000:5c:00.0: CONFIG_IWLWIFI_DEVICE_TRACING disabled
[    6.657273] iwlwifi 0000:5c:00.0: Detected Intel(R) Centrino(R) Advanced-N 6235 AGN, REV=0xB0
[    6.657374] iwlwifi 0000:5c:00.0: L1 Enabled - LTR Disabled
[    6.679835] bbswitch: version 0.8
[    6.679843] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEGP.DGFX
[    6.679852] ACPI Warning: \_SB_.PCI0.PEGP.DGFX._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150818/nsarguments-95)
[    6.680428] snd_hda_codec_idt hdaudioC0D0: autoconfig for 92HD91BXX: line_outs=1 (0xa/0x0/0x0/0x0/0x0) type:line
[    6.680431] snd_hda_codec_idt hdaudioC0D0:    speaker_outs=1 (0xd/0x0/0x0/0x0/0x0)
[    6.680433] snd_hda_codec_idt hdaudioC0D0:    hp_outs=1 (0xb/0x0/0x0/0x0/0x0)
[    6.680435] snd_hda_codec_idt hdaudioC0D0:    mono: mono_out=0x0
[    6.680436] snd_hda_codec_idt hdaudioC0D0:    inputs:
[    6.680438] snd_hda_codec_idt hdaudioC0D0:      Mic=0xc
[    6.680440] snd_hda_codec_idt hdaudioC0D0:      Internal Mic=0x11
[    6.680441] snd_hda_codec_idt hdaudioC0D0:      Line=0xf
[    6.680563] ACPI Warning: \_SB_.PCI0.PEGP.DGFX._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150818/nsarguments-95)
[    6.680894] bbswitch: No suitable _DSM call found.
[    6.681019] lis3lv02d: 8 bits 3DC sensor found
[    6.684780] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[    6.685685] iwlwifi 0000:5c:00.0 wlo1: renamed from wlan0
[    6.686727] Adding 2095100k swap on /dev/sdb1.  Priority:-1 extents:1 across:2095100k FS
[    6.693571] ACPI: Battery Slot [BAT0] (battery present)
[    6.693794] ACPI: Battery Slot [BAT1] (battery absent)
[    6.701477] SGI XFS with ACLs, security attributes, realtime, no debug enabled
[    6.717757] XFS (sdb2): Mounting V5 Filesystem
[    6.737612] input: HDA Digital PCBeep as /devices/pci0000:00/0000:00:1b.0/sound/card0/input16
[    6.738342] input: HDA Intel PCH Mic as /devices/pci0000:00/0000:00:1b.0/sound/card0/input17
```
\-\-
```
[    7.232958] ACPI Error: Method parse/execution failed [\_TZ.GFXZ._PSL] (Node ffff88017fcbb748), AE_NOT_FOUND (20150818/psparse-542)
[    7.232963] ACPI: Invalid passive threshold
[    7.316261] Bluetooth: Core ver 2.20
[    7.316286] NET: Registered protocol family 31
[    7.316288] Bluetooth: HCI device and connection manager initialized
[    7.316293] Bluetooth: HCI socket layer initialized
[    7.316297] Bluetooth: L2CAP socket layer initialized
[    7.316302] Bluetooth: SCO socket layer initialized
[    7.318935] usbcore: registered new interface driver btusb
[    7.322230] Linux video capture interface: v2.00
[    7.323924] bbswitch: version 0.8
[    7.323934] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEGP.DGFX
[    7.323947] ACPI Warning: \_SB_.PCI0.PEGP.DGFX._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150818/nsarguments-95)
[    7.324267] usbcore: registered new interface driver usbhid
[    7.324269] usbhid: USB HID core driver
[    7.324791] ACPI Warning: \_SB_.PCI0.PEGP.DGFX._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150818/nsarguments-95)
[    7.325123] bbswitch: No suitable _DSM call found.
[    7.325764] input: CHICONY HP Basic USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2/3-1.2:1.0/0003:03F0:0024.0001/input/input24
[    7.326511] uvcvideo: Found UVC 1.00 device HP HD Webcam (04f2:b3ed)
[    7.327728] input: HP HD Webcam as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/input/input25
[    7.381827] hid-generic 0003:03F0:0024.0001: input,hidraw0: USB HID v1.10 Keyboard [CHICONY HP Basic USB Keyboard] on usb-0000:00:14.0-1.2/input0
[    7.381895] usbcore: registered new interface driver uvcvideo
[    7.381897] USB Video Class driver (1.1.1)
[    7.496926] thermal LNXTHERM:01: registered as thermal_zone2
[    7.496931] ACPI: Thermal Zone [GFXZ] (20 C)
[    7.893461] thermal LNXTHERM:02: registered as thermal_zone3
[    7.893466] ACPI: Thermal Zone [EXTZ] (0 C)
```

```
uname -a
Linux rkrell 4.3.0-2-default #1 SMP PREEMPT Sat Nov 14 16:19:19 UTC 2015 (734b32c) x86_64 x86_64 x86_64 GNU/Linux
```

**Distribution:** OpenSUSE Tumbleweed 20151209
**Xorg:** version 7.6, X-Server: 1.18.0
**Driver:** nvidia 352.63

rkrell, 2015-12-14:
```
cat /proc/acpi/dump_info
0000:00:00.0 060000 
0000:00:01.0 060400 \_SB_.PCI0.PEGP
0000:00:14.0 0c0330 \_SB_.PCI0.XHC_
0000:00:16.0 078000 
0000:00:19.0 020000 \_SB_.PCI0.LANC
0000:00:1a.0 0c0320 \_SB_.PCI0.EHC2
0000:00:1b.0 040300 \_SB_.PCI0.HDEF
0000:00:1c.0 060400 \_SB_.PCI0.RP01
0000:00:1c.4 060400 \_SB_.PCI0.RP05
0000:00:1c.6 060400 \_SB_.PCI0.RP07
0000:00:1c.7 060400 \_SB_.PCI0.RP08
0000:00:1d.0 0c0320 \_SB_.PCI0.EHC1
0000:00:1f.0 060100 \_SB_.PCI0.LPCB
0000:00:1f.2 010601 \_SB_.PCI0.SATA
0000:00:1f.3 0c0500 \_SB_.PCI0.SBUS
0000:01:00.0 030000 \_SB_.PCI0.PEGP.DGFX
0000:01:00.1 040300 
0000:5c:00.0 028000 
0000:5d:00.0 ff0000
```

rkrell, 2015-12-14:
**Machine information:**:
- https://bugs.launchpad.net/lpbugreporter/+bug/752542/comments/776
- https://bugs.launchpad.net/lpbugreporter/+bug/752542/comments/777

ArchangeGabriel, 2015-12-27:
@Lekensteyn Broken handle too?

Lekensteyn, 2016-05-11:
When `\_SB.PCI0.PEGP.DGFX._DSM (UUID, RevID, func, args)` is called with the Nvidia Optimus UUID, it will invoke:
```
Name (MUID, ToUUID ("4004a400-917d-4cf2-b89c-79b62fd55665")) // no idea, "DGPU _DSM", requires RevId==0x300
Name (NUID, ToUUID ("d4a50b75-65c7-46f7-bfb7-41514cea0244")) // no idea, requires RevId==0x0102
Name (VUID, ToUUID ("a486d8f8-0bda-471b-a72b-6042a6b5bee0")) // "Optimus"

Method (_DSM, 4, NotSerialized) {
    ....
    if (\SRCP(Arg0, MUID, 0x01)) { // Is Arg0 == MUID?
        Return (MDSM (Arg0, Arg1, Arg2, Arg3))
    }
    if (\SRCP(Arg0, NUID, 0x01)) { // Is Arg0 == NUID?
        Return (NDSM (Arg0, Arg1, Arg2, Arg3))
    }
    if (\SRCP(Arg0, VUID, 0x01)) { // Is Arg0 == VUID?
        Return (NVOP (Arg0, Arg1, Arg2, Arg3))
    }
    Return (0x80000002)
}
```

But eh... NVOP is not quite implemented:
```
Scope (\_SB.PCI0.PEGP.DGFX) {
    Method (NVOP, 4, Serialized) {
        Debug = "------- NV OPTIMUS DSM --------"
        Return (0x80000002)
    }
}
```

MDSM seems not usable for power management (it can only provide info). NDSM on the other hand seems more interesting.

...but looking through the available methods, there is no `_PS3`/`_PS0` nor `_PR3`. I do not think that the video card can be disabled on this laptop. Correct me if I'm wrong, does Windows allow you to disable the GPU on this?

rkrell, 2016-05-12:
> Correct me if I'm wrong, does Windows allow you to disable the GPU on this?

I will ask a colleague which runs Windows 7 on the same notebook.
Can you recommend a reliable method how to figure this out? We are not sure...

Lekensteyn, 2016-05-12:
I don't know a very reliable method, but what could possibly be used is the Nvidia control panel. When I open the 3D settings tab with the rotating Nvidia logo, it has a short (<1sec) delay before it shows the icon. Other things to look at are BIOS settings and the manual (does it mention some kind of hybrid graphics)?

If you happen to run a checked/debug build of Windows, then you could try to attach a kernel debugger remotely and use `!amli set verboseon spewon` to log all ACPI method calls. (This is more advanced stuff which is much harder to do.)

Lekensteyn, 2016-05-28:
@rkrell The product specification mentions Optimus support and in the BIOS (sp73534.exe, M70_0113.bin, 01.13 Rev.A 28 Oct 2015) there is a trace of a functional Optimus function.

What BIOS version do you exactly have? I cannot find your version mentioned on one of these pages;
- [Revision History 01.12-01.14 Rev.A (28 Jan 2016)](http://h20564.www2.hp.com/hpsc/swd/public/detail?sp4ts.oid=6978815&swItemId=ob_160357_1&swEnvOid=4158#tab-history)
- [Revision History 01.02-01.08 Rev.A (17 Mar 2015)](http://h20564.www2.hp.com/hpsc/swd/public/detail?sp4ts.oid=6978815&swItemId=ob_146439_1&swEnvOid=4158#tab-history)

rkrell, 2016-05-28:
@Lekensteyn I have a different versioning schema for my notebook, it's an HP ZBook 15 G2 (http://h20565.www2.hp.com/portal/site/hpsc/public/kb/search/?sp4ts.oid=6978814) and I'm currently running the latest one offered by the BIOS online update: 01.34
See http://h20564.www2.hp.com/hpsc/swd/public/detail?swItemId=ob_146940_1#tab-history
Head of output of dmidecode:
```
# dmidecode 3.0
Getting SMBIOS data from sysfs.
SMBIOS 2.7 present.
36 structures occupying 1937 bytes.
Table at 0x5EE3F000.

Handle 0x0011, DMI type 0, 24 bytes
BIOS Information
        Vendor: Hewlett-Packard
        Version: L70 Ver. 01.34
        Release Date: 10/28/2015
        Address: 0xF0000
        Runtime Size: 64 kB
        ROM Size: 8192 kB
        Characteristics:
                PCI is supported
                PC Card (PCMCIA) is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                ACPI is supported
                USB legacy is supported
                Smart battery is supported
                BIOS boot specification is supported
                Function key-initiated network boot is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.34
        Firmware Revision: 148.85
```

rkrell, 2016-05-29:
> Other things to look at are BIOS settings and the manual (does it mention some kind of hybrid graphics)?

There is such a setting Hybrid Graphics - set to _Auto_. But even _Enabled_ didn't help.

Lekensteyn, 2016-05-30:
Can you post a fuller dmesg containing the list of discovered ACPI tables? At least include `dmesg | grep ACPI:`. You have a SSDT identified by NVIDIAGF which contains a stub Optimus function which does nothing. In your firmware there is also a trace of a NVSGTBL table that does have Optimus functionality. Not sure if that helps much since the firmware also has stuff for AMD cards which could indicate that the same code was used for multiple machines.

Can you ask your collegue to provide ACPI tables when running Windows? He can do so by getting the "acpidump.exe" program from the zip archive at https://www.acpica.org/downloads/binary-tools, open an admin command prompt, navigate to the directory containing and execute:

    acpidump > acpidump-windows.txt

Please include the BIOS version as well.

rkrell, 2016-05-30:
> Can you post a fuller dmesg containing the list of discovered ACPI tables? At least include dmesg | grep ACPI:

```
dmesg | grep ACPI:
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F3060 000024 (v02 HPQOEM)
[    0.000000] ACPI: XSDT 0x000000005EFFE120 0000AC (v01 HPQOEM SLIC-MPC 00000001      01000013)
[    0.000000] ACPI: FACP 0x000000005EFFC000 00010C (v05 HPQOEM 1909     00000001 HP   00000001)
[    0.000000] ACPI: DSDT 0x000000005EFD1000 02591A (v02 HPQOEM 1909     00000001 INTL 20110112)
[    0.000000] ACPI: FACS 0x000000005EE31000 000040
[    0.000000] ACPI: FACS 0x000000005EE31000 000040
[    0.000000] ACPI: HPET 0x000000005EFFB000 000038 (v01 HPQOEM 1909     00000001 HP   00000001)
[    0.000000] ACPI: APIC 0x000000005EFFA000 0000BC (v01 HPQOEM 1909     00000001 HP   00000001)
[    0.000000] ACPI: MCFG 0x000000005EFF9000 00003C (v01 HPQOEM 1909     00000001 HP   00000001)
[    0.000000] ACPI: TCPA 0x000000005EFF7000 000032 (v02 HPQOEM 1909     00000000 HP   00000001)
[    0.000000] ACPI: SSDT 0x000000005EFCE000 000466 (v01 HPQOEM SataAhci 00001000 INTL 20110112)
[    0.000000] ACPI: SSDT 0x000000005EFCD000 00048A (v01 HPQOEM PtidDevc 00001000 INTL 20110112)
[    0.000000] ACPI: SLIC 0x000000005EFCC000 000176 (v01 HPQOEM SLIC-MPC 00000001 HP   00000001)
[    0.000000] ACPI: MSDM 0x000000005EFCB000 000055 (v03 HPQOEM SLIC-MPC 00000000 HP   00000001)
[    0.000000] ACPI: FPDT 0x000000005EFCA000 000044 (v01 HPQOEM 1909     00000001 HP   00000001)
[    0.000000] ACPI: BGRT 0x000000005EFC9000 000038 (v00 HPQOEM 1909     00000001 HP   00000001)
[    0.000000] ACPI: SSDT 0x000000005EFC6000 001A30 (v01 HPQOEM NVIDIAGF 00000001 INTL 20110112)
[    0.000000] ACPI: SSDT 0x000000005EFC3000 000544 (v01 PmRef  Cpu0Ist  00003000 INTL 20110112)
[    0.000000] ACPI: SSDT 0x000000005EFC2000 000AF3 (v01 PmRef  CpuPm    00003000 INTL 20110112)
[    0.000000] ACPI: SSDT 0x000000005EFC1000 0001D5 (v01 PmRef  LakeTiny 00003000 INTL 20110112)
[    0.000000] ACPI: SSDT 0x000000005EFC0000 000706 (v01 SaSsdt SaSsdt   00003000 INTL 20110112)
[    0.000000] ACPI: ASF! 0x000000005EFF8000 0000A5 (v32 HPQOEM 1909     00000001 HP   00000001)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] ACPI: PM-Timer IO Port: 0x1808
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x00] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x01] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x02] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x03] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x04] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x05] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x06] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x07] high edge lint[0x1])
[    0.000000] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.000000] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.000000] ACPI: IRQ0 used by override.
[    0.000000] ACPI: IRQ9 used by override.
[    0.000000] ACPI: HPET id: 0x8086a201 base: 0xfed00000
[    0.000042] ACPI: Core revision 20160108
[    0.023075] ACPI: 8 ACPI AML tables successfully acquired and loaded
[    0.398620] ACPI: bus type PCI registered
[    0.422941] ACPI: Added _OSI(Module Device)
[    0.422943] ACPI: Added _OSI(Processor Device)
[    0.422944] ACPI: Added _OSI(3.0 _SCP Extensions)
[    0.422945] ACPI: Added _OSI(Processor Aggregator Device)
[    0.436815] [Firmware Bug]: ACPI: BIOS _OSI(Linux) query ignored
[    0.441423] ACPI: Dynamic OEM Table Load:
[    0.441431] ACPI: SSDT 0xFFFF88017C55B000 0003F3 (v01 PmRef  Cpu0Cst  00003001 INTL 20110112)
[    0.442370] ACPI: Dynamic OEM Table Load:
[    0.442376] ACPI: SSDT 0xFFFF88089DB95000 0005DB (v01 PmRef  ApIst    00003000 INTL 20110112)
[    0.443397] ACPI: Dynamic OEM Table Load:
[    0.443402] ACPI: SSDT 0xFFFF88017C767200 000119 (v01 PmRef  ApCst    00003000 INTL 20110112)
[    2.453652] ACPI: Interpreter enabled
[    2.453678] ACPI: (supports S0 S3 S4 S5)
[    2.453679] ACPI: Using IOAPIC for interrupt routing
[    2.456609] ACPI: Power Resource [APPR] (off)
[    2.460087] ACPI: Power Resource [COMP] (on)
[    2.460555] ACPI: Power Resource [LPP] (on)
[    2.465307] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-fe])
[    2.497248] ACPI: PCI Interrupt Link [LNKA] (IRQs 3 4 5 6 *10 11 12 14 15)
[    2.497315] ACPI: PCI Interrupt Link [LNKB] (IRQs 3 4 5 6 *10 11 12 14 15)
[    2.497379] ACPI: PCI Interrupt Link [LNKC] (IRQs 3 4 5 6 *10 11 12 14 15)
[    2.497442] ACPI: PCI Interrupt Link [LNKD] (IRQs 3 4 5 6 *10 11 12 14 15)
[    2.497505] ACPI: PCI Interrupt Link [LNKE] (IRQs 3 4 *5 6 10 11 12 14 15)
[    2.497567] ACPI: PCI Interrupt Link [LNKF] (IRQs 3 4 5 6 10 11 12 14 15) *0, disabled.
[    2.497631] ACPI: PCI Interrupt Link [LNKG] (IRQs 3 4 5 6 10 *11 12 14 15)
[    2.497693] ACPI: PCI Interrupt Link [LNKH] (IRQs 3 4 5 6 10 11 12 14 15) *0, disabled.
[    2.498076] ACPI: Enabled 6 GPEs in block 00 to 3F
[    2.515547] pnp: PnP ACPI: found 9 devices
[    4.975513] ACPI: Sleep Button [SLPB]
[    4.975683] ACPI: Lid Switch [LID]
[    4.975804] ACPI: Power Button [PWRF]
[    4.977008] ACPI: Video Device [DGFX] (multi-head: yes  rom: yes  post: no)
[    4.992494] ACPI: bus type USB registered
[    5.805379] ACPI: AC Adapter [AC] (on-line)
[... cut ...]
[    5.991851] ACPI: Battery Slot [BAT0] (battery present)
[    5.991965] ACPI: Battery Slot [BAT1] (battery absent)
[    6.401498] ACPI: Thermal Zone [CPUZ] (60 C)
[    6.677614] ACPI: Invalid passive threshold
[    6.941305] ACPI: Thermal Zone [GFXZ] (24 C)
[    7.335322] ACPI: Thermal Zone [EXTZ] (0 C)
[    7.350456] ACPI: Thermal Zone [LOCZ] (0 C)
[    7.355899] ACPI: Invalid passive threshold
[    7.365606] ACPI: Thermal Zone [BATZ] (22 C)
[    7.366029] ACPI: Thermal Zone [PCHZ] (59 C)
```
 My colleague will answer directly.

mezmerize, 2016-05-30:
Here is dump from windows, bios version is Hewlett-Packard L70 Ver. 01.08, 10.4.2014
[acpidump.txt](https://github.com/Bumblebee-Project/bbswitch/files/289490/acpidump.txt)

rkrell, 2016-05-30:
I have also made some acpidump for a different issue, see https://bugs.freedesktop.org/show_bug.cgi?id=90626, especially the attachment https://bugs.freedesktop.org/attachment.cgi?id=119334.

Lekensteyn, 2016-05-30:
@mezmerize Unfortunately acpidump on Windows seems limited to just four tables including the DSDT, but excluding the SSDT I am interested in. I found another tool that seems able to extract the SSDT, could you give that a go?

http://www.nirsoft.net/utils/firmware_tables_view.html

Select all entries with the SSDT signature, Copy Selected Items (Ctrl+C) and paste it here. If you see an entry with NVIDIA, please select that single entry and Export To Binary File (Ctrl+E).

@rkrell That dump shows the same problem, the `NVOP` method is not implemented (you can also guess that based on the "NVIDIAGF" identifier from `acpixtract -l acpidump.txt`).

mezmerize, 2016-05-31:
Here are the entries from SSDT.
[ssdt.txt](https://github.com/Bumblebee-Project/bbswitch/files/290697/ssdt.txt)

Lekensteyn, 2016-05-31:
@mezmerize That only contains the SataAhci table, could you obtain an overview of all SSDT tables as described before? (ctrl-click to select multiple entries, then Ctrl+C and paste here). Unlike DSDT, there may be multiple SSDTs.

If you happen to see an entry with "NVIDIA", please just export that file.

mezmerize, 2016-05-31:
@Lekensteyn Sorry but I have not found any entry with NVIDIA signature and there is only one SSDT entry as you can see on the image below.
[image]

Lekensteyn, 2016-05-31:
Sorry for the wrong directions, it turns out that this tool is limited to just one table. Apparently the tables can be exported from `HKLM\HARDWARE\ACPI`. Can you give that a go?

mezmerize, 2016-05-31:
There are only DSDT, FACS, FADT and RSDT entries, no SSDT :(

Lekensteyn, 2016-05-31:
Meh, I discovered that those entries only exist on Windows 8.1 (or 8?) and newer. http://rweverything.com/ seems to work though because it directly reads from memory.

mezmerize, 2016-06-01:
I had much more success with the tool you suggested, I have not found any NVIDIA signatures though. I have attached all SSDT signatures that I have found.
[ssdt1.txt](https://github.com/Bumblebee-Project/bbswitch/files/293070/ssdt1.txt)
[ssdt2.txt](https://github.com/Bumblebee-Project/bbswitch/files/293067/ssdt2.txt)
[ssdt3.txt](https://github.com/Bumblebee-Project/bbswitch/files/293069/ssdt3.txt)
[ssdt4.txt](https://github.com/Bumblebee-Project/bbswitch/files/293071/ssdt4.txt)
[ssdt5.txt](https://github.com/Bumblebee-Project/bbswitch/files/293072/ssdt5.txt)
[ssdt6.txt](https://github.com/Bumblebee-Project/bbswitch/files/293068/ssdt6.txt)
[ssdt7.txt](https://github.com/Bumblebee-Project/bbswitch/files/293073/ssdt7.txt)

rkrell, 2016-06-01:
@Lekensteyn BTW: Here is a complete version history for the HP BIOS of the dedicated hardware (HP ZBook 15 G2 (GK208):
http://h20564.www2.hp.com/hpsc/swd/public/detail?swItemId=ob_146940_1#tab-history
We're both on 01.34 at the moment.

Lekensteyn, 2016-06-01:
Wow, magic! @mezmerize Your SSDT 7 (NVSGTBL) contains a non-empty NVOP function.
Is this 01.34 as @rkrell mentioned or 01.08 as you wrote before?

Can you boot from a Linux USB on the same machine and use `acpidump` to obtain the tables? Note that the choice between legacy and UEFI BIOS might matter.

rkrell, 2016-10-04:
Thanks for this analyze and sorry for the late answer.
I'll retry this booting from UEFI soon.

rkrell, 2016-10-25:
Now I converted my OS to boot upon native UEFI with the same hardware (HP ZBook 15 G2).
The OS is openSUSE Tumbleweed 20161023, kernel 4.8.3, x64_86, using the native nvidia driver 370.28 locally compiled.
I'm still getting errors when activating the BIOS setting Hybrid Graphics set to Auto or Enabled.

If you are interested in an analyze please tell me what kind of logs I should post.
I'll try to collect some by my own.

Lekensteyn, 2016-10-26:
Well, if you disable hybrid graphics, the `No suitable _DSM call found` message is totally expected. Have you tried the nouveau driver with either hybrid graphics mode?

rkrell, 2016-10-26:
> Well, if you disable hybrid graphics, the No suitable _DSM call found message is totally expected. 

Sure, I mentioned this just for comparison and mainly to show the installed driver matches the hardware.

> Have you tried the nouveau driver with either hybrid graphics mode?

Not yet, but I have big problems with the nouveau_drm.so itself concerning multi-threading (crashes), which is currently a known issue but prevents me from seriously using it. This is a different issue, see [fdo #98039 ](https://bugs.freedesktop.org/show_bug.cgi?id=98039).

rkrell, 2016-10-26:
According to the /var/log/messages:
```
2016-10-26T00:38:49.982695+02:00 rkrell kernel: [   18.731315] NVRM: The NVIDIA GPU 0000:01:00.0 (PCI ID: 10de:12b9)
2016-10-26T00:38:49.982696+02:00 rkrell kernel: [   18.731315] NVRM: installed in this system is not supported by the 370.28
```
this sounds odd, because PCI ID: 10de:12b9 matches exactly the K610M graphics I'm using. But this problem happens just with activated Hybrid Graphics. Can there be an association with bbswitch?

Lekensteyn, 2016-10-26:
Yes, nouveau has known issues with multithreading, I was just wondering whether it could handle your GPU or not (and apparently it can).

When looking at your dmesg, I do not see "bbswitch: enabling graphics" before the nvidia driver is loaded. This is a problem, you must blacklist nvidia to prevent it from loading automatically.

rkrell, 2016-10-26:
> Yes, nouveau has known issues with multithreading, I was just wondering whether it could handle your GPU or not (and apparently it can).

Last time I still tried nouveau in legacy BIOS mode, so this is not representative by now, because bbwitch did not have the complete ACPI data like discussed above. I'll try this later and let you know, whether bbswitch works in UEFI mode with nouveau.

rkrell, 2016-11-03:
With the current hardware using native UEFI boot I'm currently not able to boot on Nouveau in neither combination of the BIOS (Hybrid Graphics Enabled|Disabled|Auto) and blacklisting/remove nouveau from the blacklist in /etc/modprobe.d (after mkinitrd), even not if I had bumblebee/primus completely uninstalled. The best result I got was a black screen which couldn't been interrupted by CTRL-ALT-BKSP.

Unfortunately, I can currently just work or try on the native nvidia driver with uninstalled bumblebee/primus, blacklisted nouveau and Hybrid Graphics set do Disabled.

If you need me something to test please give me an exact checklist what to do in particular (a list of commands) and which files to log (and how to process the logs). There are too much combinations to deliver information for.