Bumblebee-Project/bbswitch #119: 1.0 Milestone

ArchangeGabriel, 2015-12-27:
For the next release, that would probably deserve a 1.0 version number, I see three things that should be worked on:

* Issues regarding advertising of `Windows 2013` OSI since kernel 3.15 or 3.16. For some laptops (Thinkpad only?), this change the behaviour of the “`_DSM`” function, and results in card not turned OFF (so no power savings) and sometimes other errors. Also, this revealed that we might also need to handle the pcie link powerstate.

* Issues regarding suspend. They are several of them, with different problems. Most common ones are bbswitch making suspend unable to work, issues because of card being powered off or on on suspend (both seems problematic, which is potentially a big issue) —including card apparently ON during suspend, causing battery drain—, card not usable after suspend (potentially depending of state before suspend).

* Broken handles. Seems to be some more of them, maybe a generic fix could be included (not sure if this is possible, but something that tries the broken handle logic if the initial one failed).

@Lekensteyn Will you have any time soon to look at this?

ArchangeGabriel, 2015-12-27:
Also, the rework needed for the first point might be the right time to look at #18.

Lekensteyn, 2016-01-21:
Regarding changed behavior after advertisement of Windows 2013, this would mean that newer Windows versions cannot run on those machines either, can they? Maybe there is a BIOS update available or the approach is wrong.

The current suspend logic in bbswitch aims to be as compatible as possible by powering it on before suspend. Unfortunately this possibly has adverse effects on the battery (though I have not verified this yet).

The broken handles problem will probably remaing unfixed, it is a problem that needs to be addressed in the ACPI core of Linux.

BernardoGO, 2016-03-26:
I have modified some parts of the code to make suspend reliable for P650SE(had battery drain problems while suspended).
It is more like a nasty workaround but it works and since my hibernation is unreliable also I had to make something about it.
Also I cannot lose my GTX970m after suspend since I use it mostly for image processing on linux and the Intel is a key factor for me to have this laptop to last almost 8 hours.
I resolved this issues and so far I have 5 days of uptime and a battery waste of 2% per hour while suspended which is pretty nice considering that it was wasting more than 40% per hour. Now I'm trying to make it reliable while suspended in D3Cold, right now i'm using D3hot. Maybe I can get 1% per hour with it.

Anyway, as I've said, it is a workaround. To solve it in a elegant way we need to patch the kernel I guess.

ArchangeGabriel, 2016-05-05:
As pointed in #112 after a more thorough reading of it, there is at least a probable issue on PCIe port PM that should be handled somewhere in bbswitch or kernel, especially if it’s able to save around 2 W for every concerned laptop.

bluca, 2017-01-05:
@Lekensteyn in case a release can't be cut before the 25th of January, is there any patch that you think would be useful to include in the next Debian stable release? pm-rework etc. Consider that Debian 9 will ship with kernel 4.9
I'm happy to carry any needed change as quilt patches if needed

Lekensteyn, 2017-01-07:
@bluca You can try the patches on the pm-rework branch until 5c7b3f53f229c70bc49c710295967605ac5846e4 (that is, w/o the audio devices patch). Only problem is that it is not much tested and I have no time to work on it in this month (January), sorry.

(In particular, I do not think that this new version of bbswitch has been tested at all on the old models using DSM instead of PR3.)

bluca, 2017-01-08:
I have an old laptop (Latitude E5540 with 720m) and unfortunately testing doesn't look good. I see the problem already reported in another issue with the card refusing to change state from D3 and the nvidia kernel module failing to load (PM runtime for the card is reported as "bad" by powertop, which I think means off?).

So I'll hold out the patches from Stretch to avoid regressions I think, and if I'm reading correctly in other issues on newer laptops 0.8 will still work fine with pm_runtime off, so there's a way to support both old and new models as far as I can see, is that correct?

ArchangeGabriel, 2017-01-08:
Yes, you’re almost right, because most laptops supporting the newer interface also support the older one. However, it’s not the case for some, e.g. https://github.com/Bumblebee-Project/bbswitch/issues/112.

@Lekensteyn Whenever that’s needed, I’ve still have my old Asus laptop accessible, and being from 2009 or 10, it definitively represents the old category. ;)

Lekensteyn, 2017-01-09:
@bluca Can you link the specific issue you found?

Some (most?) laptops should indeed still work with `pcie_port_pm=off`. (Disabling RPM completely just results in sucking more power.) Some laptops have issues, like memory corruption (one Lenovo model, #78), fans that keep spinning, higher power consumption during suspend (mentioned by @ArchangeGabriel above).