Bumblebee-Project/bbswitch #120: Power saving is disabled or the graphic card is turned on on its own.

sudhirkhanger, 2015-12-31:
Hi,

For past few days I have noticed that the graphic card is turned on its own. Appending `OFF` to `/proc/acpi/bbswitch` doesn't turn it off. As far as I can `bumblebeed` and `bumblebee-nvidia` system files have no failure. Restarting `bumblebeed` or rebooting the system fixes the problem.

**Bumblebeed systemd file**

    systemctl status bumblebeed
    ● bumblebeed.service - Bumblebee C Daemon
    Loaded: loaded (/usr/lib/systemd/system/bumblebeed.service; enabled; vendor preset: disabled)
    Active: active (running) since Thu 2015-12-31 11:52:42 IST; 3h 7min ago
    Main PID: 873 (bumblebeed)
       CGroup: /system.slice/bumblebeed.service
               └─873 /usr/sbin/bumblebeed

    Dec 31 11:52:42 fedora systemd[1]: Started Bumblebee C Daemon.
    Dec 31 11:52:42 fedora systemd[1]: Starting Bumblebee C Daemon...
    Dec 31 11:52:50 fedora bumblebeed[873]: [   25.483712] [INFO]/usr/sbin/bumblebeed 3.2.1 started

Journalctl -b It's too long to be able to paste https://cloud.wyffy.com/index.php/s/wTEhYPA6lS4WfD3

Sorry the log is huge. I will try to figure out what's bombarding it.

[sudhir@fedora Downloads]$ journalctl -b | grep nvidia
Dec 31 11:52:50 fedora kernel: nvidia: module license 'NVIDIA' taints kernel.
Dec 31 11:52:51 fedora kernel: nvidia 0000:01:00.0: enabling device (0000 -> 0003)
Dec 31 11:52:51 fedora kernel: [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:01:00.0 on minor 1
Dec 31 11:52:56 fedora audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=bumblebee-nvidia comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[sudhir@fedora Downloads]$ journalctl -b | grep bumblebee
Dec 31 11:52:42 fedora audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=bumblebeed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Dec 31 11:52:50 fedora bumblebeed[873]: [   25.483712] [INFO]/usr/sbin/bumblebeed 3.2.1 started
Dec 31 11:52:56 fedora audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=bumblebee-nvidia comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Dec 31 14:47:16 fedora dnf[25726]: bumblebee: will expire after 160665 seconds.
Dec 31 14:47:16 fedora dnf[25726]: bumblebee-nonfree: will expire after 160662 seconds.
Dec 31 14:47:19 fedora dnf[25726]: repo: using cache for: bumblebee
Dec 31 14:47:19 fedora dnf[25726]: not found deltainfo for: bumblebee for fedora Linux 23 - x86_64 - Base
Dec 31 14:47:19 fedora dnf[25726]: not found updateinfo for: bumblebee for fedora Linux 23 - x86_64 - Base
Dec 31 14:47:26 fedora dnf[25726]: repo: using cache for: bumblebee-nonfree
Dec 31 14:47:26 fedora dnf[25726]: not found deltainfo for: bumblebee nonfree for fedora Linux 23 - x86_64 - Base
Dec 31 14:47:26 fedora dnf[25726]: not found updateinfo for: bumblebee nonfree for fedora Linux 23 - x86_64 - Base
Dec 31 15:00:42 fedora sudo[27914]:   sudhir : TTY=pts/1 ; PWD=/home/sudhir/Downloads ; USER=root ; COMMAND=/bin/systemctl status bumblebeed
Dec 31 15:00:48 fedora sudo[27936]:   sudhir : TTY=pts/1 ; PWD=/home/sudhir/Downloads ; USER=root ; COMMAND=/bin/systemctl restart bumblebeed
Dec 31 15:00:48 fedora bumblebeed[873]: [ 8614.665182] [WARN]Received Terminated signal.
Dec 31 15:00:48 fedora audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=bumblebeed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Dec 31 15:00:48 fedora audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=bumblebeed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Dec 31 15:00:48 fedora bumblebeed[27947]: [ 8614.691860] [INFO]/usr/sbin/bumblebeed 3.2.1 started
[sudhir@fedora Downloads]$ journalctl -b | grep acpi
Dec 31 11:52:26 fedora kernel: ACPI: LAPIC_NMI (acpi_id[0x00] high edge lint[0x1])
Dec 31 11:52:26 fedora kernel: ACPI: LAPIC_NMI (acpi_id[0x01] high edge lint[0x1])
Dec 31 11:52:26 fedora kernel: acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
Dec 31 11:52:26 fedora kernel: acpi PNP0C0A:01: ACPI dock station (docks/bays count: 1)
Dec 31 11:52:26 fedora kernel: acpi LNXIOBAY:00: ACPI dock station (docks/bays count: 2)
Dec 31 11:52:26 fedora kernel: acpi PNP0A08:00: _OSC: OS supports [ExtendedConfig ASPM ClockPM Segments MSI]
Dec 31 11:52:26 fedora kernel: acpi PNP0A08:00: _OSC: platform does not support [PCIeCapability]
Dec 31 11:52:26 fedora kernel: acpi PNP0A08:00: _OSC: not requesting control; platform does not support [PCIeCapability]
Dec 31 11:52:26 fedora kernel: acpi PNP0A08:00: _OSC: OS requested [PCIeHotplug PME AER PCIeCapability]
Dec 31 11:52:26 fedora kernel: acpi PNP0A08:00: _OSC: platform willing to grant [PCIeHotplug PME AER]
Dec 31 11:52:26 fedora kernel: acpi PNP0A08:00: _OSC failed (AE_SUPPORT); disabling ASPM
Dec 31 11:52:26 fedora kernel: acpi PNP0A08:00: [Firmware Info]: MMCONFIG for domain 0000 [bus 00-3f] only partially covers this bridge
Dec 31 11:52:26 fedora kernel: clocksource: acpi_pm: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 2085701024 ns
Dec 31 11:52:26 fedora kernel: acpi device:00: registered as cooling_device4
Dec 31 11:52:26 fedora kernel: acpi device:0a: registered as cooling_device5
Dec 31 11:52:36 fedora systemd[1]: Starting Load/Save Screen Backlight Brightness of backlight:acpi_video0...
Dec 31 11:52:37 fedora systemd[1]: Started Load/Save Screen Backlight Brightness of backlight:acpi_video0.
Dec 31 11:52:37 fedora kernel: audit: type=1130 audit(1451542957.105:65): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-backlight@backlight:acpi_video0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Dec 31 11:52:37 fedora audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-backlight@backlight:acpi_video0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Dec 31 11:52:37 fedora systemd[1]: Starting Load/Save Screen Backlight Brightness of backlight:acpi_video1...
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: ThinkPad ACPI Extras v0.25
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: http://ibm-acpi.sf.net/
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: ThinkPad BIOS 83ET76WW (1.46 ), EC unknown
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: Lenovo ThinkPad T420, model 4177CTO
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: detected a 16-level brightness capable ThinkPad
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: radio switch found; radios are enabled
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: This ThinkPad has standard ACPI backlight brightness control, supported by the ACPI video driver
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: Disabling thinkpad-acpi brightness events by default...
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: rfkill switch tpacpi_bluetooth_sw: radio is unblocked
Dec 31 11:52:37 fedora kernel: thinkpad_acpi: Standard ACPI backlight interface available, not loading native one
Dec 31 11:52:37 fedora kernel: input: ThinkPad Extra Buttons as /devices/platform/thinkpad_acpi/input/input11
Dec 31 11:52:37 fedora systemd[1]: Started Load/Save Screen Backlight Brightness of backlight:acpi_video1.
Dec 31 11:52:37 fedora kernel: audit: type=1130 audit(1451542957.691:66): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-backlight@backlight:acpi_video1 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Dec 31 11:52:37 fedora audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-backlight@backlight:acpi_video1 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Dec 31 11:52:44 fedora thinkfan[945]: WARNING: Using default fan control in /proc/acpi/ibm/fan.
Dec 31 11:52:50 fedora kernel: thinkpad_acpi: EC reports that Thermal Table has changed
Dec 31 11:52:51 fedora kernel: thinkpad_acpi: EC reports that Thermal Table has changed
Dec 31 13:35:46 fedora kernel: thinkpad_acpi: EC reports that Thermal Table has changed
Dec 31 14:31:46 fedora kernel: thinkpad_acpi: restoring fan level to 0x07
Dec 31 15:00:48 fedora kernel: thinkpad_acpi: EC reports that Thermal Table has changed

ArchangeGabriel, 2015-12-31:
Well your logs are indeed unusable. Could you rather reboot, give `cat /proc/acpi/bbswitch`, try `optirun glxinfo` and provide output for:
```
journalctl -b -u bumblebeed
journalctl -b -k
```
And `cat /proc/acpi/bbswitch again`?

sudhirkhanger, 2015-12-31:
Do you want these logs when problem occurs or anytime?

**cat /proc/acpi/bbswitch**

    $ cat /proc/acpi/bbswitch
    0000:01:00.0 OFF

[optirun glxinfo](https://gist.github.com/sudhirkhanger/cb1c9896c8fd50e0e8a2)
[journalctl -b -u bumblebeed](https://gist.github.com/sudhirkhanger/b4c2accee74137e1e92b)
[journalctl -b -k](https://gist.github.com/sudhirkhanger/398dbfbb364bde19a940)

**cat /proc/acpi/bbswitch**

    $ cat /proc/acpi/bbswitch
    0000:01:00.0 OFF

ArchangeGabriel, 2015-12-31:
Rather when problem occurs.

sudhirkhanger, 2015-12-31:
Sure will update it again as soon as problem occurs. Thanks.

sudhirkhanger, 2015-12-31:
Here are the logs when problem occurred.

    $ cat /proc/acpi/bbswitch
    0000:01:00.0 ON

[optirun glxinfo](https://gist.github.com/sudhirkhanger/0fe8c450178049cf023f)
[journalctl -b -u bumblebeed](https://gist.github.com/sudhirkhanger/23968afc81b2b2301125)
[journalctl -b -k](https://gist.github.com/sudhirkhanger/df13694d89a8fab04fb4)

     $ cat /proc/acpi/bbswitch
     0000:01:00.0 OFF

ArchangeGabriel, 2016-01-01:
Well, you have an issue with `nvidia` loading right after `bumblebeed` started, and enabling the device by itself:
```
Jan 01 03:10:02 fedora kernel: nvidia 0000:01:00.0: enabling device (0000 -> 0003)
```

I’ve never seen this before, it could possibly be related to bbswitch not really turning the card off because of #112, but the thing is that `nvidia` shouldn’t be loading at all. Could you check whether you correctly have a file blacklisting it in `/etc/modprobe.d`?

sudhirkhanger, 2016-01-02:
$ cat /etc/default/grub | grep -i grub_cmd
    GRUB_CMDLINE_LINUX="nouveau.modeset=0 rd.driver.blacklist=nouveau rhgb quiet resume=UUID=eb832f45-b5e9-4557-b484-12d82401d640

    $ cat blacklist-nvidia.conf 
    blacklist nvidia

    $ cat bumblebee.conf 
    blacklist nvidia
    blacklist nouveau

It looks like that.

ArchangeGabriel, 2016-01-02:
Well, that’s strange. It means something else is loading nvidia, but I can see what. Could you put again a full `journalctl -b` when it occurs?

sudhirkhanger, 2016-01-03:
It seems to occur pretty regularly.

https://gist.github.com/sudhirkhanger/f05b81a7cb3b8c9c8d0b

ArchangeGabriel, 2016-01-03:
That’s odd. We need to figure out what/why nvidia loads, but I have no idea how to do so…

gsgatlin, 2016-01-03:
Would blacklisting in grub like nouveau is help do you think?

ArchangeGabriel, 2016-01-03:
No, it’s already blacklisted by the modprobe file.

Could you try disabling (not uninstalling) bumblebee/bbswitch, reboot and see what happen?

aiguofer, 2016-02-03:
I'm having this issue as well on Fedora 23 on a T530. I checked #112 but the patch seems to be gone, so I couldn't try that.

looking at `journal -b`:
```
$journalctl -b | grep -E "nvidia|bumbleb|bbswitch|primus"
Feb 02 20:09:28 difernan.usersys.redhat.com audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=bumblebeed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Feb 02 20:09:28 difernan.usersys.redhat.com kernel: bbswitch: module verification failed: signature and/or required key missing - tainting kernel
Feb 02 20:09:28 difernan.usersys.redhat.com kernel: bbswitch: version 0.8
Feb 02 20:09:28 difernan.usersys.redhat.com kernel: bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.VID_
Feb 02 20:09:28 difernan.usersys.redhat.com kernel: bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG_.VID_
Feb 02 20:09:28 difernan.usersys.redhat.com kernel: bbswitch: detected an Optimus _DSM function
Feb 02 20:09:28 difernan.usersys.redhat.com kernel: bbswitch: disabling discrete graphics
Feb 02 20:09:28 difernan.usersys.redhat.com kernel: bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is off
Feb 02 20:09:28 difernan.usersys.redhat.com bumblebeed[1155]: [   25.255843] [INFO]/usr/sbin/bumblebeed 3.2.1 started
Feb 02 20:09:28 difernan.usersys.redhat.com kernel: nvidia: module license 'NVIDIA' taints kernel.
Feb 02 20:09:29 difernan.usersys.redhat.com kernel: nvidia 0000:01:00.0: enabling device (0000 -> 0003)
Feb 02 20:09:29 difernan.usersys.redhat.com kernel: [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:01:00.0 on minor 1
Feb 02 20:09:33 difernan.usersys.redhat.com audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=bumblebee-nvidia comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Feb 02 20:11:01 difernan.usersys.redhat.com /usr/libexec/gdm-x-session[2555]: (II) Applying OutputClass "nvidia" to /dev/dri/card1
Feb 02 20:11:01 difernan.usersys.redhat.com /usr/libexec/gdm-x-session[2555]: loading driver: nvidia
Feb 02 20:11:01 difernan.usersys.redhat.com /usr/libexec/gdm-x-session[2555]: (==) Matched nvidia as autoconfigured driver 1
Feb 02 20:11:01 difernan.usersys.redhat.com /usr/libexec/gdm-x-session[2555]: (II) LoadModule: "nvidia"
Feb 02 20:11:01 difernan.usersys.redhat.com /usr/libexec/gdm-x-session[2555]: (WW) Warning, couldn't open module nvidia
Feb 02 20:11:01 difernan.usersys.redhat.com /usr/libexec/gdm-x-session[2555]: (II) UnloadModule: "nvidia"
Feb 02 20:11:01 difernan.usersys.redhat.com /usr/libexec/gdm-x-session[2555]: (II) Unloading nvidia
Feb 02 20:11:01 difernan.usersys.redhat.com /usr/libexec/gdm-x-session[2555]: (EE) Failed to load module "nvidia" (module does not exist, 0)
Feb 02 20:12:33 difernan.usersys.redhat.com sudo[14123]: difernan : TTY=pts/4 ; PWD=/home/rdu/difernan/.config/systemd/user ; USER=root ; COMMAND=/sbin/rmmod nvidia
Feb 02 20:15:22 difernan.usersys.redhat.com bumblebeed[1155]: [  379.461404] [ERROR][XORG] (EE) /dev/dri/card1: failed to set DRM interface version 1.4: Permission denied
Feb 02 20:15:22 difernan.usersys.redhat.com bumblebeed[1155]: [  379.461733] [ERROR][XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied
Feb 02 20:15:22 difernan.usersys.redhat.com bumblebeed[1155]: [  379.461742] [WARN][XORG] (WW) Warning, couldn't open module mouse
Feb 02 20:15:22 difernan.usersys.redhat.com bumblebeed[1155]: [  379.461747] [ERROR][XORG] (EE) Failed to load module "mouse" (module does not exist, 0)
Feb 02 20:15:22 difernan.usersys.redhat.com bumblebeed[1155]: [  379.461758] [WARN][XORG] (WW) NVIDIA(0): Unable to get display device for DPI computation.
Feb 02 20:15:22 difernan.usersys.redhat.com bumblebeed[1155]: [  379.461766] [WARN][XORG] (WW) Warning, couldn't open module mouse
Feb 02 20:15:22 difernan.usersys.redhat.com bumblebeed[1155]: [  379.461773] [ERROR][XORG] (EE) Failed to load module "mouse" (module does not exist, 0)
Feb 02 20:15:22 difernan.usersys.redhat.com bumblebeed[1155]: [  379.461781] [ERROR][XORG] (EE) /dev/dri/card1: failed to set DRM interface version 1.4: Permission denied
Feb 02 20:15:22 difernan.usersys.redhat.com bumblebeed[1155]: [  379.461785] [ERROR][XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied
Feb 02 20:15:23 difernan.usersys.redhat.com bumblebeed[1155]: rmmod: ERROR: Module nvidia is in use
Feb 02 20:15:26 difernan.usersys.redhat.com bumblebeed[1155]: [  382.653931] [ERROR]Unloading nvidia driver timed out.
```

the bottom part was after playing around with `glxinfo`, it couldn't unload `nvidia` though.

what is weird is that it worked the first time I booted into my current kernel:
```
$ uname -r
4.3.3-303.fc23.x86_64
```

aiguofer, 2016-02-03:
Here's the output after disabling `bumblebeed` and `bumblebee-nvidia`:

```
Feb 02 20:51:57 difernan.usersys.redhat.com akmods[1163]: Checking kmods exist for 4.3.3-303.fc23.x86_64[  OK  ]
Feb 02 20:51:58 difernan.usersys.redhat.com kernel: nvidia: module license 'NVIDIA' taints kernel.
Feb 02 20:51:58 difernan.usersys.redhat.com kernel: Disabling lock debugging due to kernel taint
Feb 02 20:51:58 difernan.usersys.redhat.com kernel: nvidia: module verification failed: signature and/or required key missing - tainting kernel
Feb 02 20:51:58 difernan.usersys.redhat.com kernel: nvidia 0000:01:00.0: enabling device (0000 -> 0003)
Feb 02 20:51:58 difernan.usersys.redhat.com kernel: vgaarb: device changed decodes: PCI:0000:01:00.0,olddecodes=io+mem,decodes=none:owns=none
Feb 02 20:51:58 difernan.usersys.redhat.com kernel: nf_conntrack version 0.5.0 (65536 buckets, 262144 max)
Feb 02 20:51:58 difernan.usersys.redhat.com kernel: [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:01:00.0 on minor 1
Feb 02 20:51:58 difernan.usersys.redhat.com kernel: NVRM: loading NVIDIA UNIX x86_64 Kernel Module  352.63  Sat Nov  7 21:25:42 PST 2015
Feb 02 20:51:58 difernan.usersys.redhat.com systemd[1]: Started Builds and install new kmods from akmod packages.
```

I added the akmod lines because it seems interesting that this is happening between them... maybe it's related?

regardless, if I `sudo modprobe bbswitch` after loading with bumblebee disabled, it still can't turn off the card.

aiguofer, 2016-02-04:
So today `intel-virtual-output` stopped working after a suspend session... found this in `journalctl -b`:

```
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: CPU: 0 PID: 13928 Comm: Xorg Tainted: P     U     OE   4.3.4-300.fc23.x86_64 #1
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: Hardware name: LENOVO 24293E8/24293E8, BIOS G4ET98WW (2.58 ) 11/26/2013
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: task: ffff8801ded63c00 ti: ffff8802795d8000 task.ti: ffff8802795d8000
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: RIP: 0010:[<ffffffffa0a5ab4a>]  [<ffffffffa0a5ab4a>] _nv006324rm+0xda/0x150 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: RSP: 0018:ffff8802795db9f0  EFLAGS: 00010246
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: RAX: 00000000ffffffff RBX: 0000000000000000 RCX: 00000000ffffffff
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffffffffa100be40
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: RBP: ffff8803a566ae98 R08: 000001891616ea46 R09: ffff88042bf88900
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: R10: 0000000056b35f7c R11: 0000000000000000 R12: ffff88025cc28008
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: R13: 0000000000000000 R14: ffff880382cfc008 R15: 0000000000000200
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: FS:  00007f0b1b876a00(0000) GS:ffff88043e200000(0000) knlGS:0000000000000000
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: CR2: 00007f0b1b7e5000 CR3: 0000000026638000 CR4: 00000000001406f0
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: Stack:
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  0000000000010000 ffff88025cc28008 ffff880258af4008 ffff880382cfc008
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  ffff8803e0697008 ffffffffa0a54eb8 0000000000000000 ffff880258af4008
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  ffff8801a31e4008 ffff88025cc28008 ffff8800a8c9b008 ffffffffa0a4d413
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: Call Trace:
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffffa0a54eb8>] ? _nv007007rm+0x9f8/0xd90 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffffa0a4d413>] ? _nv007010rm+0x93/0x310 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffffa0ac78c0>] ? _nv002021rm+0x1b20/0x3c90 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffffa0c85f6b>] ? _nv008716rm+0x3b/0x60 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffffa0d194a7>] ? _nv012792rm+0xc57/0xda0 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffffa0d1a425>] ? _nv000689rm+0x355/0x770 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffffa0d1123a>] ? rm_init_adapter+0x6a/0x100 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffffa0d1ffcd>] ? nvidia_open+0x27d/0x8c0 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffffa0d1f0a9>] ? nvidia_frontend_open+0x49/0x90 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff81228396>] ? chrdev_open+0xb6/0x180
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff812216cf>] ? do_dentry_open+0x1ff/0x2f0
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff812282e0>] ? cdev_put+0x30/0x30
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff81222b76>] ? vfs_open+0x56/0x60
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff8123190e>] ? path_openat+0x1de/0x1250
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff81232af4>] ? putname+0x54/0x60
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff81232f91>] ? filename_lookup+0xf1/0x180
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff81233b21>] ? do_filp_open+0x91/0x100
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff812046f3>] ? kmem_cache_alloc+0x193/0x210
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff81232b56>] ? getname_flags+0x56/0x1f0
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff81240aff>] ? __alloc_fd+0x3f/0x100
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff81222f2a>] ? do_sys_open+0x13a/0x230
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff8122303e>] ? SyS_open+0x1e/0x20
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  [<ffffffff8178182e>] ? entry_SYSCALL_64_fastpath+0x12/0x71
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: Code: eb 14 48 89 d8 48 c1 e8 1d 85 c0 75 70 8d 14 dd 00 00 00 00 3b 55 04 76 03 8b 55 04 41 89 56 18 8b 75 08 8d 4c 32 ff 31 d2 89 c8 <f7> f6 41 89 46 40 89 c1 44 89 f8 0f af 45 14 41 0f af c5 41 c1
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: RIP  [<ffffffffa0a5ab4a>] _nv006324rm+0xda/0x150 [nvidia]
Feb 04 09:26:04 difernan.usersys.redhat.com kernel:  RSP <ffff8802795db9f0>
Feb 04 09:26:04 difernan.usersys.redhat.com kernel: ---[ end trace baa93d2cd917dd1b ]---
```

Not sure if it's related, but I thought I'd drop it off here in case it is.

aiguofer, 2016-02-04:
Well... it turns out the `akmod` lines were indeed related...

```
$cat /usr/lib/systemd/system/akmods.service
[Unit]
Description=Builds and install new kmods from akmod packages
Before=display-manager.service

[Service]
Type=oneshot
RemainAfterExit=true
ExecStart=/usr/sbin/akmods --from-init
# Make sure we load the nvidia module if built on startup, otherwise X will
# fail. Harmless if not needed.
ExecStart=-/usr/sbin/modprobe nvidia

[Install]
WantedBy=multi-user.target
```

Comment out the `ExecStart` and now it doesn't get loaded :+1: 

Now I just need to file a bug with someone, just not sure who... rpmfusion-free-updates akmods pkg maintainer?

ArchangeGabriel, 2016-05-05:
OK, so that’s definitively your issue: something (that service in fact) is loading nvidia (which bypass the blacklist, and we can’t work around that on our side, since we still need to be able to load it, so anyone could – it’s just autoloading that should and is disabled).

So, despite what the service file says, it is harmfull if the nvidia card isn’t the one actually running the X server, i.e. on most Optimus systems.

Now you shed some light on this, I think that OP problem is just about the same thing: he has something on his system (maybe the same akmod thing?) that load nvidia. This is undesirable and should be fixed on the side of that things.

Closing here.

closed 2016-05-05 by ArchangeGabriel