Bumblebee-Project/bbswitch #121: bbswitch does not compile on kernel 3.19.0-43

oanonymos0, 2016-01-10:
a user reports that he cannot install bbswitch, because it does not compile on kernel 3.19.0-43
and Ubuntu 14.04. See the link for more info:
http://askubuntu.com/questions/719066/problem-installing-bbswitch-on-ubuntu-14-04-3?noredirect=1#comment1063069_719066

ArchangeGabriel, 2016-01-16:
The bbswitch version is quite old. Should try with 0.8. Available on our PPA.

ArchangeGabriel, 2016-01-16:
Forgot that. Just seen it tries 0.8 too.

oanonymos0, 2016-01-17:
@ArchangeGabriel do you know what could be causing the error?

ArchangeGabriel, 2016-01-18:
Seems to be on GCC side. I don’t think the error is in bbswitch.

Lekensteyn, 2016-01-21:
bbswitch is fine, but your disk needs to be replaced. Please continue at https://bugs.launchpad.net/ubuntu/+source/bbswitch/+bug/1532538

closed 2016-01-21 by Lekensteyn