Bumblebee-Project/bbswitch #123: bbswitch doesn't turn off gpu after optirun

veekk, 2016-02-27:
```
optirun --status
Bumblebee status: Ready (3.2.1). X inactive. Discrete video card is off.

```

```
optirun glxspheres64
Polygons in scene: 62464 (61 spheres * 1024 polys/spheres)
Visual ID of window: 0x20
Context is Direct
OpenGL Renderer: GeForce GT 740M/PCIe/SSE2
94.142424 frames/sec - 82.182570 Mpixels/sec

```

```
optirun glxspheres64
Polygons in scene: 62464 (61 spheres * 1024 polys/spheres)
Visual ID of window: 0x20
Context is Direct
OpenGL Renderer: GeForce GT 740M/PCIe/SSE2
94.142424 frames/sec - 82.182570 Mpixels/sec

```

For disabling i'm wrote script, but bbswtich should doing it, how can i fix that?

```
#!/bin/sh
sudo rmmod nvidia_modeset
sudo rmmod nvidia
sudo tee /proc/acpi/bbswitch <<<OFF

```

imi415, 2016-02-28:
I think bbswitch just use `rmmod nvidia`, which fails due to `nvidia_modset` is still loaded.

veekk, 2016-02-28:
so how can i fix that?

imi415, 2016-02-28:
@veekk  Here is a [solution(AUR)](https://github.com/Bumblebee-Project/Bumblebee/issues/729) for Arch Linux, and similar patches can be found in other issues. Maybe these can help. And it seems that this issue will be fixed in 4.0 (Milestone).

ArchangeGabriel, 2016-05-06:
I’m closing this as a dupe of https://github.com/Bumblebee-Project/Bumblebee/issues/719 (fix is coming).

closed 2016-05-06 by ArchangeGabriel