Bumblebee-Project/bbswitch #125: bbswitch: No suitable _DSM call found. on MacBookPro 9.1

simonmcbrown, 2016-02-29:
Hi.

Looking at https://github.com/Bumblebee-Project/bbswitch/issues/104 and following the diagnostics it looks like I need a device handle.  Might you be able to find what it is please?

Many thanks.

Diagnostics:
Running linux mint 17.3 Cinnamon

mcbrown@simonMBP $ sudo dmidecode -s system-product-name
MacBookPro9,1

mcbrown@simonMBP $  sudo uname -a
Linux simonMBP 3.19.0-32-generic #37~14.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

mcbrown@simonMBP $ lspci |grep VGA
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
01:00.0 VGA compatible controller: NVIDIA Corporation GK107M [GeForce GT 650M Mac Edition] (rev a1)

mcbrown@simonMBP /etc/bumblebee $ inxi -Gx
Graphics:  Card-1: Intel 3rd Gen Core processor Graphics Controller bus-ID: 00:02.0
           Card-2: NVIDIA GK107M [GeForce GT 650M Mac Edition] bus-ID: 01:00.0
           Display Server: X.Org 1.17.1 drivers: nouveau (unloaded: fbdev,vesa) FAILED: intel
           Resolution: 1440x900@59.9hz
           GLX Renderer: Gallium 0.4 on NVE7 GLX Version: 3.0 Mesa 10.5.9 Direct Rendering: Yes

mcbrown@simonMBP /etc/bumblebee $ dpkg -l | grep nouv
ii  libdrm-nouveau2:amd64                       2.4.64-1~ubuntu14.04.1                              amd64        Userspace interface to nouveau-specific kernel DRM services -- runtime
ii  libdrm-nouveau2:i386                        2.4.64-1~ubuntu14.04.1                              i386         Userspace interface to nouveau-specific kernel DRM services -- runtime
ii  xserver-xorg-video-nouveau-lts-vivid        1:1.0.11-1ubuntu2build1~trusty1                     amd64        X.Org X server -- Nouveau display driver

mcbrown@simonMBP /etc/bumblebee $ glxinfo | egrep "vendor"
server glx vendor string: SGI
client glx vendor string: Mesa Project and SGI
OpenGL vendor string: nouveau
mcbrown@simonMBP /etc/bumblebee $ glxinfo | egrep "renderer"
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer,
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer,
OpenGL renderer string: Gallium 0.4 on NVE7

mcbrown@simonMBP /etc/bumblebee $  dmesg | grep acpi
[    0.000000] ACPI: LAPIC (acpi_id[0x01] lapic_id[0x00] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x02] lapic_id[0x02] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x03] lapic_id[0x04] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x04] lapic_id[0x06] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x05] lapic_id[0x01] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x06] lapic_id[0x03] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x07] lapic_id[0x05] enabled)
[    0.000000] ACPI: LAPIC (acpi_id[0x08] lapic_id[0x07] enabled)
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x01] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x02] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x03] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x04] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x05] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x06] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x07] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x08] high edge lint[0x1])
[    0.200725] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    0.218538] acpi PNP0A08:00: _OSC: OS assumes control of [PCIeHotplug SHPCHotplug AER PCIeCapability]
[    0.218722] acpi PNP0A08:00: [Firmware Info]: MMCONFIG for domain 0000 [bus 00-9b] only partially covers this bridge
[    1.236826] acpiphp: Slot [3] registered
[    1.236849] acpiphp: Slot [4] registered
[    1.237580] acpiphp: Slot [1] registered
[    3.349346] acpi device:02: registered as cooling_device8
[    3.351232] acpi device:0f: registered as cooling_device9

mcbrown@simonMBP ~ $ sudo modprobe bbswitch
modprobe: ERROR: could not insert 'bbswitch': No such device

mcbrown@simonMBP ~ $ dmesg | tail
[ 3254.092432] ERROR @wl_inform_single_bss : cfg80211_inform_bss_frame error
[ 3374.050766] ERROR @wl_inform_single_bss : cfg80211_inform_bss_frame error
[ 3494.061182] ERROR @wl_inform_single_bss : cfg80211_inform_bss_frame error
[ 3560.806201] bbswitch: version 0.7
[ 3560.806208] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.IGPU
[ 3560.806212] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.P0P2.GFX0
[ 3560.806224] ACPI Warning: \_SB_.PCI0.P0P2.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[ 3560.806251] ACPI Warning: \_SB_.PCI0.P0P2.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[ 3560.806262] ACPI Warning: \_SB_.PCI0.IGPU._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
[ 3560.806270] bbswitch: No suitable _DSM call found.

gzipped acpi.dump attached
[acpi.dump.gz](https://github.com/Bumblebee-Project/bbswitch/files/151120/acpi.dump.gz)

ArchangeGabriel, 2016-05-06:
I’m not an expert at all in MacBook, but I think that either:
1) You’re dependin on gmux (most probable), which is a tool that does a lot of things for Mac, including bbswitch job (FWIU).
2) You’re depending on something like https://github.com/Bumblebee-Project/bbswitch/tree/hack-lenovo.

I’m marking this as “Broken ACPI handle” and assign to @Lekensteyn that knows those things better than me.

ArchangeGabriel, 2016-05-06:
gmux is the answer I think. See https://github.com/Bumblebee-Project/Bumblebee/issues/585. Closing here. Feel free to reopen if I’m wrong.

closed 2016-05-06 by ArchangeGabriel