Bumblebee-Project/bbswitch #126: support for single gpu machines

karolherbst, 2016-03-09:
I have a mac mini which I want to use as a server and therefore I want to disable the gpu entiry.

```
[  453.577823] bbswitch: version 0.8
[  453.577838] bbswitch: Found discrete VGA device 0000:02:00.0: \_SB_.PCI0.IXVE.IGPU
[  453.577848] ACPI Warning: \_SB_.PCI0.IXVE.IGPU._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150930/nsarguments-95)
[  453.577885] ACPI Warning: \_SB_.PCI0.IXVE.IGPU._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150930/nsarguments-95)
[  453.577900] bbswitch: No suitable _DSM call found.
```

What can I do to get it to work?

ArchangeGabriel, 2016-05-06:
I don’t know, but it might be a problem of ACPI handle. I’m labelling this that way and assigning to @Lekensteyn that knows those things way better than me.

karolherbst, 2016-05-06:
@ArchangeGabriel I think we already dig into this and found out the ACPI handles are useless. It is an interessting feature nethertheless so I decided to leave it open.

Lekensteyn, 2016-05-11:
I think that I and @karolherbst already checked the ACPI tables, but there was nothing interesting last time. Maybe you could link/attach the acpidump nevertheless