Bumblebee-Project/bbswitch #128: NVidia card cannot be turned off on Lenovo Yoga 700

janusko, 2016-03-29:
I am trying to use bbswitch on my new laptop. It seems to be impossible to turn off my NVidia card using this module.
```
janikl janik # echo OFF > /proc/acpi/bbswitch
janikl janik # cat /proc/acpi/bbswitch
0000:02:00.0 ON
```
dmesg does not contain any relevant info about this problem. The only record about turning off NVidia is:
```
[ 4681.168621] bbswitch: disabling discrete graphics
[ 4681.168658] ACPI Warning: \_SB_.PCI0.RP09.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150930/nsarguments-95)
```
I have found that there is a way to turn off NVidia directly using ACPI method _OFF
```
janikl janik # echo "\_SB.PCI0.RP09.PEGP._OFF" >> /proc/acpi/call
janikl janik # cat /proc/acpi/bbswitch
0000:02:00.0 OFF
janikl janik # echo "\_SB.PCI0.RP09.PEGP._ON" >> /proc/acpi/call
janikl janik # cat /proc/acpi/bbswitch
0000:02:00.0 ON
```
I want to let bumblebee to manage turning the NVidia card on and off, so I need to have bbswitch module functional.

janusko, 2016-04-25:
I have made patch for this issue. I know, this is not the cleanest solution, but the _DSM method is really not working for me. The patch adds possibility to use _ON/_OFF methods directly instead. Can someone review the patch, whether it could be applied to the project or not?

[onoff.zip](https://github.com/Bumblebee-Project/bbswitch/files/234703/onoff.zip)

ArchangeGabriel, 2016-05-06:
It might be #112.

Lekensteyn, 2016-05-11:
@janusko Can you attach your `sudo acpidump > acpidump.txt` (unzipped please)?

I think that calling `_ON` on the DGPU handle is too specific for your device. Looks indeed like #112 (and others that rely on DSM instead of the parent device (PCI port))

shmerl, 2016-06-07:
I have a similar issue (Lenovo W540, Debian testing, Linux 4.5.0-2-amd64),

Running: `sudo tee /proc/acpi/bbswitch <<<OFF`

Doesn't turn the discrete Nvidia GPU off. dmesg shows this:

`ACPI Warning: \_SB.PCI0.PEG.VID._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160108/nsarguments-95)`

ArchangeGabriel, 2016-06-10:
The dmesg output is normal. bbswitch reporting of status might not be accurate, but @Lekensteyn might be able to tell you more about this if you attach the `acpidump` like he asked for @janusko. ;)

Lekensteyn, 2016-06-10:
The reported status is most likely correct, the card is indeed not off. Newer laptops require the power resources on the PCIe port to be disabled to make the graphics card really enter D3cold. I have tagged issues caused by this with "ACPI Power Resources". Will be able to look into again this in two weeks.