Bumblebee-Project/bbswitch #129: Eurocom M5 Pro draining 40% of battery per hour while suspended to ram

JeorgeMatt, 2016-03-31:
I recently decided to use Ubuntu as my main OS in Eurocom M5 Pro, but I have noticed something strange with bbswitch.
After installing it is draining a way too much battery while suspended to run. In windows or without bbswitch it uses about 3% per hour.
Everything is working perfectly but the system is impossible to use as a laptop since the suspend to ram  does not last very long and the hibernation is unreliable.

Is there any fix for this?

oanonymos0, 2016-04-04:
having an open OpenGL program on nvidia before suspend may help.

Lekensteyn, 2016-04-06:
I think this is a duplicate of #115, if not please let me know.

closed 2016-04-06 by Lekensteyn

JeorgeMatt, 2016-04-06:
It is a different laptop and i'm not sure if it is the same problem