Bumblebee-Project/bbswitch #131: MSI GE62 0% battery after 2 hours in standby

hundredyearslate, 2016-04-12:
I'm quite used to put my computer in standby and it used to work flawlessly.
But now I installed nvidia with bbswitch and everything I have ever feared come true. It does not last two hours in standby.
Is there any alternative to bbswich for my computer?

Lekensteyn, 2016-04-25:
On P651RA/P671RA laptops (GTX965M in both cases), we had to disable nouveau/bbswitch because of infinite loops in ACPI methods. So unless someone figures out how to do it properly, no alternative I guess.

ArchangeGabriel, 2016-05-06:
Could be related to PCIe root port thing in #112 maybe?

hundredyearslate, 2016-05-06:
@ArchangeGabriel 
I don't know. It seems to be turning it off while it is on. Not sure if it is entirely off but it is definitely saving energy.
The problem only happens when suspended.