Bumblebee-Project/bbswitch #132: Dedicated GPU stays on during suspend on Acer E5-574G

jacobmischka, 2016-05-23:
As requested in #115 I am creating a new issue.

When suspending, bbswitch powers on the dedicated GPU as it is intended to, but the card continues to draw power during suspend, draining the battery very rapidly and producing heat. The card is a GTX 940M.

As a workaround, I'm using systemd hooks to rmmod and modprobe bbswitch on suspend and resume. This causes suspend to work properly and not drain battery, but means I have to reboot in order for the GPU to work again. I am currently using the stable version of bumblebee, I am setting no acpi_osi parameter, and am running Linux kernel 4.6.

If there is any other information I can provide I will be happy to do so.

acpidump: https://gist.github.com/jacobmischka/e7f3d1da3039a0f207635d235c10f5a4

Lekensteyn, 2016-05-27:
Updating the BIOS will likely not make a difference. What will likely work is the `bbswitch_dev` module from the acpi-pr3 branch combined with the four patches from https://git.kernel.org/cgit/linux/kernel/git/helgaas/pci.git/commit/?h=pci/pm&id=8b71f5652eeac561acf883da01ab4810f763ee42 applied on top of your current kernel (I tested it on top of v4.6).

`bbswitch_dev` (with dev standing for "PCI device", but "in development" is also a possible explanation) will eventually be merged into bbswitch itself. For now you have to activate it manually:

    make modname=bbswitch_dev
    sudo insmod bbswitch_dev
    # replace 13d9 by your device ID from lspci -nnd10de:
    sudo tee /sys/bus/pci/drivers/bbswitch_dev/new_id <<<"10de 13d9"

Once loaded it should show:

    bbswitch_dev: disabling discrete graphics
    bbswitch_dev 0000:01:00.0: power state changed by ACPI to D3cold
    pcieport 0000:00:01.0: PME# enabled
    pcieport 0000:00:01.0: power state changed by ACPI to D3cold

If the last two messages are missing, then either the four pci/pm patches were not applied to the kernel or (not in your case) the BIOS is older than 2015.

jacobmischka, 2016-06-12:
Thanks, I've been meaning to try this but I haven't found the time lately to patch and compile a kernel.

I just tried that kernel.org link and it seems broken, the commit hash no longer seems to exist. Sorry again for the delay, thank you for your help.

Lekensteyn, 2016-06-13:
New branch is based on v4.7-rc1, see
https://git.kernel.org/cgit/linux/kernel/git/helgaas/pci.git/log/?h=pci/pm

jacobmischka, 2016-06-13:
Oh perfect that makes things much easier, I'll install it now and report back soon

Lekensteyn, 2016-07-05:
Have you found time to test nouveau? I'd like to prepare the second revision of the patchset for nouveau, see https://github.com/Bumblebee-Project/bbswitch/issues/78#issuecomment-223072012 for something based on v4.6.

If possible, can you try nouveau on v4.6 (with no patches) and nouveau (with the linked patchset)?

jacobmischka, 2016-07-10:
Sure, I'll also try the patch mentioned before with nvidia today as well finally.

Nouveau without the patches behaved as expected, high power usage. Compiling the patches now, will report back.

jacobmischka, 2016-07-10:
I'm sorry, I was wrong. Nouveau without the patches doesn't work as expected. I've never been able to get xrandr to recognize my nvidia gpu to use with PRIME, and I'm still unable to.

I assumed it was working correctly because the power usage went up, but that's just because I hadn't reenabled power management for the PCI device yet since disabling bbswitch. I have no xorg.d configs messing with my graphics config to my knowledge, xrandr just can't see it. So I can't confirm that the card actually *works*, but nouveau is loaded and power usage is low.

I'm not sure if that means it's working or not. If you have any further ideas to try to get xrandr to recognize my card I'll be happy to try them.

I'll try bbswitch_dev now. I'm gonna use the same kernel I just compiled to test nouveau, hopefully the 4 nouveau-specific commits don't affect it.

jacobmischka, 2016-07-10:
Sorry, I seem to be having problems enabling bbswitch_dev. It seems to be installed correctly, but `/sys/bus/pci/drivers/bbswitch_dev/` doesn't exist, and bumblebee is complaining that it can't find `bbswitch`. I tried changing `PMMethod=bbswitch` to `bbswitch_dev`, but that didn't seem to work.

Any ideas?

jacobmischka, 2016-07-16:
Okay, I'm not sure what I was doing wrong before but I'm able to correctly load the module now. However, I do not see the two pcieport messages.

`journalctl -r`:

```
Jul 16 01:33:20 mischka-laptop kernel: bbswitch_dev 0000:01:00.0: Refused to change power state, currently in D0
Jul 16 01:33:20 mischka-laptop kernel: bbswitch_dev: disabling discrete graphics
```

I'm pretty certain I have the patch properly applied. Is there a way to check that you know of?

Lekensteyn, 2016-07-17:
Can you post the output of `lspci -nn -s01:`? If you happen to have an audio device there that is not suspended, the pcie root port will also not suspend. It is also important to note that this module will only save power when the appropriate pci/pm patches are applied (possibly in Linux 4.8 now).

Also, do you need to use Bumblebee for acceleration or would you just like to save power?

jacobmischka, 2016-08-23:
Sorry for the delay. My output of `lspci -nn s01:` is:
```
01:00.0 3D controller [0302]: NVIDIA Corporation GM108M [GeForce 940M] [10de:1347] (rev a2)
```

I almost never use my external gpu, saving power is my primary concern, though I'd like to be able to use the gpu occasionally.

Unfortunately, it looks like 4.8 rc2 isn't automatically disabling the card either. I would prefer to use Nouveau if it'll correctly suspend the card, is there something special I'm supposed to do to get it to work?

jacobmischka, 2016-08-23:
Oh. Do you not need to set up the provider offload sink in xrandr for PRIME with DRI3?

After reading either [badly written docs—or docs that suggest that—](https://nouveau.freedesktop.org/wiki/Optimus/#dri3), I just tried using PRIME and it seems to work. I get terrible performance with nouveau, but I suppose that's more or less to be expected. Looks like everything is working now, thanks again.

Lekensteyn, 2016-08-24:
DRI3 does not require the xrandr --setprovideroffloadsink stuff. Thanks for testing, no problem for the delay, I got busy too ;)

BernardoGO, 2016-10-01:
Seems like 4.8rc8 successfully suspend the GPU using the stock bbswitch.
Sometimes the GPU does not work after resuming but it does work mostly.

Hopefully it may be solved in future releases.