Bumblebee-Project/bbswitch #135: bumblebee + bbswitch Not work module bbswitch (No suitable _DSM call found.)

xlin, 2016-07-12:
`dmesg | grep -C 10 bbswitch:`

http://pastebin.com/raw/JU73a6Jj

`[root@xlin ~]# uname -a
Linux xlin.local 4.7.0-desktop-0.rc7.1.mga6 #1 SMP Mon Jul 11 10:34:16 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

`Distributor ID: Mageia
Description:    Mageia 6
Release:        6
Codename:       thornicroft
`

Xorg and driver
http://pastebin.com/raw/smYCPApP

`sh get-acpi-info.sh`
http://pastebin.com/raw/v6nPGZHt
[ASUSTeK_COMPUTER_INC.-X751LD.tar.gz](https://github.com/Bumblebee-Project/bbswitch/files/359513/ASUSTeK_COMPUTER_INC.-X751LD.tar.gz)

acpi_dump_info

http://pastebin.com/raw/riEf0saB

xlin, 2016-07-13:
Please help me

xlin, 2016-07-13:
The new kernel 4.7.0-server-0.rc7.2.mga6 turned to run. But Nvidia module is unloaded after the closure of the program.

logfile http://pastebin.com/raw/xsqhFZuh

xlin, 2016-07-16:
reinstall dkms-bbswitch, but not nvidia module unload

[xlin@xlin ~]$ dmesg | grep -C 10 bbswitch
[   42.601362] audit: type=1130 audit(1468672952.510:38): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=alsa-state comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   42.625633] audit: type=1130 audit(1468672952.530:39): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=hddtemp comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   45.368346] audit: type=1130 audit(1468672955.269:40): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bluetooth comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   45.368385] audit: type=1130 audit(1468672955.269:41): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=dbus comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   45.442190] audit: type=1130 audit(1468672955.349:42): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=irqbalance comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   45.478480] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   45.478483] Bluetooth: BNEP filters: protocol multicast
[   45.478487] Bluetooth: BNEP socket layer initialized
[   45.497631] audit: type=1130 audit(1468672955.399:43): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bumblebeed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   45.499408] audit: type=1130 audit(1468672955.399:44): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=cpupower comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   46.377133] bbswitch: version 0.8
[   46.377141] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   46.377150] bbswitch: Found discrete VGA device 0000:04:00.0: \_SB_.PCI0.RP05.PEGP
[   46.377162] ACPI Warning: \_SB.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[   46.377281] bbswitch: detected an Optimus _DSM function
[   46.377361] bbswitch: disabling discrete graphics
[   46.377368] ACPI Warning: \_SB.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[   46.409661] bbswitch: Succesfully loaded. Discrete card 0000:04:00.0 is off
[   47.577765] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   47.590165] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   47.593236] IPv6: ADDRCONF(NETDEV_UP): enp2s0f1: link is not ready
[   47.681528] r8169 0000:02:00.1 enp2s0f1: link down
[   47.681600] IPv6: ADDRCONF(NETDEV_UP): enp2s0f1: link is not ready
[   48.243188] audit_printk_skb: 51 callbacks suppressed
[   48.243191] audit: type=1130 audit(1468672958.149:62): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=wpa_supplicant comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   48.422590] audit: type=1130 audit(1468672958.329:63): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=polkit comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   48.553390] NET: Registered protocol family 17
[   48.740767] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
--
[  360.267100] audit: type=1006 audit(1468673270.601:142): pid=4821 uid=0 old-auid=4294967295 auid=1000 tty=(none) old-ses=4294967295 ses=2 res=1
[  360.267438] audit: type=1105 audit(1468673270.601:143): pid=4821 uid=0 auid=1000 ses=2 msg='op=PAM:session_open grantors=pam_loginuid,pam_keyinit,pam_limits,pam_systemd,pam_tcb acct="xlin" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  360.327945] audit: type=1130 audit(1468673270.661:144): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=user@1000 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  360.328432] audit: type=1006 audit(1468673270.661:145): pid=4820 uid=0 old-auid=4294967295 auid=1000 tty=(none) old-ses=4294967295 ses=3 res=1
[  360.342085] audit: type=1105 audit(1468673270.671:146): pid=4820 uid=0 auid=1000 ses=3 msg='op=PAM:session_open grantors=pam_keyinit,pam_namespace,pam_keyinit,pam_limits,pam_systemd,pam_tcb,pam_loginuid acct="xlin" exe="/usr/libexec/sddm-helper" hostname=? addr=? terminal=:0 res=success'
[  363.035443] audit: type=1130 audit(1468673273.371:147): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=rtkit-daemon comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  364.452976] Bluetooth: RFCOMM TTY layer initialized
[  364.452984] Bluetooth: RFCOMM socket layer initialized
[  364.452992] Bluetooth: RFCOMM ver 1.11
[  370.452609] audit: type=1130 audit(1468673280.781:148): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=udisks2 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  439.897758] bbswitch: enabling discrete graphics
[  440.815560] nvidia: module license 'NVIDIA' taints kernel.
[  440.815564] Disabling lock debugging due to kernel taint
[  440.828246] nvidia-nvlink: Nvlink Core is being initialized, major device number 244
[  440.828264] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  367.27  Thu Jun  9 18:53:27 PDT 2016
[  441.600245] vgaarb: this pci device is not a vga device
[... cut ...]
[  441.610808] ACPI Warning: \_SB.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[  441.610846] ACPI Warning: \_SB.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[  441.610881] ACPI Warning: \_SB.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[  441.610914] ACPI Warning: \_SB.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)

Lekensteyn, 2016-07-16:
Not sure why PSPX got detected in your first post, but bbswitch seems to work without any changes I guess. Closing.

If you have an issue with Bumblebee, try that https://github.com/Bumblebee-Project/Bumblebee. Please try Bumblebee from the develop branch though (really need to get that tagged...), that version is supposed to work better with newer nvidia drivers that have multiple kernel modules.

closed 2016-07-16 by Lekensteyn

akien-mga, 2016-07-16:
> Please try Bumblebee from the develop branch though (really need to get that tagged...), that version is supposed to work better with newer nvidia drivers that have multiple kernel modules.

Oh nice, thanks for the headsup. I package bumblebee for Mageia (which @xlin is using, I sent him here) but was still only on 524ded14, I hadn't noticed the more recent developments. I'll update our package.

xlin, 2016-07-16:
Thank you so much. After the upgrade, everything was working as it should.

[xlin@xlin ~]$ sudo systemctl status bumblebeed.service 
● bumblebeed.service - Bumblebee C Daemon
   Loaded: loaded (/usr/lib/systemd/system/bumblebeed.service; enabled; vendor preset: enabled)
   Active: active (running) since Вс 2016-07-17 01:21:22 MSK; 2min 32s ago
 Main PID: 2420 (bumblebeed)
   CGroup: /system.slice/bumblebeed.service
           └─2420 /usr/sbin/bumblebeed --use-syslog

июл 17 01:21:22 xlin.local systemd[1]: Started Bumblebee C Daemon.
июл 17 01:21:24 xlin.local bumblebeed[2420]: /usr/sbin/bumblebeed 3.2.1-2016-05-19-Format:%h$ started
[xlin@xlin ~]$ sudo lsmod | grep nvidia
[xlin@xlin ~]$ sudo cat /proc/acpi/bbswitch 
0000:04:00.0 OFF
[xlin@xlin ~]$ optirun kwrite
[xlin@xlin ~]$ cat /proc/acpi/bbswitch 
0000:04:00.0 ON

clouse kwrite and nvidia module unload

[xlin@xlin ~]$ sudo lsmod | grep nvidia
[xlin@xlin ~]$ sudo cat /proc/acpi/bbswitch 
0000:04:00.0 OFF