Bumblebee-Project/bbswitch #137: X freezes when bumblebeed has started and switched off the card using bbswitch

MrBollie, 2016-08-17:
I'm not sure, if X is trying to add the card nonetheless and then fails. What helps is 
- tee /proc/acpi/bbswitch <<< ON 
- starting X
- tee /proc/acpi/bbswitch <<< OFF
and then let bumblebeed switch in on, when it needs it.

Versions:
- bbswitch 0.8-52
- bumblebee 3.2.1.r18.3c6f9c9-1
- nvidia 370.23-1

OS is arch, NVIDIA card is a GeForce GTX 960M

sodar, 2016-08-25:
I have the same issue as @MrBollie. With bbswitch installed Xorg server cannot start. Laptop hangs on attempt to start Xorg (I have tried it using `startx` with `twm` and with lightdm) and does not respond to anything. In this state I cannot get any output from dmesg.

@MrBollie workaround works, but after starting proper desktop environment and using `nvidia-smi` Laptop hangs.

My spec:

- HW:
    - Dell Inspiron 7559
    - Intel i7-6700HQ
    - NVIDIA GeForce GTX 960M
- OS: Arch Linux, kernel 4.7.1
- bbswitch 0.8-52
- bumblebee 3.2.1-12
- nvidia 370.23-2
- xorg-server 1.18.4-1

sodar, 2016-08-25:
My dmesg output after boot without Xorg server and when bbswitch is installed:

```
[    4.517649] Bluetooth: hci0: read Intel version: 370810011003110e25
[    4.517655] Bluetooth: hci0: Intel device is already patched. patch num: 25
[    4.520682] uvcvideo: Found UVC 1.00 device Integrated_Webcam_HD (1bcf:2b8a)
[    4.529287] uvcvideo 1-4:1.0: Entity type for entity Extension 4 was not initialized!
[    4.529292] uvcvideo 1-4:1.0: Entity type for entity Extension 3 was not initialized!
[    4.529296] uvcvideo 1-4:1.0: Entity type for entity Processing 2 was not initialized!
[    4.529299] uvcvideo 1-4:1.0: Entity type for entity Camera 1 was not initialized!
[    4.529591] input: Integrated_Webcam_HD as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/input/input20
[    4.529827] usbcore: registered new interface driver uvcvideo
[    4.529833] USB Video Class driver (1.1.1)
[    4.563535] bbswitch: version 0.8
[    4.563548] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[    4.563558] bbswitch: Found discrete VGA device 0000:02:00.0: \_SB_.PCI0.PEG0.PEGP
[    4.563575] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[    4.563753] bbswitch: detected an Optimus _DSM function
[    4.563772] pci 0000:02:00.0: enabling device (0006 -> 0007)
[    4.563892] bbswitch: Succesfully loaded. Discrete card 0000:02:00.0 is on
[    4.566804] bbswitch: disabling discrete graphics
[    4.566822] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[    5.005613] IPv6: ADDRCONF(NETDEV_UP): wlp5s0: link is not ready
[    5.008105] iwlwifi 0000:05:00.0: L1 Enabled - LTR Enabled
[    5.008549] iwlwifi 0000:05:00.0: L1 Enabled - LTR Enabled
[    5.075855] iwlwifi 0000:05:00.0: L1 Enabled - LTR Enabled
[    5.076570] iwlwifi 0000:05:00.0: L1 Enabled - LTR Enabled
[    5.099640] IPv6: ADDRCONF(NETDEV_UP): wlp5s0: link is not ready
[    5.110730] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
[    5.124986] r8169 0000:04:00.0 enp4s0: link down
[    5.125075] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
```

Lekensteyn, 2016-08-26:
@MrBollie @sodar Please upload your machine details following the instructions at https://bugs.launchpad.net/lpbugreporter/+bug/752542

sodar, 2016-08-26:
Sorry. I have already uploaded it according to readme on GitHub, but did not paste any reference here. Below is a link to launchpad.
https://bugs.launchpad.net/lpbugreporter/+bug/752542/comments/792

Lekensteyn, 2016-08-26:
@sodar Can you boot with `acpi_osi="!Windows 2015"` as workaround and report whether it prevents the lockup?
(Your PGON method only falls back to the old method when Linux reports itself as either `Windows 2009` or `Windows 2013`, disabling `Windows 2015` is sufficient to make it report itself as `Windows 2013`.)

See also the discussion in https://github.com/Bumblebee-Project/Bumblebee/issues/764#issuecomment-234494238

sodar, 2016-08-30:
Thank you @Lekensteyn. Adding `acpi_osi="!Windows 2015"` to the kernel command line worked. Xorg server and desktop environment now starts with graphics card off (according to `cat /proc/acpi/bbswitch`).

However after running `optirun glxgears -info` Bumblebee didn't turn off the graphics card. Manual attempt to disable it, does not work either. dmesg is showing that `nvidia` and `nvidia_modeset` modules are in use.

Here are the logs from `dmesg | grep -C 10 bbswitch` and `dmesg | grep -C 10 nvidia`: https://gist.github.com/sodar/0be5c5c4e1a2a68de6e9995af7e52197

bluca, 2016-08-30:
Please make sure your version of Bumblebee has the latest patches available in the develop branch. There are fixes for the module loading/unloading. You can check that with the maintainer of your distribution.

sodar, 2016-08-30:
I have installed bumblebee from develop branch (package `bumblebee-git` from Arch Linux's AUR) and module loading/unloading works correctly now. Thank you @bluca :)

hemenkapadia, 2016-09-12:
I have the exact same machine and after a lot of tries have been able to get it working as per my need. See if the documentation at http://hemenkapadia.github.io/blog/2016/05/07/Ubuntu-with-Nvidia-Bumblebee.html helps you.

veaceslav, 2016-10-17:
I have a similar problem with only bbswitch enabled. I can not load the bbswitch module at boot without gettting the Frozen X. Also if I have bbswitch module installed, my system will not restart. Running 4.7 Kernel of Archlinux. 

Laptop Asus GL 552VW
GTX 960M

ydajwz123, 2017-06-04:
sorry for my poor English. I have a similar problem that if I turn my card off, it will freeze when I try to get access to my gnome. So, I need to get my card on before I login, and turn it off afterwards.
BTW, I'm using the latest kali.
Hardware: i7-6700HQ & GTX960M. (HASEE, a Chinese brand)

MrBollie, 2017-09-04:
Hi there,

I'm still using the GTX960M as in my initial post and worked around this issue by starting bumblebee after X is running. Now, I have another issue: gnome-session (or whatever) is now causing the system to load the nvidia module. And while this is loaded, you can start bumblebeed, but bbswich won't get active.

I've recently installed a laptop the same way. It features a Quadro K2000M and I was quite amazed, that everything works perfect there. You can start bumblebeed before X and gnome and bbswitch switches it off properly. No issues or anything.

So, I presume, we have an issue with how GTX960M is being disabled. It seems like it's not "switched off enough" to prevent the system from loading the nvidia-module, which leads to a hanging system. 

As far as I understand it, bbswitch is a "convenience layer", offering a simple interface for card-specific ACPI calls. If I'm right, is there a way to check out, if there's "more thorough" ways to disable GTX650M, so nothing will try to load the nvidia module?

Lekensteyn, 2017-09-04:
@MrBollie "Disabling the GPU" is different from "prevent a program from loading the nvidia module". The former is done by bbswitch or nouveau. The latter has multiple possibilites. The most reliable method is probably blacklisting or aliasing the nvidia module as done on Ubuntu. You might want to check the develop branch.

MrBollie, 2017-09-06:
Ok, it seems that I've found a solution for my GE62 6QC featureing said GTX960M.
I'm running arch, blacklist didn't work, as X (or gnome) will still cause the nvidia module to be loaded.
But I've added this to the kernel boot params: 

`acpi_osi=! "acpi_osi=Windows 2009"`

I did a systemctl enable bumblebeed and so when the system boots up, bbswitch correctly disables the card, gdm starts and I can log into gnome3. So everything is no working as intended.

Lekensteyn, 2017-09-20:
Since it's working with a workaround from https://github.com/Bumblebee-Project/Bumblebee/issues/764, I'm closing this bug. Have fun with more battery power!

closed 2017-09-20 by Lekensteyn