Bumblebee-Project/bbswitch #138: bbswitch is broken with kernel 4.8.0_rc2

belandsoft, 2016-08-18:
As in the title....

Lekensteyn, 2016-08-18:
You can close issues yourself.

What was your "mistake"? It is very possible that bbswitch does not work with 4.8-rc1 or newer because some changes enabled runtime PM for the PCI root port which interferes with bbswitch. Work is in progress to fix that, but no ETA.

Should you run into new problems with Linux 4.8 and bbswitch 0.8, try to boot with `pcie_port_pm=off` which reverts to pre-4.8 behavior regarding the PCI PM functionality.

belandsoft, 2016-08-18:
I had a broken gcc package, so bbswitch wasn't compile correctly. Now i fixed the compilation and it seems to work fine. If I will find something else broken i will tell you here.

closed 2016-08-18 by belandsoft