Bumblebee-Project/bbswitch #139: nvidia module autoload mucks up bbswitch state machine

zx2c4, 2016-09-10:
I have bbswitch installed and I have the nvidia blob driver, for the times that I use it. I have in /etc/modprobe.d:

```
options bbswitch load_state=0 unload_state=1
```

When bbswitch is loaded, the nvidia card is turned off. When it's unloaded the nvidia card is turned back on.

Unfortunately, it appears that udev automatically loads the nvidia_drm/nvidia_modeset/nvidia module, I suspect due to one of the aliases:

```
$ modinfo nvidia-drm nvidia | grep alias
alias:          pci:v000010DEd00000E00sv*sd*bc04sc80i00*
alias:          pci:v000010DEd*sv*sd*bc03sc02i00*
alias:          pci:v000010DEd*sv*sd*bc03sc00i00*
alias:          char-major-195-*
alias:          pci:v000010DEd00000E00sv*sd*bc04sc80i00*
alias:          pci:v000010DEd*sv*sd*bc03sc02i00*
alias:          pci:v000010DEd*sv*sd*bc03sc00i00*
```

Upon insertion, the nvidia module _turns the card back on_, except bbswitch doesn't know about it. The only way to turn the card back off is by running this:

```
rmmod nvidia_drm
rmmod nvidia_modeset
rmmod nvidia
echo OFF > /proc/acpi/bbswitch
echo ON > /proc/acpi/bbswitch
echo OFF > /proc/acpi/bbswitch
```

Notice that I have to run this OFF/ON/OFF sequence, to get bbswitch's state machine back in order. This results in printing an ughly warning with a stack trace in my dmesg, unfortunately.

So, question: **Is there anyway that I can either prevent udev from loading the nvidia driver based on aliases, find a way to influence ordering, or generally get bbswitch working correctly and doing the right thing? What do I do here?**

Lekensteyn, 2016-09-10:
The way to prevent udev from loading the module via kmod is to blacklist the modules. You can do so by creating an `/etc/modprobe/blacklist-nvidia.conf` file containing stuff like:

```
blacklist nvidia
blacklist nvidia-uvm
blacklist nvidia-drm
blacklist nvidia-modeset
```

Then you will still be able to manually load it using `modprobe nvidia`. The `unload_state` parameter will be removed in the future, do not rely on it. It is recommended to keep bbswitch loaded and control the power state via `/proc/acpi/bbswitch`. This is possible since bbswitch currently does not take ownership of a PCI device. It does require you to manually power on the device before loading the nvidia module though.

zx2c4, 2016-09-10:
I didn't realize that blacklisting only prevented udev from loading it. I thought that it also prevented modprobe from working, but I was wrong. Thanks for the pointer. I now have the perfect setup. I can do `nvidia on && systemctl restart sddm` to start X with the nvidia driver, and then `nvidia off && systemctl restart sddm` to start X with the intel driver while disabling the nvidia driver.

```
zx2c4@thinkpad ~ $ cat /usr/share/sddm/scripts/Xsetup.nvidia 
#!/bin/sh
# Xsetup - run as root before the login dialog appears
xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto
```

```
zx2c4@thinkpad ~ $ cat /usr/share/sddm/scripts/Xsetup.intel 
#!/bin/sh
# Xsetup - run as root before the login dialog appears
PATH="/sbin:/usr/sbin:/bin:/usr/bin"
rmmod nvidia_drm
rmmod nvidia_modeset
rmmod nvidia
echo OFF > /proc/acpi/bbswitch
```

```
zx2c4@thinkpad ~ $ cat /etc/modules-load.d/10-bbswitch.conf 
bbswitch
```

```
zx2c4@thinkpad ~ $ cat /etc/modprobe.d/nvidia-blacklist.conf 
blacklist nvidia
blacklist nvidia_drm
blacklist nvidia_modeset
```

```
zx2c4@thinkpad ~ $ cat /etc/modprobe.d/nvidia-drm.conf 
# PRIME requires the drm driver, but X won't automatically load it, so we force it here
install nvidia /sbin/modprobe --ignore-install nvidia; /sbin/modprobe nvidia_drm
```

```
zx2c4@thinkpad ~ $ cat /etc/X11/xorg.conf.nvidia 
Section "Device"
        Identifier "NVIDIA Graphics"
        Driver "nvidia"
        BusID "PCI:1:0:0"
        Option "AllowEmptyInitialConfiguration"
        Option "Coolbits" "28"
        Option "ConnectToAcpid" "0"
        Option "NoLogo" "1"
EndSection
```

```
Section "Module"
        Load "modesetting"
EndSection    
```

```                                                                                                         
zx2c4@thinkpad ~ $ cat /etc/X11/xorg.conf.intel 
Section "Device"                                                                                                       
        Identifier "Intel Graphics"                                                                                    
        Driver "intel"                                                                                                 
EndSection        
```

```                                                                                                     
zx2c4@thinkpad ~ $ cat /etc/mpv/mpv.conf.nvidia 
vo=vdpau
```

```
zx2c4@thinkpad ~ $ cat /etc/mpv/mpv.conf.intel 
vo=vaapi
```

```
zx2c4@thinkpad ~ $ cat /usr/local/bin/nvidia 
#!/bin/bash

[[ $UID != 0 ]] && exec sudo "$0" "$@"

on() {
        eselect opengl set nvidia
        eselect opencl set nvidia
        ln -svfn xorg.conf.nvidia /etc/X11/xorg.conf
        ln -svfn Xsetup.nvidia /usr/share/sddm/scripts/Xsetup
        ln -svnf mpv.conf.nvidia /etc/mpv/mpv.conf
        echo ON > /proc/acpi/bbswitch
}

off() {
        eselect opengl set xorg-x11
        eselect opencl set intel
        ln -svfn xorg.conf.intel /etc/X11/xorg.conf
        ln -svfn Xsetup.intel /usr/share/sddm/scripts/Xsetup
        ln -svnf mpv.conf.intel /etc/mpv/mpv.conf
}

case "$1" in
        on) on ;;
        off) off ;;
        *) echo "Usage: $0 on|off" >&2 ;;
esac
```

closed 2016-09-10 by zx2c4