Bumblebee-Project/bbswitch #141: Asus UX501VW discrete card is still ON after attempt to power off

vlasenko, 2016-10-20:
Hi,

I'm trying to use bbswitch for power saving on Asus UX501VW. 
After attempt to power off discrete card with bbswitch it is reported to still stay powered ON.
```
$ echo OFF > /proc/acpi/bbswitch
$ cat /proc/acpi/bbswitch
0000:01:00.0 ON
```

Linux kernel version: 4.8.2-040802-generic
XOrg: 1.18.4
xorg-xserver-video-intel: 2.99.917

Nouveau or proprietary NVidia drivers are not loaded

All the info as requested by README for issuing bug report is attached to this post.

Thank you for great module!

[acpi_dump_info.txt](https://github.com/Bumblebee-Project/bbswitch/files/540873/acpi_dump_info.txt)
[ASUSTeK_COMPUTER_INC.-N501VW.tar.gz](https://github.com/Bumblebee-Project/bbswitch/files/540877/ASUSTeK_COMPUTER_INC.-N501VW.tar.gz)
[dmesg.txt](https://github.com/Bumblebee-Project/bbswitch/files/540876/dmesg.txt)
[kernel.txt](https://github.com/Bumblebee-Project/bbswitch/files/540874/kernel.txt)
[proc_acpi_bbswitch.txt](https://github.com/Bumblebee-Project/bbswitch/files/540875/proc_acpi_bbswitch.txt)
[xorg_ver.txt](https://github.com/Bumblebee-Project/bbswitch/files/540878/xorg_ver.txt)

Lekensteyn, 2016-10-21:
If you just want to save power, I recommend you to remove bbswitch (since you do not use bumblebee). The nouveau kernel module in kernel 4.8 supports a newer PM method which is superior to the current method in bbswitch (for newer laptops).

Can you give that a go? I have some notes at https://github.com/Bumblebee-Project/Bumblebee/issues/808#issuecomment-252312116

Lekensteyn, 2016-10-22:
vgaswitcheroo should not be used for ON/OFF, as you can see the nvidia device (01:00.0) is listed as DynOff which shows that nouveau manages the power using the runtime power management framework (and indeed, it is currently off).

So you do not have to do anything else now, it seems working now with proper power savings. If you want to use an external monitor later, then you might have to run `lspci` or `xrandr` to trigger power-on, but otherwise everything is automatic.

vlasenko, 2016-10-22:
I see. Thank you for your help!

Lekensteyn, 2016-10-22:
You're welcome!

closed 2016-10-22 by Lekensteyn