Bumblebee-Project/bbswitch #143: dGPU unusable after suspend

eke2152, 2016-11-07:
I have a Asus laptop specs:
i5-6200u, 940mx, realtek rtl8723 wifi-card

I tried with kernel 4.4, 4.8 and 4.9rc3, fedora, ubuntu and manjaro.
Computer runs ok and bbswitch can power on and off the card properly with bumblebee and nvidia-prime but after suspend-resume dGPU becomes unusable.

Ubuntu 16.04 kernel 4.4:
with nvidia-prime:
-PC works ok with dGPU
-Suspend & resume -> freeze
with bumblebee:
-PC works ok with primusrun.
-Suspend & resume -> bumblebee says NVIDIA GPU couldn't initialized but system works ok.

Fedora 25 kernel 4.8:
with nouveau:
-DRI_PRIME=1 glxgears -> works ok
-Suspend & resume -> DRI_PRIME=1 glxgears -> whole system freezes.

Also i have another problem too but im not sure its related. realtek wifi card giving pcie aer correctable errors flooding dmesg (i use pci=noaer as a workaround functionally no problem.) after suspend&resume these messages go away.
[ASUSTeK_COMPUTER_INC.-X555UQ.tar.gz](https://github.com/Bumblebee-Project/bbswitch/files/575347/ASUSTeK_COMPUTER_INC.-X555UQ.tar.gz)

Lekensteyn, 2016-11-07:
Try to boot with `acpi_osi=! acpi_osi="!Windows 2009"`, otherwise the power might not be restored as you have noticed. No idea about the Wi-Fi card.

Possibly related: #134

eke2152, 2016-11-07:
acpi_osi workaround doesn't work in my case. I noticed something that if i unplug powercord when sleep state computer shuts off. (Both in windows 10 & linux) I dont know if this is expected behaviour with new laptops.

Lekensteyn, 2016-11-07:
Complete shutdown when unplugging power cord sounds like a firmware issue. Have you looked for a BIOS update?

eke2152, 2016-11-08:
There is no bios update however i re-flashed same BIOS and shutdown problem solved but other issues remaining so should i attach another acpidump in case something changed with re-flash?

Lekensteyn, 2016-11-08:
It is worth creating a new acpidump and check for differences (just `acpidump.txt` is sufficient if the dmidecode output did not change). Re-flashing your BIOS has possibly reverted some changes to their defaults.

eke2152, 2016-11-13:
Sorry i couldn't reply earlier so i think im getting somewhere. Latest bios Update description says update nv vbios so when i reflashed bios there is a line in bios settings showing nv vbios version but after booting linux

Suspend resume --> freeze poweroff from button

And nv vbios line from bios is gone. After that pc power off instead of sleeping when in battery.
So in linux nvidia vbios somehow deleted from bios (is this possible?)

My understanding is that i have a device specific vbios and in linux nvidia driver loads its generic vbios and this causing problem so this is unrelated to bbswitch. I am closing this

closed 2016-11-13 by eke2152

Lekensteyn, 2016-11-13:
Hm, can you also reproduce the issue after full shutdown instead of reboot? Rebooting might be diifferent from shutdown.

eke2152, 2016-11-14:
Yes reboot and full shutdown same issue but i think this is related to the prop drivers when using nouveau system freezes after suspend but there is no bios issue.