Bumblebee-Project/bbswitch #144: Work to ASUS N53SV is G540M

TryAztec, 2016-11-12:
[    9.905038] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: data=ordered
[   10.190379] systemd-journald[166]: Received request to flush runtime journal from PID 1
[   13.283279] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
[   13.552316] r8169 0000:04:00.0 enp4s0: link down
[   13.552327] r8169 0000:04:00.0 enp4s0: link down
[   13.552407] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
[   13.556212] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   13.571569] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   13.593905] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   14.195628] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   14.947749] bbswitch: loading out-of-tree module taints kernel.
[   14.948253] bbswitch: version 0.8
[   14.948264] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   14.948275] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.GFX0
[   14.948293] ACPI Warning: \_SB.PCI0.PEG0.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[   14.948570] bbswitch: detected an Optimus _DSM function
[   14.948592] pci 0000:01:00.0: enabling device (0000 -> 0003)
[   14.948715] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[   14.952106] bbswitch: disabling discrete graphics
[   14.952124] ACPI Warning: \_SB.PCI0.PEG0.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[   15.474492] r8169 0000:04:00.0 enp4s0: link up
[   15.474508] IPv6: ADDRCONF(NETDEV_CHANGE): enp4s0: link becomes ready

uname -a
Linux notebook 4.8.7-1-ARCH #1 SMP PREEMPT Thu Nov 10 17:22:48 CET 2016 x86_64 GNU/Linux

Xorg -version
X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.5.4-1-ARCH x86_64 
Current Operating System: Linux notebook 4.8.7-1-ARCH #1 SMP PREEMPT Thu Nov 10 17:22:48 CET 2016 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=ea98cd72-9352-4cac-8cb7-eb7549df390f rw quiet
Build Date: 19 July 2016  05:54:24PM
 
Current version of pixman: 0.34.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.

==========================================
Dont start is discrete videocard. Start is integrated.

Lekensteyn, 2016-11-12:
Sorry, can you give more details? What commands have you ran, what is the output?

Did it ever work (is it a regression)?

TryAztec, 2016-11-12:
tee /proc/acpi/bbswitch <<<ON
I'm install package bumblebee, bbswitch and nvidia.

Lekensteyn, 2016-11-12:
What happens after that command? Anything in dmesg?

What is your goal?

TryAztec, 2016-11-12:
I want use discrete card. Or using integrated and discrete card is only.
[   14.947749] bbswitch: loading out-of-tree module taints kernel.
[   14.948253] bbswitch: version 0.8
[   14.948264] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   14.948275] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.GFX0
[    0.852525] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[    0.780609] vgaarb: setting as boot device: PCI:0000:00:02.0
[    0.780611] vgaarb: device added: PCI:0000:00:02.0,decodes=io+mem,owns=io+mem,locks=none
[    0.780615] vgaarb: device added: PCI:0000:01:00.0,decodes=io+mem,owns=none,locks=none
[    0.780615] vgaarb: loaded
[    0.780616] vgaarb: bridge control possible 0000:01:00.0
[    0.780617] vgaarb: no bridge control possible 0000:00:02.0

[root@notebook kosten]# lspci
00:00.0 Host bridge: Intel Corporation 2nd Generation Core Processor Family DRAM Controller (rev 09)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200/2nd Generation Core Processor Family PCI Express Root Port (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
00:16.0 Communication controller: Intel Corporation 6 Series/C200 Series Chipset Family MEI Controller #1 (rev 04)
00:1a.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 (rev 05)
00:1b.0 Audio device: Intel Corporation 6 Series/C200 Series Chipset Family High Definition Audio Controller (rev 05)
00:1c.0 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 1 (rev b5)
00:1c.1 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 2 (rev b5)
00:1c.5 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 6 (rev b5)
00:1d.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 (rev 05)
00:1f.0 ISA bridge: Intel Corporation HM65 Express Chipset Family LPC Controller (rev 05)
00:1f.2 SATA controller: Intel Corporation 6 Series/C200 Series Chipset Family 6 port SATA AHCI Controller (rev 05)
00:1f.3 SMBus: Intel Corporation 6 Series/C200 Series Chipset Family SMBus Controller (rev 05)
01:00.0 VGA compatible controller: NVIDIA Corporation GF108M [GeForce GT 540M] (rev ff)
03:00.0 Network controller: Qualcomm Atheros AR9285 Wireless Network Adapter (PCI-Express) (rev 01)
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 06)

TryAztec, 2016-11-12:
14.947749] bbswitch: loading out-of-tree module taints kernel.
[   14.948253] bbswitch: version 0.8
[   14.948264] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   14.948275] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.GFX0
[   14.948293] ACPI Warning: \_SB.PCI0.PEG0.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)
[   14.948570] bbswitch: detected an Optimus _DSM function
[   14.948592] pci 0000:01:00.0: enabling device (0000 -> 0003)
[   14.948715] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[   14.952106] bbswitch: disabling discrete graphics
[   14.952124] ACPI Warning: \_SB.PCI0.PEG0.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160422/nsarguments-95)

Lekensteyn, 2016-11-12:
I think you misunderstand what `bbswitch` is for. It is a module to toggle the power of the NVIDIA GPU, it does not allow you to switch between integrated and discrete.

To use your discrete, see the options listed at https://wiki.archlinux.org/index.php/NVIDIA_Optimus. In particular:

 - https://wiki.archlinux.org/index.php/PRIME
 - https://wiki.archlinux.org/index.php/Bumblebee

I'll close this issue since there is no issue with bbswitch. You can still reply though.

closed 2016-11-12 by Lekensteyn