Bumblebee-Project/bbswitch #145: Shutdown Freeze Kernel 4.8

Adrixan, 2016-11-23:
Hi, I have an Alienware M11x (the first generation) with a GeForce 335M.
Running Kernel 4.8 on OpenSUSE Tumbleweed, I can't shutdown/reboot/hibernate my system anymore when the discrete graphics was switched off once. I'm using the NVidia proprietary drivers.

Steps to reproduce:
1) Start system
2) load bbswitch, rmmod nvidia
3) issue tee /cat/proc/acpi/bbswitch<<<OFF
4) re-enable discrete graphics, or leave it turned off, doesn't matter
5) shutdown system
6) Gets stuck after seemingly finishing all shutdown jobs.
Trying various acpi_osi or pcie_port_pm=off kernel parameters didn't help.
Sometimes the laptop will shutdown when I hit the Esc key in this 'stuck' state, but not reliably so.

Lekensteyn, 2016-11-23:
Got dmesg? What was the last working kernel? Does system sleep (suspend) work?

Adrixan, 2016-11-23:
I unfortunately don't have dmesg.
The latest kernel where I've had it confirmed working was 4.6.7.
Hibernation doesn't work either. Basically the computer fails to shut down in any case.
Maybe also worth mentioning is, that I use the latest nvidia driver 340.98.

Lekensteyn, 2016-11-23:
Hibernation is not the same as suspend-to-RAM. Does that work?

Adrixan, 2016-12-27:
Hi again,

sorry for the long pause. I've meanwhile done several updated but the problem is basically still the same:
Kernel version 4.9 
output from uname -a: Linux M11x.suse 4.9.0-2.g6fbc0c0-default #1 SMP PREEMPT Fri Dec 16 19:51:27 UTC 2016 (6fbc0c0) x86_64 x86_64 x86_64 GNU/Linux
Proprietary nvidia driver at version 340.101
My kernel command line is: resume=/dev/sda1 showopts rcutree.rcu_idle_gp_delay=1 acpi_osi=Linux acpi=force acpi_enforce_resources=lax i915.enable_rc6=1 i915.enable_fbc=1 i915.lvds_downclock=1 i915.semaphores=1 drm.vblankoffdelay=1 acpi_backlight=vendor

No type of shutdown works, this also includes suspend to ram.

A dmesg of the current configuration is attached.

What is currently working is LTS kernel version 4.4.38.
[nvidia.txt](https://github.com/Bumblebee-Project/bbswitch/files/673870/nvidia.txt)

DistantThunder, 2017-01-29:
Hi, I'm using Arch and had the exact same problem on 4.8.x so I had to disable bbswitch.

Now that I have switched to Kernel 4.9.x, the issue seems to be gone and bumblebee and bbswitch seem to be working under optimal conditions again.

I wrote a more detailed report here:
https://github.com/Bumblebee-Project/bbswitch/issues/140#issuecomment-275912317

As for my kernel command line, here it is: `pci=nomsi modprobe.blacklist=nouveau intel_iommu=on acpi_osi=! acpi_osi="Windows 2009"`

Now, acpi support is non-standardized mess "thanks" to vendors of various proficiency and willingness to document said mess but hope this can help.