Bumblebee-Project/bbswitch #147: Asus n551vw with bbswitch >> off the fan speed 25500

raidhon, 2017-01-06:
Hi all.
Can't fix a problem with the fan speed.
When you run the app through `optirun glxgears` ( or any other application ) everything works fine.
But if you close the app 30 seconds, turn on speed fan 25500 rpm. Although this can not be.

Latest BIOS is 205.
Ways with ACPI hence https://github.com/Bumblebee-Project/Bumblebee/issues/764#issuecomment-234494238 tried all not working.
Any idea how to fix it ?

Lekensteyn, 2017-01-07:
Maybe the info for Asus UX501VW (see #134) helps? Can you create a tarball and upload it according to https://bugs.launchpad.net/lpbugreporter/+bug/752542

raidhon, 2017-01-07:
)) Should fix it. Today, recompiled my kernel on 4.9.1, has enabled a couple of new drm and a few modules and it worked!
Now can not list all of what I included need to remember. 
I collected with the `make oldconfig` kernel 4.8.15 transfer configs for 4.9.1.
And walked new possibilities of 4.9.1. 
I hope this information will help someone.
Links to my build of the kernel.
https://drive.google.com/file/d/0BzmvAX7m3n4xZURla0t4UF9NbVk/view?usp=sharing
https://drive.google.com/file/d/0BzmvAX7m3n4xa2VuekhpNWNJNHM/view?usp=sharing
Yes ACPI here https://github.com/Bumblebee-Project/Bumblebee/issues/764#issuecomment-234494238
now you don't.

Lekensteyn, 2017-01-09:
Is bbswitch loaded in your new kernel? Please upload dmesg

DistantThunder, 2017-01-29:
I confirm kernel mainline 4.9.x seems to fix this issue:

https://github.com/Bumblebee-Project/bbswitch/issues/140#issuecomment-275912317

glats, 2017-06-18:
I don't know if create another issue but I have the same problem with Asus GL553VD.
I have ubuntu 17.04 with kernel version 4.12.0-041200rc5-generic

raidhon, 2017-06-29:
@glats Try adding a line to /etc/default/grub
GRUB_CMDLINE_LINUX="acpi_osi=! acpi_osi=\"Windows 2009\""
And reboot.

glats, 2017-06-29:
@raidhon I can't because my touchpad stop working :(

Littlejth, 2017-07-13:
@raidhon @glats I have the FX53VD which has the same internals as the GL553VD and I also have this problem. The ACPI thing you described did not work for me; the fan still spun up after about 30 seconds of the card being disabled. Should we create a new issue for this?

Lekensteyn, 2017-07-16:
@Littlejth Does this occur also if you have absolutely no other program running (e.g. Xorg is stopped and you are at a text console)? (This is to rule out programs like temperature applets that keep resuming the dGPU). Search for any hints in dmesg too.

Littlejth, 2017-07-19:
Even at the text console, the fan still spins up after about 30 seconds. Here is the last portion of my dmesg where I loaded the module, then proceeded to turn off the card using /proc/acpi/bbswitch

>   214.334583] bbswitch: loading out-of-tree module taints kernel.
[  214.335338] bbswitch: version 0.8
[  214.335341] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[  214.335346] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[  214.335352] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20170303/nsarguments-95)
[  214.335421] bbswitch: detected an Optimus _DSM function
[  214.335430] pci 0000:01:00.0: enabling device (0006 -> 0007)
[  214.335536] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[  228.080180] bbswitch: disabling discrete graphics
[  228.080187] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20170303/nsarguments-95)

FallingSnow, 2017-07-21:
Same thing happens to me. modprobe bbswitch and about 30 seconds later fans max out. Haven't tested it in terminal only.
```
[    6.002718] iwlwifi 0000:03:00.0: L1 Enabled - LTR Enabled
[    6.021786] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[    6.149779] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[    6.511338] fuse init (API version 7.26)
[    9.583813] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   10.648895] wlp3s0: authenticate with d8:5d:e2:7c:39:80
[   10.654981] wlp3s0: send auth to d8:5d:e2:7c:39:80 (try 1/3)
[   10.657222] wlp3s0: authenticated
[   10.664018] wlp3s0: associate with d8:5d:e2:7c:39:80 (try 1/3)
[   10.665170] wlp3s0: RX AssocResp from d8:5d:e2:7c:39:80 (capab=0x1011 status=0 aid=3)
[   10.666789] wlp3s0: associated
[   10.666818] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
[   60.842822] bbswitch: version 0.8
[   60.842827] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   60.842833] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[   60.842841] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20170119/nsarguments-95)
[   60.842926] bbswitch: detected an Optimus _DSM function
[   60.843032] bbswitch: disabling discrete graphics
[   60.843035] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20170119/nsarguments-95)
[   61.061091] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is off
```

mex990, 2017-08-21:
same here on my ASUS UX550VE on Debian Testing (buster) Kernel 4.12.0

fogzot, 2018-07-23:
Same problem here. ASUS UX550GE on Debian unstable, kernel 4.18.0-rc5.
I'll upload the needed information to Launchpad soon.

fogzot, 2018-07-23:
Here is the ACPI information:
https://bugs.launchpad.net/lpbugreporter/+bug/752542/comments/837

GrubbyHalo, 2018-08-16:
I have the same issue. Initially after installing bumblebee-nvidia my laptop and ASUS FX553VD would freeze on booting. I then added this kernel param:
`GRUB_CMDLINE_LINUX_DEFAULT="quiet acpi_osi=! acpi_osi='Windows 2009'"`

and i can now boot however the fan still comes on about 30 seconds later at full speed. I have blacklisted nouveau and have verified that bbswitch reports the nvidia card as being off

`cat /proc/acpi/bbswitch
0000:01:00.0 OFF
`
I am running debian buster. uname -a gives me
`Linux asus 4.17.0-1-amd64 #1 SMP Debian 4.17.8-1 (2018-07-20) x86_64 GNU/Linux
`
BTW the above kernel parameter is the only one that lets me boot without freezing.. i.e. i have tried Windows2013 and Windows2015 and they both cause a freeze when booting.
attached is dump from get-acpi-info.sh

[ASUSTeK_COMPUTER_INC.-GL553VD.tar.gz](https://github.com/Bumblebee-Project/bbswitch/files/2293389/ASUSTeK_COMPUTER_INC.-GL553VD.tar.gz)

GrubbyHalo, 2018-08-16:
### Follow up:

After searching through some other related issues i have discovered that if i suspend my laptop TWICE in succession, the fan winds down and doesn't come back up again as was suggested by someone over   [here](https://github.com/Bumblebee-Project/Bumblebee/issues/904)

I have also come to the conclusion that the easiest fix for this issue right now for me is to just let everything run through the [nvidia card .](https://github.com/Bumblebee-Project/Bumblebee/issues/904) Having done this i can't even say i have noticed a considerable drop in power consumption.

GrubbyHalo, 2019-12-12:
It seems that nvidia have finally added support for PRIME offloading onto the nvidia card for linux. I can confirm that this work well. 
[](https://www.phoronix.com/scan.php?page=news_item&px=NVIDIA-435.17-Linux-Driver)