Bumblebee-Project/bbswitch #148: Dell XPS 15 9560: System crashes when turning GPU on/off repeatedly

skoehler, 2017-02-04:
On my brand new Dell XPS 9560, the bbswitch kernel module loads and repeatedly turning the GPU on and off lets the system hang. I don't know why and I can't see any kernel output, unfortunately. I'm using kernel 4.9.7.

My Distro is Gentoo. When booting with systemd, my system would actually hang before I could reach the login prompt. (Probably because the bumblebee daemon tries to disable the GPU). When booting with OpenRC (very serial booting process), I could reach the graphical login primpt, but turning GPU on and off repeatedly would result in a system freeze and crash. I just an echo ON/OFF > /proc/acpi/bbswitch.

It might be that the new 10 series GPU come with new ACPI tables for turning on/off the GPU. On a Dell XPS 15 9550 (960M GPU) everything was working fine.

ArchangeGabriel, 2017-02-04:
Those issues are quite none/expected I think. Did you take a look at https://github.com/Bumblebee-Project/bbswitch/issues/140?

skoehler, 2017-02-04:
Something else is at work here. I added the suggest workaround pcie_port_pm=off to my grub config and it didn't help. The system still freezes during bootup, probably when bumblebeed is loaded.

skoehler, 2017-02-04:
Update: starting X will lock up the machine if the Nvidia device is OFF. So bbswitch itself seems to work, but X won't start. Neither nouveau or nvidia kernel module is loaded. Does X try to talk to the PCI device, somehow?

ArchangeGabriel, 2017-02-05:
Sorry, answered in a hurry. I have two laptop at the moment:

– Dell XPS 9530, which works OK with Bumblebee, not sure with bbswitch alone, should test that.
– HP zBook Studio G3, on which I use bbswitch alone (need to try Bumblebee), but don’t load it at boot since it then freezes my system. So that looks like your case (but it’s not a Pascal GPU).

Both are affected by #140 btw.

I didn’t have the time until now to look into those issues, but I’ll do so next week. I did some tests with nouveau too, but don’t remember the results, and a new kernel requires new tests. ;)

I’ll keep you updated, and we’ll see with @Lekensteyn what can be done then.

skoehler, 2017-02-06:
Something loads the nvidia kernel module when I start sddm. I blame X for that, yet the Xorg logs mention nothing. That seems to be the reason why my laptop hangs when starting X with the dGPU turned off.

From bug #140 I gather that support for PCIe PM is in the works and that will eventually replace the DSM method?

Lekensteyn, 2017-02-06:
The hang issue is possibly not related to pcie_port_pm, but https://bugzilla.kernel.org/show_bug.cgi?id=156341 (#764). Try the acpi_osi workaround listed there.

ArchangeGabriel, 2017-02-06:
@Lekensteyn OK that’s definitively it for me (my machine is already listed by someone else). Looks like I’ve been missing a lot of fun lately (my comment review queue for Bumblebee is at 1098 right now…). I’ve added myself to the CC list.

Lekensteyn, 2017-02-06:
For me Github still shows 143 unread notifications for Bumblebee (none for bbswitch), but there are lots of open issues that still need a response (luckily users are helping each other). Hmm

bluca, 2017-02-06:
I think I've given asnwers to all Ubuntu/Debian related issues, if I missed some please do tag me

stefansedich, 2017-02-08:
I am also seeing a hang on startup too on my Precision 5200 which seems to be ACPI related, if I do no install the module and manually load it everything is fine, but loading it I just end up hanging on boot.

zeraien, 2017-02-11:
I also have the startup issue on Dell XPS 9560, and it seems to be mainly related to X, as it does not seem to affect Wayland.

stefansedich, 2017-02-12:
I can confirm this @zeraien I just installed 3.22 and moved over to it today using wayland, I can boot fine with bbswitch installed although my system does lock up a short time after booting.

zeraien, 2017-02-12:
I also have issues with freezing while in Wayland, but it seems to be related to the launching of older X-based applications, which probably start the X machinery and this probably triggers the nvidia driver/bbswitch and thus you get a crash.
I'm on Fedora 25 (kernel 4.9.8), bumblebee 3.2.1

stefansedich, 2017-02-12:
If I just manually load bbswitch `sudo make load` on each boot everything is working fine for me without any crashes, Only seems to be an issue for me when installed, I might just run a script to load it this way after boot.

pronobis, 2017-02-15:
@stefansedich can you explain what exactly you're doing with `make load`? I'm on XPS 15 9560 and having the freeze with bbswitch off. Indeed acpi_osi=! acpi_osi=Windows 2009 helps, but that disables touchpad as discussed above.

stefansedich, 2017-02-15:
Sure @pronobis so basically I did not run make install, I booted to recovery mode, then ran `make uninstall`. Once my machine starts back up I simply run `make load` to load the driver, which does mean if I reboot I need to do this again but this has been working nice and stable for me until I find out how to get it working properly when installed.

pronobis, 2017-02-15:
Does that mean that you can run the commands that cause freeze after you load bbswitch? For instance, in my case it's nvidia-smi, lspci, lshw etc.

stefansedich, 2017-02-15:
I don't seem to get any freeze running lspci or lshw, I could be wrong here
mind you but I can tell from powertop that my nvidia card is turned off so
have assumed that it was all working as expected.

pronobis, 2017-02-15:
@stefansedich Thanks for the info. Just to summarize:
- cat /proc/acpi/bbswitch gives OFF
- you are on X, not wayland
- you can run lspci several times in a row and no freeze happens?

What is your hardware?

stefansedich, 2017-02-15:
I am actually on wayland now but was working under X too, cat
/proc/acpi/bbswitch gives me a file not found, but looking at gpu-manager I
can see it turns the card off, and looking at my idle watts it appears the
card is off as if I reboot running the nvidia my watts are 5w or so higher.

And yes can run lscpi many times fine, my hardware is a Precision 5200 so
might be a different story due to the different card.

skoehler, 2017-02-19:
I was using `acpi_osi='!Windows 2013' pcie_port_pm=off` and doing `echo OFF > /proc/acpi/bbswitch` followed by running `lspci` twice lets my machine freeze (Dell XPS 9560).

pronobis, 2017-02-20:
@stefansedich The fact that  /proc/acpi/bbswitch gives you file not found means your bbswitch module is likely not loaded. I think I also did not have the problem with simply no NVidia driver installed and used. The problem exists only if you power down a card that is managed by an nvidia driver (nvidia-smi displays the card) and X is using the Intel card with nvidia-prime.

The bug is likely to be the same or related to: https://bugzilla.kernel.org/show_bug.cgi?id=156341

I would urge all Kaby Lake Dell laptop users to post there, since this issue has been previously reported mostly for SkyLake laptops.

pronobis, 2017-02-21:
There is a patch in https://bugzilla.kernel.org/show_bug.cgi?id=156341 that solves the freeze when applied against kernel 4.10.

skoehler, 2017-02-22:
I tried the patch with Linux 4.10 and my Dell XPS 9560 is not freezing anymore. Nice! However, the nvidia driver doesn't load. It gives some errors and I couldn't make them go away yet.

bsdfirst, 2017-02-22:
@skoehler It's good to know that it did indeed work.  Downgrading your nvidia module version is reported to solve the loading issue.  See comment #18 in the Archlinux thread below:

https://bbs.archlinux.org/viewtopic.php?id=223056

skoehler, 2017-02-22:
Downgrading the nvidia driver worked. I can now play games such as Dota 2 with my nvidia card.

stefansedich, 2017-02-22:
Interesting since upgrading to 4.10 I just installed the module again and rebooted, and everything is working as expected! no lockup on boot for me now.

bpinto, 2017-02-28:
@stefansedich that's weird... aparrently the fix has not been accepted upstream:

> Update: the patch didn't get accepted, but I got the hint to try booting with acpi_rev_override. 
> acpi_rev_override=5 instead of the acpi_osi stuff works for me (currently on kernel 4.8), no lockups and the touchpad issues also seem to be gone.

bsdfirst, 2017-02-28:
For what it's worth, it doesn't solve the issue for me - kernel 4.9.11, bios 1.1.3.  I still get the hard lock when I switch the nvidia card off.

skoehler, 2017-03-02:
OK, I tried the kernel option `acpi_rev_override` (it's Boolean, it doesn't take a value like 5, see kernel documentation) and it works just as well as the patch.

bsdfirst, 2017-03-02:
Fair point - although briefly skimming the code, 5 should still have had the same effect (i.e. evaluated to true).  Just tried `=1` in any event.  Still had the lockup - happened to be on the console this time there was a stacktrace.  @skoehler What BIOS version are you running - there were apparently changes to improve nvidia stability which I applied while trying to improve things.

bsdfirst, 2017-03-02:
For those playing at home...your Kernel must include the `CONFIG_ACPI_REV_OVERRIDE_POSSIBLE` option.  The default ArchLinux kernel does not - I can't speak for other distributions.

$ zcat /proc/config.gz | grep CONFIG_ACPI_REV_OVERRIDE_POSSIBLE
# CONFIG_ACPI_REV_OVERRIDE_POSSIBLE is not set

One kernel rebuild later, and setting `acpi_rev_override` solves the lockup issue.

domenkozar, 2017-03-25:
On NixOS I was getting kernel panics as well on XPS 9560.

Following works:

```nix
hardware.nvidiaOptimus.disable = true;
boot.kernelParams = [ "acpi_rev_override=1" ];
```

saroele, 2017-05-04:
I also own the dell xps 15 9560, running ubuntu 17.04.  Everything I read about managing the built-in graphic chip and the nvidia graphic card confuses me.  Is there a working solution today that is safe and does not require expert linux skills?  Could someone write a summary of the steps to take? Thanks a lot!

etiago, 2017-05-29:
@saroele can't help with Ubuntu specifics but essentially what you currently need (and keep in mind that this is a dynamic situation) is to install the latest NVIDIA drivers, install bumblebee and install bbswitch. This is documented in the Arch Linux [wiki](https://wiki.archlinux.org/index.php/Dell_XPS_15_9560#Proprietary_driver_with_bumblebee) and at least the general idea can be applied to any distro, you only have to then use your own package manager, etc. Once you have these three moving parts, your dedicated GPU should be off by default and thus not using power. Whenever you want to run an app on the dedicated GPU, you run it with "optirun <whatever_app>". So if you'd like to run Chrome with dedicated GPU, you'd run optirun chrome.

This should get you a basic working system. What some users experience is that sometimes switching the GPU on and/or off will cause the system to freeze. To get rid of that, you can set the kernel parameter ```acpi_rev_override=1```. This is something that your bootloader (normally GRUB) passes onto the Linux kernel so it allows an ACPI override. Once you've done that, you're all set.

Personally I'm running Arch Linux which is great because it polishes your Linux skills and gives you a better understanding of what's running where. It does also cost quite some initial time to understand it so tread carefully. An alternative is Manjaro Linux, which is Arch with training wheels. You'll get all the benefits of Arch and quite some helpers to get you up and running.

parasquid, 2017-06-11:
I've got the 9560 FHD running Ubuntu 17.04 and nvidia-375 and I can confirm that adding the `acpi_rev_override=1` option in grub does the trick.

Here's the benchmarking results: http://openbenchmarking.org/result/1706111-TR-XPS95602060

I'm posting here because this came up as one of the search results while googling :)

BTW this guy https://www.reddit.com/r/Dell/comments/63cavx/fixed_nvidia_1050_freezing_in_ubuntu_linux/ has got good instructions for whoever wants to try this out.

I'm not sure what are the side effects of this kernel option is though, so I consider this to be a temporary issue until Ubuntu ships us a updated kernel with the proper patches.

XVilka, 2017-06-23:
Have the same bug on 4.12_rc6 kernel on Gentoo: https://bugs.freedesktop.org/show_bug.cgi?id=101553

XVilka, 2017-09-15:
Nobody in kernel/nouveau seems to working on this (no even requests for debugging/testing/more information, whatever). So do not expect any fixes soon.

LordTyrius, 2017-09-27:
> the `acpi_rev_override=1` option in grub does the trick
I just wanted to report that I found a change to this since recently. While the kernel parameter worked for me as well, it does no longer. I'm on manjaro stable kernel 4.13, and since the last update my system will hang on boot with the override set to 1. Luckily I have not experienced the original issues again since I have disabled it again to be able to boot. Just wanted to leave this here if others find themselves in a similar situation.

domenkozar, 2017-10-12:
I can report that 4.13 does break again and for me `acpi_rev_override` has no effect either way, machine won't boot up.

skoehler, 2017-10-12:
My Dell XPS 9560 works without problems with 4.13.x. I'm just using `acpi_rev_override`.

domenkozar, 2017-10-12:
@skoehler so `acpi_rev_override=1`?

carymrobbins, 2017-10-12:
On my Dell XPS 9560 with Linux 4.13.3-1 I have `acpi_rev_override=1` which does indeed set bbswitch to `OFF`; however, I can't turn `bbswitch` to `ON`, neither manually  with `echo ON | sudo tee /proc/acpi/bbswitch` nor via optirun, e.g. `optirun -vv glxgears`.

Both attempts yield the same output in `journalctl` -
```
... kernel: bbswitch: enabling discrete graphics
... kernel: pci 0000:01:00.0: Refused to change power state, currently in D3
```

skoehler, 2017-10-12:
My kernel parameters are `pcie_port_pm=off acpi_rev_override` beside the usual root device etc.

domenkozar, 2017-10-13:
Apparently kernel 4.13.6 doesn't work for me, could be unrelated to bumblebee since it won't boot even with ON or OFF.

Lekensteyn, 2017-10-14:
@domenkozar sounds like an unrelated problem? Unless "won't boot" means that it does not get to the login screen or something. What was the last working kernel version?

Lekensteyn, 2017-10-25:
FWIW, my brother also has this laptop (Dell Inc. XPS 15 9560/05FFDN, BIOS 1.3.3 05/08/2017) now and I can confirm issues with it (using Arch Linux with Linux 4.13.7).

The `acpi_rev_override` boot parameter (value does not matter as @skoehler observed) combined with "Windows 2015" (i.e. the default without `acpi_osi` override) will force the old logic. In theory it should work, but with nouveau it hung (see references from @XVilka above).

One major issue that it sucks much battery during system sleep (51% in 9h, approx. 49Wh with the extended battery model). To be investigated later.

skoehler, 2017-11-02:
Just upgraded vom 4.13.4 to 4.13.11 and my systems hangs during boot. Probably due to this issue. So disregard my above comment that everything is working fine with 4.13.x. Clearly, something is broken after 4.13.4.

XVilka, 2017-11-26:
See also this pull request - should improve the situation after landing in the kernel https://github.com/acpica/acpica/pull/330

lf-, 2018-03-18:
> The fact that /proc/acpi/bbswitch gives you file not found means your bbswitch module is likely not loaded. I think I also did not have the problem with simply no NVidia driver installed and used. [...]

I've reproduced this with a blacklisted nouveau and no bbswitch/bumblebee.

I use the following to turn off the GPU:

`/etc/tmpfiles.d/gpu-off.conf`:
```
w /proc/acpi/call - - - - \\_SB.PCI0.PEG0.PEGP._OFF
```

<details><summary>logs</summary><p>
<pre><code>
~ » uname -a
Linux archenemy 4.15.5-1-ARCH #1 SMP PREEMPT Thu Feb 22 22:15:20 UTC 2018 x86_64 GNU/Linux
~ » cat /proc/cmdline
initrd=\initramfs-linux.img root=UUID=3879c851-c423-45ae-853e-320fa144a04c rootflags=subvol=@ pci=nommconf
~ » lsmod
Module                  Size  Used by
ccm                    20480  6
xt_CHECKSUM            16384  1
ipt_REJECT             16384  2
xt_tcpudp              16384  6
tun                    45056  1
bridge                188416  0
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
devlink                49152  0
ebtable_filter         16384  0
ebtables               36864  1 ebtable_filter
ip6table_filter        16384  0
ip6_tables             32768  1 ip6table_filter
iptable_nat            16384  0
nf_nat_ipv4            16384  1 iptable_nat
nf_nat                 36864  1 nf_nat_ipv4
iptable_mangle         16384  1
iptable_filter         16384  1
tpm_crb                16384  0
nft_reject_inet        16384  1
nf_reject_ipv4         16384  2 ipt_REJECT,nft_reject_inet
nf_reject_ipv6         16384  1 nft_reject_inet
nft_reject             16384  1 nft_reject_inet
nft_meta               16384  4
nf_conntrack_ipv6      16384  2
nf_defrag_ipv6         36864  1 nf_conntrack_ipv6
nf_conntrack_ipv4      16384  3
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
nft_ct                 20480  2
nf_conntrack          155648  5 nft_ct,nf_conntrack_ipv6,nf_conntrack_ipv4,nf_nat_ipv4,nf_nat
libcrc32c              16384  2 nf_conntrack,nf_nat
crc32c_generic         16384  0
nft_set_bitmap         16384  0
nft_set_hash           28672  1
nft_set_rbtree         16384  0
nf_tables_inet         16384  4
nf_tables_ipv6         16384  1 nf_tables_inet
nf_tables_ipv4         16384  1 nf_tables_inet
nf_tables             106496  28 nft_ct,nft_set_bitmap,nft_reject,nft_set_hash,nf_tables_ipv6,nf_tables_ipv4,nft_reject_inet,nft_meta,nft_set_rbtree,nf_tables_inet
nfnetlink              16384  1 nf_tables
nls_iso8859_1          16384  1
nls_cp437              20480  1
vfat                   20480  1
fat                    77824  1 vfat
arc4                   16384  2
snd_hda_codec_hdmi     57344  1
ath10k_pci             65536  0
ath10k_core           466944  1 ath10k_pci
snd_hda_codec_realtek   110592  1
ath                    32768  1 ath10k_core
snd_hda_codec_generic    86016  1 snd_hda_codec_realtek
[... 107 lines cut ...]
btrfs                1331200  1
xor                    24576  1 btrfs
zstd_decompress        94208  1 btrfs
zstd_compress         196608  1 btrfs
xxhash                 16384  2 zstd_compress,zstd_decompress
raid6_pq              122880  1 btrfs
rtsx_pci_sdmmc         28672  0
mmc_core              172032  1 rtsx_pci_sdmmc
serio_raw              16384  0
atkbd                  32768  0
libps2                 16384  2 atkbd,psmouse
ahci                   40960  0
libahci                40960  1 ahci
xhci_pci               16384  0
crc32c_intel           24576  2
libata                278528  2 ahci,libahci
xhci_hcd              258048  1 xhci_pci
rtsx_pci               65536  2 rtsx_pci_sdmmc,rtsx_pci_ms
scsi_mod              258048  1 libata
usbcore               286720  4 uvcvideo,xhci_pci,btusb,xhci_hcd
usb_common             16384  1 usbcore
i8042                  32768  1 dell_laptop
serio                  28672  6 serio_raw,atkbd,psmouse,i8042
```
</pre></code></details>

XVilka, 2018-05-07:
For everyone who had the issues with ACPI that required acpi_rev_override - important fixes were merged into the mainline kernel git, so starting from 4.17-rc2 kernel you can try without ACPI override option. I have tried with Dell XPS 9560 and it really works.

chenxiaolong, 2018-05-07:
@XVilka That's great news! Do you happen to know which exact patches from 4.17-rc2 fix the issue?

XVilka, 2018-05-08:
@chenxiaolong I don't know exact commit - the MLC fixes landed in multiple commits, but it is mostly updated `drivers/acpi/acpica` from https://github.com/acpica/acpica/

PumbaPe, 2018-06-25:
@XVilka I just upgraded from 4.16 to 4.17.2 (Fedora 28), but unfortunately it still doesn't work without acpi_rev_override. The system still hangs before the login screen.

XVilka, 2018-06-25:
Very strange, can you record dmesg for exact ACPI error?

PumbaPe, 2018-06-27:
@XVilka Actually there aren't any errors. The log shows exactly the same messages regarding ACPI (except the boot parameter of course). The last couple of messages from a failed boot:

```
$ journalctl -k -b -1 | grep -i acpi
Jun 28 01:13:13 kernel: ACPI: AC Adapter [AC] (on-line)
Jun 28 01:13:13 kernel: ACPI: Lid Switch [LID0]
Jun 28 01:13:13 kernel: ACPI: Power Button [PBTN]
Jun 28 01:13:13 kernel: ACPI: Sleep Button [SBTN]
Jun 28 01:13:13 kernel: ACPI: Power Button [PWRF]
Jun 28 01:13:13 kernel: ACPI: Thermal Zone [THM] (25 C)
Jun 28 01:13:13 kernel: ACPI: Battery Slot [BAT0] (battery present)
Jun 28 01:13:14 kernel: ACPI: Video Device [GFX0] (multi-head: yes  rom: no  post: no) 
Jun 28 01:13:14 kernel: ACPI: Video Device [PEGP] (multi-head: no  rom: yes  post: no) 
```

And from a successful boot:

```
$ journalctl -k -b -2 | grep -i acpi
Jun 28 01:14:50 kernel: ACPI: AC Adapter [AC] (on-line)
Jun 28 01:14:50 kernel: ACPI: Lid Switch [LID0]
Jun 28 01:14:50 kernel: ACPI: Power Button [PBTN]
Jun 28 01:14:50 kernel: ACPI: Sleep Button [SBTN]
Jun 28 01:14:50 kernel: ACPI: Power Button [PWRF]
Jun 28 01:14:50 kernel: ACPI: Thermal Zone [THM] (25 C)
Jun 28 01:14:50 kernel: ACPI: Battery Slot [BAT0] (battery present)
Jun 28 01:14:51 kernel: ACPI: Video Device [GFX0] (multi-head: yes  rom: no  post: no) 
Jun 28 01:14:51 kernel: ACPI: Video Device [PEGP] (multi-head: no  rom: yes  post: no) 
Jun 27 23:14:53 kernel: ACPI Warning: \_SB.IETM._TRT: Return Package has no elements (empty) (20180313/nsprepkg-96)
Jun 27 23:14:56 kernel: ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20180313/nsarguments-66)
```

The only error in the whole log is

```
$ journalctl -b -1 -p 3
Jun 28 01:13:13 kernel: rtc_cmos 00:02: nvmem registration failed
```

It's the same for a successful boot though.

mxncr, 2018-07-03:
@XVilka Same as @PumbaPe, I have upgrade to kernel 4.17.3 and I still need acpi_rev_override=1 to boot.

demurgos, 2018-07-17:
Using Arch Linux, updated to the latest versions, I still had to use `acpi_rev_override=1` (or `bbswitch` caused a black screen). I had some trouble finding this info, so I sent the PR #174 to make this information easier to find.