Bumblebee-Project/bbswitch #150: Not working after suspend

NoMore201, 2017-03-14:
Distro: Archlinux
Machine: ASUS S551LN
Kernel: Linux hostname 4.10.1-1-ARCH #1 SMP PREEMPT Sun Feb 26 21:08:53 UTC 2017 x86_64 GNU/Linux

The module is working fine, but after resuming the notebook from suspend the nvidia card stays on, there's no way I can disable it again (unless rebooting the system) and I get the following messages:
```
[  993.517572] ACPI: Low-level resume complete
[  993.517691] ACPI : EC: EC started
[  993.517692] PM: Restoring platform NVS memory
[  993.518287] Suspended for 7.841 seconds
[  993.523514] Enabling non-boot CPUs ...
[  993.523579] x86: Booting SMP configuration:
[  993.523581] smpboot: Booting Node 0 Processor 1 APIC 0x2
[  993.525077]  cache: parent cpu1 should not be sleeping
[  993.525311] microcode: sig=0x40651, pf=0x40, revision=0x17
[  993.528568] microcode: updated to revision 0x1f, date = 2016-04-01
[  993.528835] CPU1 is up
[  993.528877] smpboot: Booting Node 0 Processor 2 APIC 0x1
[  993.530687]  cache: parent cpu2 should not be sleeping
[  993.530930] microcode: sig=0x40651, pf=0x40, revision=0x1f
[  993.531147] CPU2 is up
[  993.531196] smpboot: Booting Node 0 Processor 3 APIC 0x3
[  993.532552]  cache: parent cpu3 should not be sleeping
[  993.533027] CPU3 is up
[  993.543913] ACPI: Waking up from system sleep state S3
[  993.551234] ACPI : EC: interrupt unblocked
[  993.570098] ehci-pci 0000:00:1d.0: System wakeup disabled by ACPI
[  993.676497] xhci_hcd 0000:00:14.0: System wakeup disabled by ACPI
[  993.676693] PM: noirq resume of devices complete after 125.749 msecs
[  993.677426] PM: early resume of devices complete after 0.668 msecs
[  993.677698] ACPI : EC: event unblocked
[  993.678968] ACPI : button: The lid device is not compliant to SW_LID.
[  993.679090] rtc_cmos 00:02: System wakeup disabled by ACPI
[  993.689223] sd 0:0:0:0: [sda] Starting disk
[  993.689252] sd 1:0:0:0: [sdb] Starting disk
[  993.742562] xhci_hcd 0000:00:14.0: port 7 resume PLC timeout
[  993.757828] xhci_hcd 0000:00:14.0: port 4 resume PLC timeout
[  993.758224] proc_thermal 0000:00:04.0: Unsupported event [0x84]
[  993.986694] usb 2-8: reset high-speed USB device number 5 using xhci_hcd
[  994.003367] ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[  994.003638] ata2.00: supports DRM functions and may not be fully accessible
[  994.004458] ata2.00: disabling queued TRIM support
[  994.005156] ata2.00: supports DRM functions and may not be fully accessible
[  994.005802] ata2.00: disabling queued TRIM support
[  994.006182] ata2.00: configured for UDMA/133
[  994.230015] usb 2-5: reset high-speed USB device number 4 using xhci_hcd
[  994.437639] PM: resume of devices complete after 760.200 msecs
[  994.437980] PM: Finishing wakeup.
[  994.437982] Restarting tasks ... done.
[  994.471242] bbswitch: disabling discrete graphics
[  994.471251] ACPI Warning: \_SB.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[  994.625118] usb 2-4: new full-speed USB device number 6 using xhci_hcd
[  996.126567] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[  996.130783] ata1.00: configured for UDMA/133
[  996.164779] bbswitch: disabling discrete graphics
[  996.164810] ACPI Warning: \_SB.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[  996.178322] ------------[ cut here ]------------
[  996.178327] WARNING: CPU: 2 PID: 1823 at drivers/pci/pci.c:1641 pci_disable_device+0xa8/0xd0
[  996.178328] pci 0000:04:00.0: disabling already-disabled device
[  996.178328] Modules linked in: ctr ccm xt_tcpudp iptable_filter snd_hda_codec_hdmi joydev arc4 iTCO_wdt iTCO_vendor_support intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul nls_iso8859_1 asus_nb_wmi ghash_clmulni_intel asus_wmi nls_cp437 pcbc sparse_keymap vfat mxm_wmi fat iwlmvm mac80211 aesni_intel aes_x86_64 crypto_simd glue_helper cryptd iwlwifi intel_cstate rtsx_usb_ms memstick intel_rapl_perf psmouse input_leds pcspkr evdev mousedev cfg80211 intel_pch_thermal mac_hid rfkill i2c_i801 lpc_ich thermal r8168(O) i915 snd_soc_rt5640 snd_hda_codec_realtek snd_hda_codec_generic snd_soc_rl6231 drm_kms_helper snd_soc_core wmi snd_hda_intel drm battery int3402_thermal snd_hda_codec int3406_thermal snd_soc_sst_acpi video snd_soc_sst_match snd_compress
[  996.178367]  snd_hda_core snd_pcm_dmaengine ac97_bus elan_i2c i2c_hid snd_hwdep intel_gtt i2c_designware_platform mei_me snd_pcm syscopyarea 8250_dw i2c_designware_core spi_pxa2xx_platform sysfillrect ac processor_thermal_device sysimgblt fjes int340x_thermal_zone fb_sys_fops i2c_algo_bit snd_timer int3400_thermal acpi_thermal_rel mei intel_soc_dts_iosf intel_smartconnect snd asus_wireless soundcore button tpm_tis tpm_tis_core tpm shpchp sch_fq_codel bbswitch(O) ip_tables x_tables btrfs xor rtsx_usb_sdmmc rtsx_usb raid6_pq sd_mod hid_generic usbhid hid serio_raw atkbd libps2 crc32c_intel ahci libahci libata xhci_pci ehci_pci scsi_mod xhci_hcd ehci_hcd usbcore usb_common i8042 serio sdhci_acpi sdhci led_class mmc_core
[  996.178405] CPU: 2 PID: 1823 Comm: tee Tainted: G           O    4.10.1-1-ARCH #1
[  996.178405] Hardware name: ASUSTeK COMPUTER INC. S551LN/S551LN, BIOS S551LN.209 07/08/2014
[  996.178406] Call Trace:
[  996.178411]  dump_stack+0x63/0x83
[  996.178413]  __warn+0xcb/0xf0
[  996.178415]  warn_slowpath_fmt+0x5f/0x80
[  996.178417]  ? __pci_set_master+0x3b/0xf0
[  996.178419]  pci_disable_device+0xa8/0xd0
[  996.178422]  bbswitch_off+0xad/0x240 [bbswitch]
[  996.178424]  bbswitch_proc_write+0xb0/0xc7 [bbswitch]
[  996.178427]  proc_reg_write+0x42/0x70
[  996.178429]  __vfs_write+0x37/0x140
[  996.178431]  vfs_write+0xb6/0x1a0
[  996.178433]  SyS_write+0x55/0xc0
[  996.178435]  entry_SYSCALL_64_fastpath+0x1a/0xa9
[  996.178436] RIP: 0033:0x7f842de4d3c0
[  996.178437] RSP: 002b:00007ffd67e41238 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
[  996.178439] RAX: ffffffffffffffda RBX: 0000000000ddd090 RCX: 00007f842de4d3c0
[  996.178440] RDX: 0000000000000004 RSI: 00007ffd67e41330 RDI: 0000000000000003
[  996.178441] RBP: 00007f842e10a400 R08: 0000000000ddd3a0 R09: 00007f842e2fb4c0
[  996.178442] R10: 0000000000000846 R11: 0000000000000246 R12: 0000000000000000
[  996.178442] R13: 00007ffd67e40f20 R14: 0000000000dde590 R15: 0000000000ddd090
[  996.178444] ---[ end trace b6e16084b879c424 ]---
```

Lekensteyn, 2017-03-20:
Did this occur with older kernel versions too?

Have you tried whether the nouveau driver with runtime PM works for you? That is the recommended approach if you don't use the binary nvidia driver.

NoMore201, 2017-03-21:
Yes, it occurs with older kernel versions, unfortunately I cannot tell exactly from which version.
I use the official nvidia driver, but I can check with the nouveau one.

Fincer, 2017-03-22:
I have exactly the same issue as NoMore201. After waking up the laptop Nvidia card is enabled and there's no way to disable it. There were no any issues with this before.

I'm using Arch Linux with the latest kernel (4.10.4) & mesa/Nvidia drivers (17.0.1 & 378.13, respectively).
Intel HD4000 + Geforce GTX 760M.

Fincer, 2017-03-22:
I just booted up my laptop from powered-off state and well, Nvidia card is running by default. So, this issue may not relate only to suspend state. I don't know.

Fincer, 2017-03-23:
Not sure if related but worth noting:

[Bumblebee + Nvidia 378.13 driver: RmInitAdapter failed](https://bbs.archlinux.org/viewtopic.php?id=223969)

Partial output of `dmesg | grep -C 10 bbswitch:` command on my Arch Linux:

`...
[  376.216768] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[  376.419570] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[  377.744949] bbswitch: Unloaded. Discrete card 0000:01:00.0 is on
[  377.773648] bbswitch: version 0.8
[  377.773653] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[  377.773658] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[  377.773669] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[  377.773745] bbswitch: detected an Optimus _DSM function
[  377.773752] bbswitch: device 0000:01:00.0 is in use by driver 'nvidia', refusing OFF
[  377.773753] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[  379.141629] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[  379.141683] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
...`

Kernel: 4.10.4-1-ARCH

mirh, 2017-03-25:
Isn't this basically the same of the closed-non-closed #142?

Lekensteyn, 2017-03-25:
@mirh Not sure, @NoMore201 can you post your info according to https://bugs.launchpad.net/lpbugreporter/+bug/752542?

mirh, 2017-03-25:
You could find guides that arrived to contemplate tables are now exposed under `/sys/firmware/acpi/tables/` and there's no need to install additional programs 😥

NoMore201, 2017-03-25:
@Lekensteyn here is the output of the script
[ASUSTeK_COMPUTER_INC.-S551LN.tar.gz](https://github.com/Bumblebee-Project/bbswitch/files/870180/ASUSTeK_COMPUTER_INC.-S551LN.tar.gz)

tlvince, 2017-04-17:
FWIW, I'm experiencing the same issue here on 4.10.10 "disabling already-disabled device". Unloading and reloading the bbswitch module on resume works. ACPI DSDT info: 
[Timi-TM1613.tar.gz](https://github.com/Bumblebee-Project/bbswitch/files/925355/Timi-TM1613.tar.gz).

Fincer, 2017-09-30:
Any update on this issue?

NoMore201, 2018-08-10:
This occurs also using the nouveau module. To be more precise, after resuming from suspend
when using nouveau, `/sys/kernel/debug/vgaswitcheroo/switch` shows

```
0:IGD:+:Pwr:0000:00:02.0
1:DIS: :DynOff:0000:04:00.0
```

which seems ok, but if I check power consumption using powertop, I get ~11W battery discharge rate
against the ~6W on a fresh boot. Could it be an ACPI bug? Where should I report?