Bumblebee-Project/bbswitch #151: [Fedora 25] Cannot access secondary GPU - error: [XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied

hebertjulio, 2017-04-04:
**OS**
> Fedora 25

**Kernel**
> Linux 5470 4.10.6-200.fc25.x86_64 #1 SMP Mon Mar 27 14:06:23 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

**This error occurs when I execute the command: vblank_mode=0 optirun -b primus glxspheres64**
> [ 2107.529582] [ERROR]Cannot access secondary GPU - error: [XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied
> [ 2107.529668] [ERROR]Aborting because fallback start is disabled.

**bumblebee-nvidia -c**
> nvidia.ko compiled into in the kernel tree ok.
> modinfo output for NVIDIA:
> 
> filename:       /lib/modules/4.10.6-200.fc25.x86_64/kernel/drivers/video/nvidia.ko
> alias:          char-major-195-*
> version:        375.39
> supported:      external
> license:        NVIDIA
> srcversion:     02F9F40DB2D0FC489009343
> alias:          pci:v000010DEd00000E00sv*sd*bc04sc80i00*
> alias:          pci:v000010DEd*sv*sd*bc03sc02i00*
> alias:          pci:v000010DEd*sv*sd*bc03sc00i00*
> depends:        
> vermagic:       4.10.6-200.fc25.x86_64 SMP mod_unload 
> parm:           NVreg_Mobile:int
> parm:           NVreg_ResmanDebugLevel:int
> parm:           NVreg_RmLogonRC:int
> parm:           NVreg_ModifyDeviceFiles:int
> parm:           NVreg_DeviceFileUID:int
> parm:           NVreg_DeviceFileGID:int
> parm:           NVreg_DeviceFileMode:int
> parm:           NVreg_UpdateMemoryTypes:int
> parm:           NVreg_InitializeSystemMemoryAllocations:int
> parm:           NVreg_UsePageAttributeTable:int
> parm:           NVreg_MapRegistersEarly:int
> parm:           NVreg_RegisterForACPIEvents:int
> parm:           NVreg_CheckPCIConfigSpace:int
> parm:           NVreg_EnablePCIeGen3:int
> parm:           NVreg_EnableMSI:int
> parm:           NVreg_TCEBypassMode:int
> parm:           NVreg_UseThreadedInterrupts:int
> parm:           NVreg_MemoryPoolSize:int
> parm:           NVreg_RegistryDwords:charp
> parm:           NVreg_RmMsg:charp
> parm:           NVreg_AssignGpus:charp
> 
> Check bbswitch kernel module...
> 
> Warning! bbswitch is not loaded into this kernel!
> Since bbswitch is not loaded into the kernel, lets do additional checks on it...
> dkms status bbswitch output:
> bbswitch, 0.8.0, 4.10.5-200.fc25.x86_64, x86_64: installed
> bbswitch, 0.8.0, 4.10.6-200.fc25.x86_64, x86_64: installed
> 
> if the bbswitch package is installed, you may wish to try:
> /usr/sbin/dkms add -m bbswitch -v 0.8.0
> /usr/sbin/dkms build -m bbswitch -v 0.8.0
> /usr/sbin/dkms install -m bbswitch -v 0.8.0
> /sbin/modprobe bbswitch
> 
> To get clues as to why bbswitch is failing on this kernel...
> All checks completed successfully! NVIDIA driver appears to have compiled ok.
> 
> Documentation on bumblebee for RHEL / CentOS / fedora can be found at:
> https://www.linux.ncsu.edu/bumblebee/

**bumblebee-nvidia -f --debug**
> --debug mode selected.
> --force compile selected.
> Building NVIDIA video drivers: Creating directory NVIDIA-Linux-x86_64-375.39
> Verifying archive integrity... OK
> Uncompressing NVIDIA Accelerated Graphics Driver for Linux-x86_64 375.39.........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
> patching file kernel/common/inc/nv-mm.h
> patching file kernel/nvidia-drm/nvidia-drm-fence.c
> patching file kernel/nvidia-drm/nvidia-drm-gem.h
> patching file kernel/nvidia-drm/nvidia-drm-priv.h
> patching file kernel/nvidia/nv-pat.c
> 
> WARNING: Unable to determine the path to install the libglvnd EGL vendor library config files. Check that you have pkg-config and the libglvnd
>          development libraries installed, or specify a path with --glvnd-egl-config-path.
> 
> 
> WARNING: The symbolic link '/usr/lib/libGL.so.1' does not point to '/usr/lib64/nvidia-bumblebee/libGL.so.1' as is necessary for correct operation of
>          the NVIDIA Accelerated Graphics Driver for Linux-x86_64.  It is possible that `ldconfig` has created this incorrect symbolic link because
>          libGL.so.1.2.0;5846061a's "soname" conflicts with that of /usr/lib64/nvidia-bumblebee/libGL.so.1.  It is recommended that you remove or
>          rename the file 'libGL.so.1.2.0;5846061a' and create the necessary symbolic link by running `ln -sf /usr/lib64/nvidia-bumblebee/libGL.so.1
>          /usr/lib/libGL.so.1`.
> 
>                    [  OK  ]

**dmesg -H**
> [  +0.012033] bbswitch: version 0.8
> [  +0.000007] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
> [  +0.000009] bbswitch: Found discrete VGA device 0000:09:00.0: \_SB_.PCI0.RP05.PXSX
> [  +0.000018] bbswitch: failed to evaluate \_SB_.PCI0.RP05.PXSX._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0}
> [  +0.000029] bbswitch: failed to evaluate \_SB_.PCI0.RP05.PXSX._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4}
> [  +0.000012] ACPI Warning: \_SB.PCI0.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
> [  +0.000060] bbswitch: No suitable _DSM call found.

There are a similar issue #824 reported at Dec 9, 2016, but no solution.

JohnyGemityg, 2017-05-21:
Is it happening all the time ? I have similar problem but only after suspend. After restart it works ....

gforceg, 2017-07-02:
Hello,
I'm running Fedora 25, kernel 4.11.7-200
Finally decided to try and enable NVIDIA but BIOS only has the option of "switchable graphics" and "UMA graphics" so I tried to follow this tutorial: <https://fedoraproject.org/wiki/Bumblebee>
bumblebee / VirtualGL / primus all "seemed" to install fine but I can't run anything w/ the GPU:
```
$ optirun glxgears
[ 1504.789337] [ERROR]Cannot access secondary GPU - error: [XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied

[ 1504.789356] [ERROR]Aborting because fallback start is disabled.
```

```
$ primusrun glxgears
primus: fatal: Bumblebee daemon reported: error: [XORG] (EE) /dev/dri/card0: failed to set DRM interface version 1.4: Permission denied
```

```
# modprobe bbswitch
modprobe: ERROR: could not insert 'bbswitch': No such device
```

gforceg, 2017-07-02:
I'm using a [Lenovo Yoga 710 15isk](http://www3.lenovo.com/us/en/laptops/yoga/700-series/Lenovo-YOGA-710-15IKB/p/88YG7000772)

gforceg, 2017-07-02:
This solution worked for me.<https://github.com/Bumblebee-Project/Bumblebee/issues/875>

closed 2020-04-12 by hebertjulio