Bumblebee-Project/bbswitch #152: MSI GL62 6QD:

darknyaz, 2017-04-07:


Lekensteyn, 2017-04-07:
If you don't use Bumblebee, do you have problems? What if you use the nouveau driver instead of the proprietary driver? And note that in both cases you might need the `acpi_osi` workaround because of #140 and https://github.com/Bumblebee-Project/Bumblebee/issues/764

darknyaz, 2017-04-09:
It seems that I have problem with fan in all cases. I don't remember what was when I've just installed OS, but to successfully start sddm I had to add `nouveau.modeset=0`. nouveau also report such error:
> nouveau 0000:01:00.0: Refused to change power state, currently in D3
I just didn't try nouveau + bumblebee combination, should I?
I also don't need to add `acpi_osi=! "acpi_osi=Windows 2009"` options when using nouveau.

darknyaz, 2017-04-22:
Turns out that this problem exists on Windows too and it's a known problem of this laptop.

closed 2017-04-22 by darknyaz

darknyaz, 2017-04-23:
Workaround for my MSI GL62 6QD: update EC to 16J6EMS1.104 version and install bumblebee-nvidia package with all dependences.