Bumblebee-Project/bbswitch #155: Can't turn off dGPU after running optirun

gautamr95, 2017-04-18:
I am running Ubuntu 16.04 dual booted with Windows 10 on a Thinkpad T460p with an Nvidia 940mx card. I have it set so that the dGPU is off when I boot up the laptop so that when I run `sudo cat /proc/acpi/bbswitch` I get `0000:02:00.0 OFF`. However once I run for instance `optirun glxspheres` and check the status of the card `0000:02:00.0 ON`.

I am using the proprietary nvidia-375 driver and have blacklisted `nvidia-375` to my `/etc/modprobe.d/bumblebee.conf`as well as blacklisted `nvidia` in my `/etc/modprobe.d/blacklist.conf`. In my `/etc/modules` I have `i915` and `bbswitch load_state=0`. I also tried the fix using grub [here](https://github.com/Bumblebee-Project/bbswitch/issues/96#issuecomment-257187040) but that just made the dGPU turn on upon booting. If I need to post any actual files let me know and I can upload gists.

bluca, 2017-04-18:
Please install bumblebee-nvidia and bumblebee from the testing PPA repository and try again:

https://launchpad.net/~bumblebee/+archive/ubuntu/testing

gautamr95, 2017-04-18:
I installed the two from testing and it starts up just fine with it off and then I can run optirun programs fine. But once I try to `sudo tee /proc/acpi/bbswitch <<<OFF` it fails and my dmesg output is

`[  478.596657] nvidia: probe of 0000:02:00.0 failed with error -1`
`[  478.596711] nvidia-nvlink: Nvlink Core is being initialized, major device number 244`
`[  478.596808] NVRM: The NVIDIA probe routine failed for 1 device(s).`
`[  478.596810] NVRM: None of the NVIDIA graphics adapters were initialized!`
`[  478.596813] nvidia-nvlink: Unregistered the Nvlink Core, major device number 244`
`[  478.872996] nvidia 0000:02:00.0: enabling device (0000 -> 0003)`
`[  478.873088] NVRM: The NVIDIA GPU 0000:02:00.0 (PCI ID: 10de:134d)`
`               NVRM: installed in this system is not supported by the 375.39`
`               NVRM: NVIDIA Linux driver release.  Please see 'Appendix`
`               NVRM: A - Supported NVIDIA GPU Products' in this release's`
`               NVRM: README, available on the Linux driver download page`
`               NVRM: at www.nvidia.com.`

And then when I try to optirun something again it cannot load GPU driver. When I installed the new bumblebee it asked whether I wanted to overwrite my `/etc/bumblebee/bumblebee.conf` and I did and then manually changed it to use nvidia-375 instead of current and forced bbswitch, primus, and nvidia.

bluca, 2017-04-18:
What are these set to in /etc/bumblebee/bumblebee.conf ?

KernelDriver=
LibraryPath=
XorgModulePath=

gautamr95, 2017-04-18:
The nvidia section of my `bumblebee.conf` is
```
[driver-nvidia]
  # Module name to load, defaults to Driver if empty or unset
  KernelDriver=nvidia
  PMMethod=bbswitch
  # colon-separated path to the nvidia libraries
  LibraryPath=/usr/lib/nvidia-378:/usr/lib32/nvidia-378
  # comma-separated path of the directory containing nvidia_drv.so and the
  # default Xorg modules path
  XorgModulePath=/usr/lib/nvidia-378/xorg,/usr/lib/xorg/modules
  XorgConfFile=/etc/bumblebee/xorg.conf.nvidia
```

I actually just updated to the nvidia-378 driver and tested with that since according to http://www.nvidia.com/object/unix.html the 375 driver explicitly supports the 940M and the 378 driver explicitly supports the 940M and 940MX, which is what I have.

If I run `optirun glxspheres` for the first time it works correctly. On the second time I get `[  527.402347] [ERROR]Cannot access secondary GPU - error: Could not load GPU driver`. And subsequent tries I get 
```
Xlib:  extension "GLX" missing on display ":8".
ERROR in line 620:
Could not obtain RGB visual with requested properties
```

bluca, 2017-04-18:
I don't have my Ubuntu laptop here so I can't check, shouldn't it be

KernelDriver=nvidia-378

Could you please try? You'll want to reboot to reset the hardware

gautamr95, 2017-04-18:
Seems to result in the same errors as before. Thanks for your help so far though!

bluca, 2017-04-18:
Where does the nvidia-378 come from? Don't see it in the Ubuntu repos, is it from a PPA?

gautamr95, 2017-04-18:
Yea its from some PPA

bluca, 2017-04-18:
After a fresh reboot, if you run optirun, what's in the journal?

gautamr95, 2017-04-18:
Assuming `journalctl`:
Before optirun:
`journalctl | grep nvidia`

```
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia: loading out-of-tree module taints kernel.
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia: module license 'NVIDIA' taints kernel.
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia: module verification failed: signature and/or required key missing - tainting kernel
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia 0000:02:00.0: enabling device (0004 -> 0007)
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia-nvlink: Nvlink Core is being initialized, major device number 244
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  375.39  Tue Jan 31 19:41:48 PST 2017
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: [drm] [nvidia-drm] [GPU ID 0x00000200] Loading driver
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: bbswitch: device 0000:02:00.0 is in use by driver 'nvidia', refusing OFF
Apr 18 15:52:18 gautam-ThinkPad-T460p kernel: nvidia-uvm: Loaded the UVM driver in 8 mode, major device number 242
Apr 18 15:52:19 gautam-ThinkPad-T460p systemd-udevd[369]: Process '/usr/bin/nvidia-smi' failed with exit code 12.
Apr 18 15:52:30 gautam-ThinkPad-T460p systemd[984]: nvidia-persistenced.service: Failed at step EXEC spawning /usr/bin/nvidia-persistenced: No such file or directory
Apr 18 15:52:30 gautam-ThinkPad-T460p kernel: [drm] [nvidia-drm] [GPU ID 0x00000200] Unloading driver
Apr 18 15:52:30 gautam-ThinkPad-T460p kernel: nvidia-modeset: Unloading
Apr 18 15:52:30 gautam-ThinkPad-T460p kernel: nvidia-uvm: Unloaded the UVM driver in 8 mode
Apr 18 15:52:30 gautam-ThinkPad-T460p kernel: nvidia-nvlink: Unregistered the Nvlink Core, major device number 244
Apr 18 15:52:31 gautam-ThinkPad-T460p systemd[1]: nvidia-persistenced.service: Control process exited, code=exited status=203
Apr 18 15:52:31 gautam-ThinkPad-T460p systemd[1]: nvidia-persistenced.service: Unit entered failed state.
Apr 18 15:52:31 gautam-ThinkPad-T460p systemd[1]: nvidia-persistenced.service: Failed with result 'exit-code'.
```
`journalctl | grep bbswitch`

```
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: bbswitch: version 0.8
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: bbswitch: Found discrete VGA device 0000:02:00.0: \_SB_.PCI0.PEG2.PEGP
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: bbswitch: detected an Optimus _DSM function
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: bbswitch: device 0000:02:00.0 is in use by driver 'nvidia', refusing OFF
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: bbswitch: Succesfully loaded. Discrete card 0000:02:00.0 is on
Apr 18 15:52:15 gautam-ThinkPad-T460p systemd-modules-load[288]: Inserted module 'bbswitch'
Apr 18 15:52:31 gautam-ThinkPad-T460p kernel: bbswitch: disabling discrete graphics
```
After optirun:

`journalctl | grep nvidia`

```
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia: loading out-of-tree module taints kernel.
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia: module license 'NVIDIA' taints kernel.
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia: module verification failed: signature and/or required key missing - tainting kernel
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia 0000:02:00.0: enabling device (0004 -> 0007)
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia-nvlink: Nvlink Core is being initialized, major device number 244
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  375.39  Tue Jan 31 19:41:48 PST 2017
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: [drm] [nvidia-drm] [GPU ID 0x00000200] Loading driver
Apr 18 15:52:15 gautam-ThinkPad-T460p kernel: bbswitch: device 0000:02:00.0 is in use by driver 'nvidia', refusing OFF
Apr 18 15:52:18 gautam-ThinkPad-T460p kernel: nvidia-uvm: Loaded the UVM driver in 8 mode, major device number 242
Apr 18 15:52:19 gautam-ThinkPad-T460p systemd-udevd[369]: Process '/usr/bin/nvidia-smi' failed with exit code 12.
Apr 18 15:52:30 gautam-ThinkPad-T460p systemd[984]: nvidia-persistenced.service: Failed at step EXEC spawning /usr/bin/nvidia-persistenced: No such file or directory
Apr 18 15:52:30 gautam-ThinkPad-T460p kernel: [drm] [nvidia-drm] [GPU ID 0x00000200] Unloading driver
Apr 18 15:52:30 gautam-ThinkPad-T460p kernel: nvidia-modeset: Unloading
Apr 18 15:52:30 gautam-ThinkPad-T460p kernel: nvidia-uvm: Unloaded the UVM driver in 8 mode
Apr 18 15:52:30 gautam-ThinkPad-T460p kernel: nvidia-nvlink: Unregistered the Nvlink Core, major device number 244
Apr 18 15:52:31 gautam-ThinkPad-T460p systemd[1]: nvidia-persistenced.service: Control process exited, code=exited status=203
Apr 18 15:52:31 gautam-ThinkPad-T460p systemd[1]: nvidia-persistenced.service: Unit entered failed state.
Apr 18 15:52:31 gautam-ThinkPad-T460p systemd[1]: nvidia-persistenced.service: Failed with result 'exit-code'.
Apr 18 15:56:48 gautam-ThinkPad-T460p kernel: nvidia-nvlink: Nvlink Core is being initialized, major device number 244
Apr 18 15:56:48 gautam-ThinkPad-T460p systemd[3513]: nvidia-persistenced.service: Failed at step EXEC spawning /usr/bin/nvidia-persistenced: No such file or directory
Apr 18 15:56:48 gautam-ThinkPad-T460p systemd[1]: nvidia-persistenced.service: Control process exited, code=exited status=203
[... cut ...]
Apr 18 15:52:15 gautam-ThinkPad-T460p systemd-modules-load[288]: Inserted module 'bbswitch'
Apr 18 15:52:31 gautam-ThinkPad-T460p kernel: bbswitch: disabling discrete graphics
Apr 18 15:56:47 gautam-ThinkPad-T460p kernel: bbswitch: enabling discrete graphics
Apr 18 15:56:49 gautam-ThinkPad-T460p kernel: bbswitch: disabling discrete graphics
Apr 18 15:56:55 gautam-ThinkPad-T460p kernel: bbswitch: enabling discrete graphics
Apr 18 15:57:00 gautam-ThinkPad-T460p kernel: bbswitch: disabling discrete graphics
```
Do I need to grep for anything else?

gautamr95, 2017-04-18:
Ok so I have no idea what happened but things were working as they were supposed to for a little bit. I was able to run optirun then checked and then closed glxspheres and checked `/etc/acpi/bbswitch` and saw it was OFF. I was able to do this a few times correctly and then when I checked `/etc/acpi/bbswitch` while running `optirun glxspheres` and saw it was ON, I then got the same error with not being able to access GPU. No idea what made it work those first few times.

gautamr95, 2017-04-19:
I looked at [this](https://github.com/Bumblebee-Project/bbswitch/issues/96#issuecomment-257187040) comment and tried adding that line to my `/etc/default/grub` and then did a `grub-mkconfig -o /boot/grub/grub.cfg`. Things tentatively seem to be working as intended. I'll keep testing and will update in a little bit.

gautamr95, 2017-04-20:
Well I had a chance to reinstall everything today so I did `sudo apt-get purge '.*nvidia.*' '.*bumblebee.*' '.*primus.*'` and then installed `nvidia-prime`, `nvidia-375`, `bumblebee bumblebee-nvidia` in that order. I then went through all the configuration files and corrected them such as replacing `nvidia-current` with `nvidia-375` and forcing `bbswitch` and `nvidia` as well as making sure the blacklist was correct. I then made [this](https://wiki.archlinux.org/index.php/Bumblebee#.2Fdev.2Fdri.2Fcard0:_failed_to_set_DRM_interface_version_1.4:_Permission_denied) change to the `xorg.conf.nvidia` file. After a reboot I am able to run an optirun program like glxspheres and then close it and see that my GPU is in fact off. I haven't run into any issues yet after starting and stopping optirun many times. I really hope this issue doesn't show up again. Thanks @bluca for all your help!

gautamr95, 2017-04-20:
Ugh after rebooting everything went back to how it was before.

gautamr95, 2017-04-22:
Hopefully final update. Added `pcie_port_pm=off` to my `/etc/default/grub` so that it now has the line `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash pcie_port_pm=off"`. I then ran `sudo update-grub` and rebooted. I have not run into any issues since then and have been able to play csgo and dota just fine.

closed 2017-04-22 by gautamr95

bluca, 2017-04-22:
Great, thanks for reporting back