Bumblebee-Project/bbswitch #156: echo "OFF"  > /proc/acpi/bbswitch results in no power change and error in dmesg

jdevelop, 2017-05-16:
```
[  172.680994] bbswitch: disabling discrete graphics
[  172.681014] ACPI Warning: \_SB.PCI0.PEG.VID._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[  172.681761] ------------[ cut here ]------------
[  172.681772] WARNING: CPU: 3 PID: 8221 at drivers/pci/pci.c:1641 pci_disable_device+0xa8/0xd0
[  172.681774] pci 0000:01:00.0: disabling already-disabled device
[  172.681776] Modules linked in: fuse ctr ccm uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media joydev mousedev arc4 snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic iTCO_wdt mei_wdt iTCO_vendor_support intel_rapl iwlmvm x86_pkg_temp_thermal intel_powerclamp mac80211 coretemp kvm_intel kvm irqbypass int
el_cstate iwlwifi intel_rapl_perf psmouse pcspkr input_leds cfg80211 i2c_i801 snd_hda_intel snd_hda_codec rtsx_pci_ms memstick snd_hda_core snd_hwdep ie31200_edac e1000e snd_pcm lpc_ich snd_timer mei_me mei ptp edac_core shpchp pps_core thinkpad_acpi wmi thermal nvram snd soundcore rfkill fjes led_class ac battery evdev tpm_tis sch_fq_codel tpm_tis_core ma
c_hid tpm vboxnetflt(O) vboxnetadp(O) pci_stub vboxpci(O) vboxdrv(O) sg acpi_call(O) ip_tables
[  172.681855]  x_tables ext4 crc16 jbd2 fscrypto mbcache algif_skcipher af_alg dm_crypt dm_mod sr_mod cdrom sd_mod rtsx_pci_sdmmc mmc_core serio_raw atkbd libps2 crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel ahci pcbc libahci aesni_intel aes_x86_64 crypto_simd glue_helper libata cryptd xhci_pci ehci_pci xhci_hcd ehci_hcd scsi_mod rtsx_pci
 usbcore usb_common i8042 serio bbswitch(O) i915 video button intel_gtt i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm
[  172.681913] CPU: 3 PID: 8221 Comm: bash Tainted: G           O    4.10.13-1-ARCH #1
[  172.681914] Hardware name: LENOVO 20BECTO1WW/20BECTO1WW, BIOS GMET70WW (2.18 ) 03/05/2015
[  172.681916] Call Trace:
[  172.681927]  dump_stack+0x63/0x83
[  172.681932]  __warn+0xcb/0xf0
[  172.681937]  warn_slowpath_fmt+0x5f/0x80
[  172.681941]  ? __pci_set_master+0x3b/0xf0
[  172.681944]  pci_disable_device+0xa8/0xd0
[  172.681951]  bbswitch_off+0xad/0x240 [bbswitch]
[  172.681955]  bbswitch_proc_write+0xb0/0xc7 [bbswitch]
[  172.681961]  proc_reg_write+0x42/0x70
[  172.681965]  __vfs_write+0x37/0x140
[  172.681971]  ? filp_close+0x56/0x80
[  172.681974]  vfs_write+0xb6/0x1a0
[  172.681977]  SyS_write+0x55/0xc0
[  172.681983]  ? __close_fd+0x8f/0xb0
[  172.681987]  entry_SYSCALL_64_fastpath+0x1a/0xa9
[  172.681990] RIP: 0033:0x7fe0b0d9b3c0
[  172.681992] RSP: 002b:00007ffd2cb2d1a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
[  172.681996] RAX: ffffffffffffffda RBX: 00007fe0b105bb38 RCX: 00007fe0b0d9b3c0
[  172.681997] RDX: 0000000000000004 RSI: 000000000257c2d0 RDI: 0000000000000001
[  172.681999] RBP: 000000000000270f R08: 00007fe0b105d740 R09: 00007fe0b1693b40
[  172.682001] R10: 0000000000000000 R11: 0000000000000246 R12: 00007fe0b105bb38
[  172.682002] R13: 0000000000000fa0 R14: 00000000025983e0 R15: 00007fe0b105bae0
[  172.682005] ---[ end trace 1a19acee2f4f7984 ]---
[  172.802476] thinkpad_acpi: EC reports that Thermal Table has changed
```

Kernel boot line:

`GRUB_CMDLINE_LINUX_DEFAULT="quiet acpi_osi='!Windows 2013'"`

Kernel version:

`Linux devbox 4.10.13-1-ARCH #1 SMP PREEMPT Thu Apr 27 12:15:09 CEST 2017 x86_64 GNU/Linux`

Hardware: ThinkPad T540P

jdevelop, 2017-05-17:
Resolved in #96

closed 2017-05-17 by jdevelop

cnlohr, 2020-01-23:
I have exactly the same error (letter-for-letter), except on a PowerSpec 1520 with a GeForce RTX 2070. adding ```"acpi_osi=!Windows 2013"``` does not resolve the issue for me.  I'm on kernel ```Linux cnlohr-1520 5.3.0-26-generic #28~18.04.1-Ubuntu SMP Wed Dec 18 16:40:14 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux```  Any ideas?