Bumblebee-Project/bbswitch #160: No suitable _DSM call found

runningyz, 2017-09-07:
There are some problems when I start up my ubuntu 16.04 ,I can't enter into GUI,the screen keeps printing 'bbswitch no suitable _DSM call found',how to fix this.

[ 1843.188259] bbswitch: No suitable _DSM call found.
[ 1843.448777] bbswitch: version 0.8
[ 1843.448782] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[ 1843.448787] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[ 1843.448798] bbswitch: failed to evaluate \_SB_.PCI0.PEG0.PEGP._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[ 1843.448803] bbswitch: failed to evaluate \_SB_.PCI0.PEG0.PEGP._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[ 1843.448808] ACPI Warning: \_SB_.PCI0.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150930/nsarguments-95)

Lekensteyn, 2017-09-20:
Please see the README, you need to provide more details like kernel version, lspci, laptop model, etc.

mirh, 2017-09-23:
This smells like [this](https://github.com/Bumblebee-Project/bbswitch/commit/ee0591b9b8d0a42f60eae6f9c644e6a1be4323d6#commitcomment-3857600)

Lekensteyn, 2017-09-23:
The last message can be ignored for the above  reason, but the other "failed to evaluate" warnings are more problematic.

LukeC8, 2017-10-09:
Same bug here.

* [Hardware info] https://bugs.launchpad.net/lpbugreporter/+bug/752542/comments/827
* [dmesg output] https://gist.github.com/LukeC8/c8f9e33b8b418449d4b5ebf344b6ed0b
* [acpi_dump_info output]  https://gist.github.com/LukeC8/b0e0859fa8a81bb22b86cafbdffe3da6
* [uname -a] `Linux lucas 4.4.0-96-generic #119~14.04.1-Ubuntu SMP Wed Sep 13 08:40:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
`
if you need more information about logs files I'll be ready to help.

Lekensteyn, 2017-10-11:
@LukeC8 It should have invoked `\_SB.PCI0.RP01.PEGP._DSM`. It probably picked `\_SB.PCI0.RP01.D055` because that is device 0, function 0 under PCI device 00:1c.0. I think this was a problem that got fixed in later kernel versions, can you try upgrading your kernel? Ubuntu 14.04 is also the previous LTS, consider upgrading to 16.04.

rdjondo, 2017-10-29:
I have the same issue with 16.04

rdjondo, 2017-10-29:
I have the same issue with 16.04 since I upgraded to Linux 4.4.0-97.
Linux 4.4.0-93 seems fine but booting on Linux 4.4.0-97 booting fails on my machine.

dominiksp, 2017-10-31:
I have the same issue as  runningyz  and rdjondo. Since upgrading on Ubuntu 16.04 to Linux 4.4.0-98, I cannot boot into GUI and get the error message 'bbswitch no suitable _DSM call found'. Booting  4.4.0-97 does not work either. However, booting  4.4.0-83 does work.
My hardware is an Acer Aspire V5-573G with NVIDIA GeForce GTX 850M.

geoffmelnick, 2017-11-06:
I have the same issue,using GT1060 and Ubuntu 16.04.3  lts

geoffmelnick, 2017-11-07:
Sorry, lost most of that last message.  Installed Ubuntu 16.04.03 and it worked fine.  Then I looked for a driver for the new GPU, which is a GTX 1060.  I used the one suggested by the software installer, and then all reboots failed, just getting a screen full of the error message from BBSwitch.  The message log indicated that the GPU had been found but then it couldn't find a corresponding display device so it deinstalled it. 
Any suggestions would be gratefully received.  I can provide more information but only to a limited extent, as the thing won't boot.

tobixx, 2017-11-11:
Same here. Last working Ubuntu 16.04 kernel 4.4.0.93 with upstart.
Acer Aspire V5-573G,

mbacilieri, 2017-11-11:
Same here, on Acer Aspire E5-771G-51HQ with nvidia GeForce 820M.
Kernel 4.4.0-98 and Ubuntu 16.04: when I try to start in nvidia mode, a list of error "bbswitch: no suitable _DKMS call found" appears then the system start in graphic recovery mode.
The series 4.x seems to be very problematic on Acers also for suspend; this is a further issue.
For now I'm stuck on kernel 3.19.0-71 that seems to work good.

cesarizu, 2017-11-11:
I'm having a similar problem: https://bugs.launchpad.net/lpbugreporter/+bug/752542/comments/829

shuklaayush, 2017-11-12:
Updating to the HWE kernel (4.10) seems to resolve the  issue for me. Try running:
'sudo apt install --install-recommends linux-image-generic-hwe-16.04 xserver-xorg-hwe-16.04'

shuklaayush, 2017-11-12:
Updating to the HWE kernel (4.10) seems to resolve the  issue for me. Try running:
`sudo apt install --install-recommends linux-image-generic-hwe-16.04 xserver-xorg-hwe-16.04`

mx1010, 2017-11-22:
Same thing, but only after install drivers for 1070 MSI Aero GPU. I've tried the PPA drivers and Nvidia's latest.

shuklaayush, 2017-11-22:
@mx1010 Did you try updating the kernel as I have described? It fixed the issue for me.

mx1010, 2017-11-24:
@shuklaayush Yeah, I was still seeing the message in 4.10, depending on which TTY I chose. It was odd bc I could goto F7 and get Unity with no issue but when I went into F1-F5 it'd spam console with "bbswitch: No suitable _DSM call found". 

I finally ended up just doing a fresh install on a different storage device, reseating all devices, and it seems to have disappeared, so it was possibly just a bad config setting, or something else ¯\_(ツ)_/¯

mbacilieri, 2017-11-24:
Well, without changing anything, 4.4.0-101 seems to work properly in my Acer E5-771G.

mbacilieri, 2017-12-10:
More details: now I'm running kernel 4.4-0.103, and in dmesg I see:

```
[   20.682682] bbswitch: failed to evaluate \_SB_.PCI0.RP05.PXSX._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[   20.682689] bbswitch: failed to evaluate \_SB_.PCI0.RP05.PXSX._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[   20.682697] ACPI Warning: \_SB_.PCI0.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150930/nsarguments-95)
[   20.682739] bbswitch: No suitable _DSM call found.
```

Listing the provider for xrandr, there is no nvidia (it should be):
```
$ xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel
```

The nvidia card is recognized in pci device list:
```
$ lspci
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 0b)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 0b)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI #0 (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 1 (rev e4)
00:1c.2 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 3 (rev e4)
00:1c.3 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 4 (rev e4)
00:1c.4 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 5 (rev e4)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 8 Series SATA Controller 1 [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
01:00.0 3D controller: NVIDIA Corporation GF117M [GeForce 610M/710M/810M/820M / GT 620M/625M/630M/720M] (rev a1)
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
05:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8723BE PCIe Wireless Network Adapter
```

If I start in "intel" mode (prime-select intel) I can boot and system works, although I can't use 3d card at all:

```
$ optirun glxgears
[  775.358917] [ERROR]Cannot access secondary GPU - error: [XORG] (EE) Failed to load module "mouse" (module does not exist, 0)
[  775.358962] [ERROR]Aborting because fallback start is disabled.
```

If I run "glxgears" in "intel" mode, it runs. Obviously it's not using the nvidia card.
If I try to boot/login after "prime-select nvidia" (in nvidia mode) the system hangs in console repeating "No suitable _DSM call found.".

I tryed with nvidia driver 384 and 340, with no difference: strangely, for some reason 340 seems the only  "officially recognized" by Ubuntu, for in "software and updates / additional drivers" the only nvidia option is "340". Using the 384, Ubuntu think I'm using the nouveau: that _should be_ impossibile for I have blacklisted nouveau everywhere, starting from grub start line:
```
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash modprobe.blacklist=nouveau"
```
added to the blacklisting in nvidia modprobe configuration files.
Despite this, Xorg after fails to load nvidia module, try to load nouveau (from Xorg.0.log):
```
[    36.576] (II) LoadModule: "nvidia"
[    36.576] (WW) Warning, couldn't open module nvidia
[    36.576] (II) UnloadModule: "nvidia"
[    36.576] (II) Unloading nvidia
[    36.576] (EE) Failed to load module "nvidia" (module does not exist, 0)
[    36.576] (II) LoadModule: "nouveau"
[    36.576] (II) Loading /usr/lib/xorg/modules/drivers/nouveau_drv.so
[    36.960] (II) Module nouveau: vendor="X.Org Foundation"
[    36.960]    compiled for 1.18.1, module version = 1.0.12
[    36.960]    Module class: X.Org Video Driver
[    36.960]    ABI class: X.Org Video Driver, version 20.0
```

It's a very strange behaviour: kernel 3.19 - with the **same** configuration (I mean: just choosing the kernel at grub menu) - works properly and recognize the nvidia discrete card.

mirh, 2017-12-10:
> It's a very strange behaviour: kernel 3.19 - with the same configuration works

There's nothing strange about some tens of thousands of lines of code more or less producing.. an *impact*. 
And bugs/logs should just be reported against latest kernel (given it's the only 'point' where you could pretend a fix to be added)
Otherwise discussing about older releases may still make sense, if somebody wants something that *just works*™... 
But there's not point in pretending something as massive as acpi (what causes these problems) to be reworked there.

mbacilieri, 2017-12-11:
> And bugs/logs should just be reported against latest kernel (given it's the only 'point' where you could pretend a fix to be added)
I understand the point and I agree such policy, although the involved kernel is the 4.4.0-103 which is for now the latest officially shipped to Ubuntu 16.04 LTS. So it's difficult to me to consider it an obsolete kernel.

mirh, 2017-12-11:
You should consider ubuntu as obsolete 😄

mbacilieri, 2018-01-11:
No way, also the last update of yesterday (kernel 4.4.0-109) didn't change the issue.
The discrete card is still unaccessible due to the bbswitch error.
It begins to be a serious issue, for the kernel 4.4.0-109 patches the Meltdown and Spectre bugs, and being forced to use an old kernel 3.19 means to run unpatched.

anneb, 2018-01-12:
For me kernel 4.4.0-83 does not have this issue, the next version I have is 4.4.0-104 which has the issue. Version 4.4.0-109 still has the issue.

My graphics card:
sudo lspci -v | grep VGA -A 2
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 5500 (rev 09) (prog-if 00 [VGA controller])
	DeviceName:  Onboard IGD
	Subsystem: Dell Broadwell-U Integrated Graphics

dominiksp, 2018-01-13:
I can confirm that the issue still exists with kernel version 4.4.0-109.

I would really appreciate a solution.

Summary: 
4.4.0-98 does not work
4.4.0-97 does not work
4.4.0-83 still works
My output of dmesg | grep -C 10 bbswitch:
[    3.369457] audit: type=1400 audit(1515833541.740:10): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/snapd/snap-confine" pid=824 comm="apparmor_parser"
[    3.406481] usb 2-5.3: new low-speed USB device number 8 using xhci_hcd
[    3.438555] cgroup: new mount options do not match the existing superblock, will be ignored
[    3.502407] usb 2-5.3: New USB device found, idVendor=10d5, idProduct=000d
[    3.502412] usb 2-5.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.502415] usb 2-5.3: Product: HA2-A3
[    3.502417] usb 2-5.3: Manufacturer: No brand
[    3.502419] usb 2-5.3: SerialNumber: 0\xffffffc2\xffffff92
[    3.502564] usb 2-5.3: ep 0x2 - rounding interval to 64 microframes, ep desc says 80 microframes
[    3.535528] hidraw: raw HID events driver (C) Jiri Kosina
[    3.535599] bbswitch: version 0.8
[    3.535604] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[    3.535611] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.RP05.PEGP
[    3.535620] ACPI Warning: \_SB_.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150930/nsarguments-95)
[    3.535695] bbswitch: detected an Optimus _DSM function
[    3.535703] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[    3.560040] usbhid 2-5.3:1.1: couldn't find an input interrupt endpoint
[    3.560061] usbcore: registered new interface driver usbhid
[    3.560063] usbhid: USB HID core driver
[    3.563334] input: Microsoft Microsoft® Nano Transceiver v2.1 as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.1/2-5.1:1.0/0003:045E:07A5.0001/input/input16
[    3.618614] hid-generic 0003:045E:07A5.0001: input,hidraw0: USB HID v1.11 Keyboard [Microsoft Microsoft® Nano Transceiver v2.1] on usb-0000:00:14.0-5.1/input0
[    3.618827] input: Microsoft Microsoft® Nano Transceiver v2.1 as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.1/2-5.1:1.1/0003:045E:07A5.0002/input/input17
[    3.618932] hid-generic 0003:045E:07A5.0002: input,hidraw1: USB HID v1.11 Mouse [Microsoft Microsoft® Nano Transceiver v2.1] on usb-0000:00:14.0-5.1/input1
[    3.631048] input: Microsoft Microsoft® Nano Transceiver v2.1 as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.1/2-5.1:1.2/0003:045E:07A5.0003/input/input18
[    3.709575] NET: Registered protocol family 4
[    3.715732] NET: Registered protocol family 5

Last running kernel version
uname -a
4.4.0-83-generic #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Distribution and version:
Ubuntu 16.04.3 LTS

Xorg: unable to identify xorg version

TomLisankie, 2018-01-14:
My issue is similar. I will be using my computer (specs [here](https://pcpartpicker.com/user/tomlis/saved/#view=b6Kd6h)) for no more than 10 minutes. Then all graphics on the screen completely freeze for about 30 seconds to a minute. My screen then completely goes black except for a single underscore in the top left corner. A couple of lines of text quickly flash and then this black screen starts outputting `bbswitch: No suitable _DSM call found.` repeatedly.

Upgrading the kernel did not help either.

mbacilieri, 2018-01-17:
**SOLVED**
The issue is due to the upgrade to 16.04.3 that requires a 4.13 kernel "hwe".
When updating the 16.04 at a certain point you must install che "hwe" kernel.
First make a full update:
# apt-get update && apt-get full-upgrade
then install hwe kernel and hwe xorg:
# sudo apt-get install --install-recommends linux-generic-hwe-16.04 xserver-xorg-hwe-16.04
Please pay attention for it will **uninstall** the old xorg and, for nvidia users, the bumblebee/primusrun system.
If you have nvidia proprietary drivers, do not forget a:
# dpkg-reconfigure nvidia-384
(or whatever version you have), just to ensure the dkms-modules will be configured against new xorg/kernel version.
After all, you will reboot with the 4.13 kernel that works.

TomLisankie, 2018-01-18:
That worked! Thank you!

dominiksp, 2018-01-19:
Thanks a lot, this seems to have solved the issue. I am now booting into 
4.13.0-26-generic #29~16.04.2-Ubuntu SMP Tue Jan 9 22:00:44 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
By the way, for some of the commands given, I needed an additional 'sudo'.

mbacilieri, 2018-02-03:
Anyway, it wasn't correct that kernel hwe works. After a while, I realized that:
- shutdown doesn't work, computer freezes instead to poweroff.
- doesn't work neither suspend (it suspends but never wake up).
- touchpad doesn't work.
- bbswitch removed, so I cannot "optirun" a program to use 3D when in "intel" (power save) mode.
At the end, I removed all hwe configuration e reverted to kernel 3.19. Quite worried, for it's EOL: anyway it looks like there is no 4.x kernel suitable.

TomLisankie, 2018-02-15:
@mbacilieri I'm getting these same issues all over again. What a shame it's not 4.x suitable. But you're running 16.04 with kernel 3.19?

mbacilieri, 2018-02-15:
@TomLisankie I ran 16.04 with 3.19 until last days, when I relized I couldn't be stuck for more.
So I began to test all combination of kernel, parameters and drivers I could.
Yesterday I found a working setup with kernel 4.13 "hwe", with xorg "hwe".
I removed all nvidia drivers, using nouveau. The trouble looked in the interaction between nvidia (also newest 390) and kernel of 4.x series.
Then I added "bbswitch" in /etc/modules, which for unknown reason it's not automatically loaded at boot time. bbswitch is needed to switch **on** the discrete card at shutdown, otherwise system hangs instead to power off.
With this configuration, all looks ok, both 3D with DRI_PRIME=1 and power management with suspend/shutdown. It's not the best performance, but at least it works properly.

Trevortds, 2018-05-11:
This issue may have reoccurred in the release version of 18.04. The HWE kernel no longer fixes it. See #169

whatisor, 2018-09-10:
In my case, power of graphic card is problematic.