Bumblebee-Project/bbswitch #161: Can't install on my lenovo Y470, running 4.12.0-kali2-amd64

MorosithIi, 2017-10-06:
I met an error when tried to install acpi-handle-hack from branch hack-lenovo,
here is the details:

dkms install -m acpi-handle-hack -v 0.0.2

Kernel preparation unnecessary for this kernel.  Skipping...
Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.12.0-kali2-amd64 KVERSION=4.12.0-kali2-amd64 modname=acpi-handle-hack...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.12.0-kali2-amd64 (x86_64)
Consult /var/lib/dkms/acpi-handle-hack/0.0.2/build/make.log for more information.

cat /var/lib/dkms/acpi-handle-hack/0.0.2/build/make.log

DKMS make.log for acpi-handle-hack-0.0.2 for kernel 4.12.0-kali2-amd64 (x86_64)
Fri Oct  6 21:43:08 CST 2017
make -C /lib/modules/4.12.0-kali2-amd64/build M="$(pwd)" modules
make[1]: Entering directory '/usr/src/linux-headers-4.12.0-kali2-amd64'
  CC [M]  /var/lib/dkms/acpi-handle-hack/0.0.2/build/acpi-handle-hack.o
/var/lib/dkms/acpi-handle-hack/0.0.2/build/acpi-handle-hack.c: In function ‘dev_set_acpi_handle’:
/var/lib/dkms/acpi-handle-hack/0.0.2/build/acpi-handle-hack.c:68:20: error: ‘struct dev_archdata’ has no member named ‘acpi_handle’
  pdev->dev.archdata.acpi_handle = handle;
                    ^
/usr/src/linux-headers-4.12.0-kali2-common/scripts/Makefile.build:313: recipe for target '/var/lib/dkms/acpi-handle-hack/0.0.2/build/acpi-handle-hack.o' failed
make[4]: *** [/var/lib/dkms/acpi-handle-hack/0.0.2/build/acpi-handle-hack.o] Error 1
/usr/src/linux-headers-4.12.0-kali2-common/Makefile:1532: recipe for target '_module_/var/lib/dkms/acpi-handle-hack/0.0.2/build' failed
make[3]: *** [_module_/var/lib/dkms/acpi-handle-hack/0.0.2/build] Error 2
Makefile:152: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.12.0-kali2-amd64'
Makefile:13: recipe for target 'default' failed
make: *** [default] Error 2

what should I do to fix this?

Lekensteyn, 2017-10-11:
The hack is not needed for recent kernel versions, what makes you think that you need it?

MorosithIi, 2017-10-14:
Because when I type  lspci  | grep -i vga
It just printed this:
00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
I search for it and it leads me to here

Lekensteyn, 2017-10-14:
@MorosithIi don't see any nvidia GPU. Try `lspci -nn | grep -E "3D|VGA"` to find other possible GPUs.
Consider attaching the output of `dmesg > dmesg.txt` too for inspection.

What are you trying to do?

MorosithIi, 2017-10-26:
I just wanted to install Cuda, but it seems that the hardware doesn't work anymore

closed 2018-04-23 by MorosithIi