Bumblebee-Project/bbswitch #162: System freeze after login on Debian 9 with discrete cards disabled

niltonvasques, 2017-11-06:
I can avoid this freeze doing `tee /proc/acpi/bbswitch <<<ON` before login. 

There is a way to always keep discrete card enabled during login?

```
Distributor ID: Debian
Description:    Debian GNU/Linux 9.1 (stretch)
Release:        9.1
Codename:       stretch
GNOME Shell 3.21.91
```

Find below the screenshot logs that appear during the freezes on login:

[image]

Queuecumber, 2017-11-09:
I also have this issue on ubuntu 17.10 (and I had it on 17.04 as well) I think rather than needing the discrete card on during login it should just be fixed to not freeze.

That being said you could try setting the default state on 'on', which I'm pretty sure you can do with bumblebee, and then have your .profile turn off the card or something like that

bluca, 2017-11-09:
Adding this to a file in /etc/modprobe.d should do it: ```options bbswitch load_state=1```
Remember to run ```sudo depmod``` after adding the file

Queuecumber, 2017-11-12:
Yeah it turns out if wasn't working because bumblebee actually forces the card off when it starts regardless of what that option is set to. So if I want to use bumblebee (which I'm not actually sure  I need since I only use the card for CUDA), I need another solution which I am still cooking up, in the meantime it isn't *too* bad to just pop over to a tty and turn the card on befoire logging in then turn it back off after my session is started.

bluca, 2017-11-12:
You could disable the bumblebeed service and simply start it when you need it.

axelzaro, 2018-03-05:
The same problem here...