Bumblebee-Project/bbswitch #163: Unable to Power Off Card (nvidia_drm)

Queuecumber, 2017-11-09:
For whatever reason the discrete card fails to turn off again. I had this problem a long time ago but I thought it had been fixed, maybe nvidia went and messed something up again. After I run something to completion with optimus I find that the decrete card is on and `lsmod | grep nvidia` gives

```
nvidia_drm             45056  1
nvidia_modeset        843776  1 nvidia_drm
nvidia              13004800  1 nvidia_modeset
drm_kms_helper        167936  2 i915,nvidia_drm
drm                   364544  20 i915,nvidia_drm,drm_kms_helper
```

Trying to unload `nvidia_drm` of course gives

`
ERROR: Module nvidia_drm is in use
`

Not sure why it is holding on to drm like that

molecularentropy, 2018-01-17:
I can confirm. Same issue.

Lekensteyn, 2018-02-02:
You can probably check for the user with `sudo lsof -n /dev/dri/card*`

Crashdummyy, 2018-03-03:
Any news on this ? 
I am facing the same problem with:
Zorin OS 12.2 ( derives from Ubuntu 16.04 ),
Driver Versions: 340.104 || 384.111
Kernel: 4.13.0-36-generic

I can't use optirun, once I try to start something with optirun it enables the discrete Card.
Afterwards I can't switch back...

mbacilieri, 2018-04-12:
I faced the same problem in Ubuntu 16.04, and I solved removing xserver-xorg-**hwe** and reinstalling the "normal" one (xserver-xorg architecture). For unknown to me reasons the hwe version blocks bbswitch to work, and removes it too.
Using the "not-hwe" version you can reinstall bbswitch and manage the power of the discrete card again.
Nvidia drivers must be 340: following ones to me result bugged in various way.