Bumblebee-Project/bbswitch #164: Noob--reinstalled `bbswitch` but the file `/proc/acpi/bbswitch` is missing?

xtiansimon, 2017-11-26:
My laptop has Nvidia Optimus (Intel integrated GPU/NVIDIA dedicated GPU). I've been testing bumblebee on Lenovo W541 and troubleshooting the "cannot turn off" nvidia issue. I uninstalled bumblebee and bbswitch on Arch install using `sudo pacman -Rs [package]` (only uninstalled the target). Now, I've reinstalled bbswitch-lts/bbwsitch twice, but the `/proc/acpi/bbswitch` file is missing. What am I doing wrong?

Lekensteyn, 2017-11-26:
bbswitch is not loaded automatically, it is usually loaded via bumblebee. Alternative is to put it in /etc/modules-load.d/bbswitch.conf to make it load automatically at boot.

xtiansimon, 2017-11-26:
I should have mentioned what I did to install `bbswitch`. The [Arch Wiki page for Nvidia-xrun](https://wiki.archlinux.org/index.php/Nvidia-xrun#Use_bbswitch_to_manage_nvidia) gives instructions for installation:

> Load bbswitch module on boot 
>     `# echo 'bbswitch ' > /etc/modules-load.d/bbswitch`
> Disable the nvidia module on boot:
>     `# echo 'options bbswitch load_state=0 unload_state=1' /etc/modprobe.d/bbswitch.conf 
> After a reebot, the nvidia card will be off. This can bee seen by querying bbswitch's status:
>     `$ cat /proc/acpi/bbswitch`  

The files in `/etc/modules-load.d/bbswitch` and `bbswitch.conf` have permissions `644`.

The directory `/proc/acpi/` has just two directries and a file: `button`, `ibm`, and `wakeup`.

`dmesg | grep bbswitch` and `lsmod | grep bbswitch` don't show anything.

```
$ pacman -Qs bbswitch
local/bbswitch 0.8-83
local/bbswitch-lts 0.8.40
```

Right--as lsmod shows--the module isn't loading.

Tknott95, 2018-11-07:
I have a config up and running after the issue (also using lts) if anything arises I can compare our packages installed. It looks like you had the missing piece to my issue as I may to yours.