Bumblebee-Project/bbswitch #167: Cannot find ACPI handle for VGA device

alferum, 2018-04-06:
Hi,

I have a bug with bbswitch version 0.7 and 0.8 on Debian 9 Stretch.

I am using [C847MS-E33](https://www.msi.com/Motherboard/C847MS-E33.html) with integrated Intel CPU C847 (built-in graphics capabilities), and NVIDIA Geforce GTX 960.
```
$ sudo dmesg
bbswitch: version 0.8
bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
bbswitch: cannot find ACPI handle for VGA device 0000:01:00.0
bbswitch: No discrete VGA device found

$ lspci -vnn | grep '\''[030[02]\]'
00:02.0 VGA compatible controller [0300]: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller [8086:0106] (rev 09)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM206 [GeForce GTX 960] [10de:1401] (rev a1)

$ uname -a
Linux debian 4.9.0-6-amd64 #1 SMP Debian 4.9.82-1+deb9u3 (2018-03-02) x86_64 GNU/Linux
```
My acpi_dump_info output:

0000:00:00.0 060000 
0000:00:01.0 060400 \_SB_.PCI0.PEG0
0000:00:02.0 030000 \_SB_.PCI0.GFX0
0000:00:16.0 078000 
0000:00:1a.0 0c0320 \_SB_.PCI0.EHC2
0000:00:1b.0 040300 \_SB_.PCI0.HDEF
0000:00:1c.0 060400 \_SB_.PCI0.RP01
0000:00:1c.2 060400 \_SB_.PCI0.RP03
0000:00:1c.3 060401 \_SB_.PCI0.RP04
0000:00:1d.0 0c0320 \_SB_.PCI0.EHC1
0000:00:1f.0 060100 \_SB_.PCI0.LPCB
0000:00:1f.2 010601 \_SB_.PCI0.SAT0
0000:00:1f.3 0c0500 \_SB_.PCI0.SBUS
0000:01:00.0 030000 
0000:01:00.1 040300 
0000:03:00.0 020000 \_SB_.PCI0.RP03.PXSX
0000:04:00.0 060400 \_SB_.PCI0.RP04.PXSX
0000:05:00.0 028000 
0000:05:01.0 020000

My [get-acpi-info.sh  output](http://alferum.zzz.com.ua/upload/MSI-MS-7835.tar.gz)

It's not work with kernels 4.9.0-3-amd64, 4.9.0-6-amd64, 4.14.0-0-amd64, 4.15.0-2-amd64.
But good work with kernel 4.9.0-3-686-pae:**i386**

Xorg 1:7.7+19
xserver-xorg-video-nouveau 1:1.0.13-3
Kernel driver in use: i915

Lekensteyn, 2018-04-06:
Can you try with the upstream kernel without Debian patches?

alferum, 2018-04-12:
It's not work with kernel without Debian patches too, build [linux-4.15.tar.gz](https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.15.tar.gz)
```
$ uname -a
Linux debian 4.15.0 #6 SMP Thu Apr 12 21:09:32 EEST 2018 x86_64 GNU/Linux

$ sudo dmesg | grep bbswitch
bbswitch: loading out-of-tree module taints kernel.
bbswitch: version 0.8
bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
bbswitch: cannot find ACPI handle for VGA device 0000:01:00.0
bbswitch: No discrete VGA device found

$ lspci -vnn | grep '\''[030[02]\]'
00:02.0 VGA compatible controller [0300]: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller [8086:0106] (rev 09) (prog-if 00 [VGA controller])
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM206 [GeForce GTX 960] [10de:1401] (rev a1) (prog-if 00 [VGA controller])

$ cat /proc/acpi/dump_info
0000:00:00.0 060000 
0000:00:01.0 060400 \_SB_.PCI0.PEG0
0000:00:02.0 030000 \_SB_.PCI0.GFX0
0000:00:16.0 078000 
0000:00:1a.0 0c0320 \_SB_.PCI0.EHC2
0000:00:1b.0 040300 \_SB_.PCI0.HDEF
0000:00:1c.0 060400 \_SB_.PCI0.RP01
0000:00:1c.2 060400 \_SB_.PCI0.RP03
0000:00:1c.3 060401 \_SB_.PCI0.RP04
0000:00:1d.0 0c0320 \_SB_.PCI0.EHC1
0000:00:1f.0 060100 \_SB_.PCI0.LPCB
0000:00:1f.2 010601 \_SB_.PCI0.SAT0
0000:00:1f.3 0c0500 \_SB_.PCI0.SBUS
0000:01:00.0 030000 
0000:01:00.1 040300 
0000:03:00.0 020000 \_SB_.PCI0.RP03.PXSX
0000:04:00.0 060400 \_SB_.PCI0.RP04.PXSX
0000:05:00.0 028000 
0000:05:01.0 020000
```

alferum, 2018-04-22:
My computer is **desktop**, so bbswitch does not work probably.
For me, bumblebee works without bbswitch.
Thank you everyone.
You can close this issue.

muradm, 2018-12-03:
Having exactly same issue here on Lenovo Carbon X1C6 with Lenovo Thunderbolt 3 Graphics Dock.
```
$ dmesg | grep bbswitch 
[   13.307469] bbswitch: version 0.8
[   13.307474] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   13.307486] bbswitch: cannot find ACPI handle for VGA device 0000:10:00.0
[   13.307487] bbswitch: No discrete VGA device found
```
```
00:02.0 VGA compatible controller [0300]: Intel Corporation UHD Graphics 620 [8086:5917] (rev 07) (prog-if 00 [VGA controller])
10:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] [10de:1c8d] (rev a1) (prog-if 00 [VGA controller])
```
```
Linux 4.19.6-arch1-1-ARCH #1 SMP PREEMPT Sat Dec 1 14:03:49 UTC 2018 x86_64 GNU/Linux
```

kurt-o-sys, 2020-06-17:
This is still an issue in Debian Buster and Bullseye:
* https://egpu.io/forums/thunderbolt-linux-setup/bbswitch-no-discrete-vga-device-found-on-debian-buster/#post-81613
* https://egpu.io/forums/thunderbolt-linux-setup/bbswitch-no-acpi-handle-on-debian-bullseye/