Bumblebee-Project/bbswitch #168: Surface Book 2 : No suitable _DSM call found.

deb75, 2018-04-23:
Hello,

I run debian buster/10, kernel 4.15.18, on a surface book 2. This laptop has two graphic cards :

```
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
02:00.0 3D controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
```

When trying to load bbswitch I get :
```
modprobe: ERROR: could not insert 'bbswitch': No such device
```

and in logs :

```
[ 1418.488811] bbswitch: loading out-of-tree module taints kernel.
[ 1418.489114] bbswitch: version 0.8
[ 1418.489118] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[ 1418.489123] bbswitch: Found discrete VGA device 0000:02:00.0: \_SB_.PCI0.RP05.PXSX
[ 1418.489129] ACPI Warning: \_SB.PCI0.RP05.PXSX._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20170831/nsarguments-100)
[ 1418.489206] ACPI Warning: \_SB.PCI0.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20170831/nsarguments-100)
[ 1418.489214] bbswitch: No suitable _DSM call found.
```
Current nvidia driver version is 390.48, but I do not think it matters, bbswitch does not require the nvidia driver.

X.Org X Server 1.19.6

deb75, 2018-05-18:
Hi

Any hints ?

jaythespacehound, 2018-07-14:
Have been looking into this a bit but don't really understand ACPI stuff.
Sounds like it can't find a _DSM call for the nvidia card to power control it.
Looking at the DSDT this seems to be the method but I don't know enough about ACPI to figure out why it isn't recognised (or why it is the wrong one):

```
Scope (_SB.PCI0.RP05.PXSX)                                                                                                                                                                                       
    {                                                                                                                                                                                                                
        Name (TGPC, Buffer (0x04)                                                                                                                                                                                    
        {                                                                                                                                                                                                            
             0x00, 0x00, 0x00, 0x00                           // ....                                                                                                                                                
        })                                                                                                                                                                                                           
        Name (ACNT, Zero)                                                                                                                                                                                            
        Name (PSAP, Zero)                                                                                                                                                                                            
        Name (GPSP, Buffer (0x24){})                                                                                                                                                                                 
        CreateDWordField (GPSP, Zero, RETN)                                                                                                                                                                          
        CreateDWordField (GPSP, 0x04, VRV1)                                                                                                                                                                          
        CreateDWordField (GPSP, 0x08, TGPU)                                                                                                                                                                          
        CreateDWordField (GPSP, 0x0C, PDTS)                                                                                                                                                                          
        CreateDWordField (GPSP, 0x10, SFAN)                                                                                                                                                                          
        CreateDWordField (GPSP, 0x14, SKNT)                                                                                                                                                                          
        CreateDWordField (GPSP, 0x18, CPUE)                                                                                                                                                                          
        CreateDWordField (GPSP, 0x1C, TMP1)                                                                                                                                                                          
        CreateDWordField (GPSP, 0x20, TMP2)                                                                                                                                                                          
        Method (_DSM, 4, NotSerialized)  // _DSM: Device-Specific Method                                                                                                                                             
        {                                                                                                                                                                                                            
            If ((Arg0 == ToUUID ("a3132d01-8cda-49ba-a52e-bc9d46df6b81")))                                                                                                                                           
            {                                                                                                                                                                                                        
                If ((Arg1 == 0x0100))                                                                                                                                                                                
                {                                                                                                                                                                                                    
[... cut ...]
                 0x00                                             // .                                                                                                                                               
            })                                                                                                                                                                                                       
        } 
```

I think I  was able to turn off the card using acpi_call '\_SB.PCI0.RP05.PRP5._OFF' (judging by powertop) but ymmv

Schroedi, 2018-11-07:
Jaythespacehound's proposed acpi_call works for me. I followed these steps to make it persist reboots: https://wiki.archlinux.org/index.php/Hybrid_graphics#Using_acpi_call