Bumblebee-Project/bbswitch #169: Failure to start in Ubuntu 18.04

Trevortds, 2018-05-06:
I'm scouring the installation guides and forum posts for what might have gone wrong here, but I'm stumped. The critical error seems to be AE_NOT_FOUND, but I can't find any documentation on it. 

`dmesg | grep -c 10 bbswitch`

```
[    5.840698] hid-generic 0003:04D9:0141.0011: hiddev6,hidraw12: USB HID v1.11 Device [USB Keyboard] on usb-0000:00:14.0-3.4/input1
[    5.841390] input: USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4/1-3.4:1.2/0003:04D9:0141.0012/input/input10
[    5.843441] input: Logitech M525 as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.2/0003:046D:C52B.000C/0003:046D:4013.0013/input/input11
[    5.900060] hid-generic 0003:04D9:0141.0012: input,hidraw13: USB HID v1.11 Mouse [USB Keyboard] on usb-0000:00:14.0-3.4/input2
[    5.900116] logitech-hidpp-device 0003:046D:4013.0013: input,hidraw14: USB HID v1.11 Mouse [Logitech M525] on usb-0000:00:14.0-3.2:1
[    5.905496] input: Logitech MX Master as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.2/1-3.2:1.2/0003:046D:C52B.000C/0003:046D:4041.0014/input/input12
[    5.906337] logitech-hidpp-device 0003:046D:4041.0014: input,hidraw15: USB HID v1.11 Keyboard [Logitech MX Master] on usb-0000:00:14.0-3.2:2
[    5.913488] input: Logitech MX Master as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:046D:C52B.000F/0003:046D:4041.0015/input/input13
[    5.914504] logitech-hidpp-device 0003:046D:4041.0015: input,hidraw16: USB HID v1.11 Keyboard [Logitech MX Master] on usb-0000:00:14.0-8:1
[    5.925505] PKCS#7 signature not signed with a trusted key
[    5.926190] bbswitch: loading out-of-tree module taints kernel.
[    5.926889] bbswitch: module verification failed: signature and/or required key missing - tainting kernel
[    5.927797] bbswitch: version 0.8
[    5.928533] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[    5.929249] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[    5.929974] bbswitch: failed to evaluate \_SB_.PCI0.PEG0.PEGP._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    5.931450] bbswitch: failed to evaluate \_SB_.PCI0.PEG0.PEGP._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    5.933009] ACPI Warning: \_SB.PCI0.GFX0._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20170831/nsarguments-100)
[    5.933824] bbswitch: No suitable _DSM call found.
[    6.214449] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
[    6.290784] random: crng init done
[    6.302903] ip_tables: (C) 2000-2006 Netfilter Core Team
[    6.395818] systemd[1]: systemd 237 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN2 +IDN -PCRE2 default-hierarchy=hybrid)
[    6.416454] systemd[1]: Detected architecture x86-64.
[    6.423609] systemd[1]: Set hostname to <tulkas>.
[    6.433756] systemd-fstab-generator[343]: Failed to create unit file /run/systemd/generator/dev-disk-by\x2duuid-7c5d4fa6\x2d1c18\x2d49da\x2db454\x2df5680de9d32c.swap, as it already exists. Duplicate entry in /etc/fstab?
[    6.487307] systemd[1]: Reached target User and Group Name Lookups.
[    6.489238] systemd[1]: Reached target Remote File Systems.
[    6.491066] systemd[1]: Reached target Libvirt guests shutdown.
--
[    7.276909] audit: type=1400 audit(1525644154.466:8): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/man" pid=885 comm="apparmor_parser"
[    7.276911] audit: type=1400 audit(1525644154.466:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="man_filter" pid=885 comm="apparmor_parser"
[    7.276912] audit: type=1400 audit(1525644154.466:10): apparmor="STATUS" operation="profile_load" profile="unconfined" name="man_groff" pid=885 comm="apparmor_parser"
[    7.276997] audit: type=1400 audit(1525644154.466:11): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/sbin/cups-browsed" pid=892 comm="apparmor_parser"
[    7.469894] Adding 9765884k swap on /dev/sdb1.  Priority:-3 extents:1 across:9765884k FS
[    7.648534] input: HDA NVidia HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input16
[    7.648561] input: HDA NVidia HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input17
[    7.648583] input: HDA NVidia HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input18
[    7.648603] input: HDA NVidia HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input19
[    7.688035] PKCS#7 signature not signed with a trusted key
[    7.688605] bbswitch: version 0.8
[    7.688610] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[    7.688613] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[    7.688623] bbswitch: failed to evaluate \_SB_.PCI0.PEG0.PEGP._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    7.688627] bbswitch: failed to evaluate \_SB_.PCI0.PEG0.PEGP._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    7.688659] bbswitch: No suitable _DSM call found.
[    7.968508] IPv6: ADDRCONF(NETDEV_UP): enp6s0: link is not ready
[    7.986884] r8169 0000:06:00.0 enp6s0: link down
[    7.986967] IPv6: ADDRCONF(NETDEV_UP): enp6s0: link is not ready
[    7.989088] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
[    8.001099] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
[    8.078753] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
[    8.656713] ip6_tables: (C) 2000-2006 Netfilter Core Team
[    8.762968] Ebtables v2.0 registered
[    9.189399] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    9.189915] virbr0: port 1(virbr0-nic) entered blocking state
```

On the kernel that comes with 18.04. I've tried some older ones like 4.4.75 and 4.13.0, but neither work, so I doubt it's the kernel's fault. 

`uname -a`

`Linux tulkas 4.15.0-20-generic #21-Ubuntu SMP Tue Apr 24 06:16:15 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

Nouveau and nvidia are successfully blacklisted

`xrandr --listproviders`
```
Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel
```

get-acpi-info.sh dump here: https://bugs.launchpad.net/lpbugreporter/+bug/752542/comments/832

acpi-stuff info dump. 

```
0000:00:00.0 060000 \_SB_.PCI0.DSC0
0000:00:01.0 060400 \_SB_.PCI0.PEG0
0000:00:02.0 030000 \_SB_.PCI0.GFX0
0000:00:14.0 0c0330 \_SB_.PCI0.XHC_
0000:00:14.2 050000 
0000:00:16.0 078000 \_SB_.PCI0.HECI
0000:00:17.0 010601 \_SB_.PCI0.SAT0
0000:00:1c.0 060400 \_SB_.PCI0.RP03
0000:00:1d.0 060400 \_SB_.PCI0.RP09
0000:00:1d.2 060400 \_SB_.PCI0.RP11
0000:00:1d.3 060400 \_SB_.PCI0.RP12
0000:00:1f.0 060100 \_SB_.PCI0.LPCB
0000:00:1f.3 040300 \_SB_.PCI0.HDAS
0000:00:1f.4 0c0500 \_SB_.PCI0.SBUS
0000:00:1f.5 0c8000 
0000:01:00.0 030000 \_SB_.PCI0.PEG0.PEGP
0000:01:00.1 040300 
0000:02:00.0 028000 \_SB_.PCI0.RP03.PXSX
0000:04:00.0 060400 \_SB_.PCI0.RP11.PXSX
0000:06:00.0 020000 \_SB_.PCI0.RP12.PXSX
```

Motherboard, 2018-05-15:
I got the same thing in Ubuntu 17.10:

```
[127506.827564] bbswitch: version 0.8
[127506.827567] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[127506.827570] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[127506.827577] bbswitch: failed to evaluate \_SB_.PCI0.PEG0.PEGP._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[127506.827581] bbswitch: failed to evaluate \_SB_.PCI0.PEG0.PEGP._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[127506.827603] bbswitch: No suitable _DSM call found.
```

lspci -knn:

```
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 630 [8086:5912] (rev 04)
	Subsystem: ASUSTeK Computer Inc. HD Graphics 630 [1043:872f]
	Kernel driver in use: i915
	Kernel modules: i915

...

01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] [10de:1b06] (rev a1)
	Subsystem: Gigabyte Technology Co., Ltd GP102 [GeForce GTX 1080 Ti] [1458:3752]
	Kernel driver in use: nvidia
	Kernel modules: nvidiafb, nouveau, nvidia_390, nvidia_390_drm
01:00.1 Audio device [0403]: NVIDIA Corporation GP102 HDMI Audio Controller [10de:10ef] (rev a1)
	Subsystem: Gigabyte Technology Co., Ltd GP102 HDMI Audio Controller [1458:3752]
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel

```

uname -r

`4.13.0-41-generic`

acpidump available [here](https://bugs.launchpad.net/lpbugreporter/+bug/752542/+attachment/5140014/+files/System_manufacturer-System_Product_Name.tar.gz)

Trevortds, 2018-05-18:
Tried again in a brand new fresh install with no results. 

using this guide https://gist.github.com/Misairu-G/616f7b2756c488148b7309addc940b28

and this addition. https://bugs.launchpad.net/ubuntu/+source/bumblebee/+bug/1758243

zoulock, 2018-05-22:
I'm having exactly the same problem with Manjaro. I tried 4.14 and 4.16 kernels and had the same error messages. 
```
lspci -knn | grep VGA
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 530 [8086:1912] (rev 06)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP104 [GeForce GTX 1080][10de:1b80] (rev a1)
```

vladislav06, 2022-06-08:
same error on:
82JW Legion 5 15ACH6
5.17.9-1-MANJARO 

```[    4.390138] bbswitch: version 0.8
[    4.390150] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.GPP0.PEGP
[    4.390153] bbswitch: Found discrete VGA device 0000:05:00.0: \_SB_.PCI0.GP17.VGA_
[    4.390169] bbswitch: failed to evaluate \_SB_.PCI0.GP17.VGA_._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    4.390175] bbswitch: failed to evaluate \_SB_.PCI0.GP17.VGA_._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    4.390177] bbswitch: No suitable _DSM call found.
```