Bumblebee-Project/bbswitch #170: bbswitch doesn't seem to work on kernel in recent kernel

Elv13, 2018-05-16:
Hi,

I don't think I am experiencing https://github.com/Bumblebee-Project/bbswitch/issues/140 directly. However since Linux 4.8, `bbswitch` (with or without `pcie_port_pm=off`) doesn't seem to do "it's job". The power usage on 4.7 is ~9watts and ~20watts in 4.8+. I can enable and disable the card, but it doesn't affect the power usage at all. The fans still spin even with low CPU temp.

```
cat /proc/acpi/bbswitch                                                 
0000:01:00.0 OFF
```

I also tried `echo 1 >/sys/bus/pci/devices/0000:01:00.0/remove` but that also doesn't restore the power usage back to normal. I really, really don't care or want to NVIDIA GPU to be active. I don't use it and I need the longer battery life. There is nothing suspicious in `dmesg`. It seems to "everything is fine", but in reality nothing is done. Is there any reliable way to debug this (I am ok patching bbswitch or the kernel).

Device:
`01:00.0 VGA compatible controller: NVIDIA Corporation GK107GLM [Quadro K1100M] (rev ff)`

Kernel:

```
linux	/boot/vmlinuz-4.16.8 root=/dev/sda1 ro acpi_osi="!Windows 2013" acpi_osi=Linux acpi_backlight=vendor i915.allow_pc8=1 i915.enable_psr=1 i915.powersave=1 i915.enable_rc6=1 i915.enable_fbc=1 i915.lvds_downclock=1 i915.semaphores=1 modprobe.blacklist=nouveau pcie_port_pm=off
```

ArchangeGabriel, 2018-05-16:
Hi,

I’ve also had suboptimal powersaving using recent kernels. I’m now using nouveau, with `pcie_port_pm=off` because else powersavings are not as good as expected. I suspect this is because of the forced `acpi_osi`… Have you tried using nouveau instead of bbswitch?

Also, note that `i915.enable_rc6`, `i915.lvds_downclock` and `i915.semaphores` as well as `i915.powersave` have been removed, so specifying them is useless.

Elv13, 2018-05-16:
> Have you tried using nouveau instead of bbswitch?

I don't want `nouveau`, I want the NVIDIA GPU to be turned off and use the Intel one exclusively. Also, I never had luck with `nouveau`. On my Mac turns the screen `#00ff00` and then overheat and on this laptop it's a Quadro, so it's untested, laggy and crashy. I ran the NVIDIA propriatary driver for a while, but it eats the battery in no time. This GPU is good for CAD, but it's very power hungry.

What I am looking for is getting of the last pre-4.8 LTS kermel before it's discontinued. Also, 4.17 is supposed to have magical new scheduling capable of saving ~10% energy. This was my main motivation for trying again. I can't believe there is no way to turn off whatever causes this regression. Still, can't find any solution that works and we are not that many with this GPU so the pool of knowledge seems too small.

ArchangeGabriel, 2018-05-16:
nouveau turns off the GPU. I use it mostly only for that (and the fact my HDMI/Thunderbolt 3 outputs are wired to it).

Elv13, 2018-07-04:
Hi again,

Initially it didn't work, but with the latest git-master kernel the `nouveau` team fixed PM for my card, so at least it doesn't take 35w anymore.

From what I sent on `nouveau` IRC:

I tried with Kernel 4.4 and git-master with my script and with laptop-mode-tools. Totally idle with backlight off, 4.4 takes 6.4w and git-master+nouveau 12.0, almost twice as much. The test setup was running the kernel in single user mode, starting X using `X :0 & sleep 5 && DISPLAY=:0 xterm -c 'powertop'`. laptop-mode and my ACPI script have the same result because they do the same thing. I checked everything and the git-master kernel has all the PM features it can have, so the NVIDIA card is really the only possible cause for the doubled power usage

So to recap (fully idle single user mode tests):

type | power consumption
-----|-----
Linux 4.4 + bbswitch | 6.4w
Linux 4.8-4.17 + broken bbswitch |  ~18w
Linux 4.4-17(?) + Nouveau | 35.5w
Linux 4.18-rc3 + Nouveau + forced PM | 12w

kernel args:

    linux	/boot/vmlinuz-4.18.0-rc3 root=/dev/sda1 ro acpi_osi="!Windows 2013" acpi_osi=Linux acpi_backlight=vendor i915.allow_pc8=1 i915.enable_psr=1 i915.powersave=1 i915.enable_fbc=1 i915.lvds_downclock=1 nouveau.config=NvPmEnableGating=1 pcie_aspm=force pcie_port_pm=on pcie_aspm.policy=powersupersave

adhefe, 2018-07-09:
Here, it seems that dkms doesn't goes fine because of some new kernel 4.16 features:

`# dkms install -m bbswitch -v 0.8-6`

DKMS make.log for bbswitch-0.8-6 for kernel 4.16.13-desktop-1omv (x86_64)
qui jun  7 21:08:29 -03 2018
make -C /lib/modules/4.16.13-desktop-1omv/build M="$(pwd)" modules
make[1]: Entering directory '/usr/src/linux-4.16.13-desktop-1omv'
  CC [M]  /var/lib/dkms/bbswitch/0.8-6/build/bbswitch.o
/var/lib/dkms/bbswitch/0.8-6/build/bbswitch.c: In function ‘acpi_call_dsm’:
/var/lib/dkms/bbswitch/0.8-6/build/bbswitch.c:138:28: note: byref variable will be forcibly initialized
         struct acpi_buffer buf = { ACPI_ALLOCATE_BUFFER, NULL };
                            ^~~
/var/lib/dkms/bbswitch/0.8-6/build/bbswitch.c:109:29: note: byref variable will be forcibly initialized
     struct acpi_object_list input;
                             ^~~~~
/var/lib/dkms/bbswitch/0.8-6/build/bbswitch.c:108:24: note: byref variable will be forcibly initialized
     struct acpi_buffer output = { ACPI_ALLOCATE_BUFFER, NULL };
                        ^~~~~~
/var/lib/dkms/bbswitch/0.8-6/build/bbswitch.c: In function ‘bbswitch_init’:
/var/lib/dkms/bbswitch/0.8-6/build/bbswitch.c:398:28: note: byref variable will be forcibly initialized
         struct acpi_buffer buf = { ACPI_ALLOCATE_BUFFER, NULL };
                            ^~~
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /var/lib/dkms/bbswitch/0.8-6/build/bbswitch.mod.o
  LD [M]  /var/lib/dkms/bbswitch/0.8-6/build/bbswitch.ko
ld: warning: Disabling --icf due to -r
make[1]: Leaving directory '/usr/src/linux-4.16.13-desktop-1omv'