Bumblebee-Project/bbswitch #175: Querying PCI bus turns Nvidia on

haarp, 2018-08-03:
Greetings.

I'm running kernel 4.14 on a Thinkpad W530. As soon as the PCI bus is queried, e.g. by lspci, the card instantly switches on. I do not remember if this had been an issue with earlier kernels.

The nvidia module remains untouched and unloaded throughout all this. The Bumblebee also wasn't running.

    > cat /proc/acpi/bbswitch 
    0000:01:00.0 OFF
    > lspci
    ....
    ....
    > cat /proc/acpi/bbswitch 
    0000:01:00.0 ON

Weird, isn't it?

haarp, 2018-08-03:
It seems reading ```/sys/bus/pci/devices/0000:01:00.0/config``` (01:00.0 being the Nvidia) is what triggers the switch-on.

Lekensteyn, 2018-08-11:
Yes, that is runtime PM that is becoming active. There is not much you can do about that other than setting `pcie_port_pm=off`, but then maybe it is better just to remove bbswitch. Related: https://github.com/Bumblebee-Project/Bumblebee/issues/978