Bumblebee-Project/bbswitch #177: nvidia card refuses to change power state

lumag, 2018-08-15:
Hello,

After one of updates I've noticed following error in dmesg on my Dell Inspiron 5558.
Relevant parts of kernel log:
```
[    0.000000] Linux version 4.18.0-rc4-amd64 (debian-kernel@lists.debian.org) (gcc version 7.3.0 (Debian 7.3.0-25)) #1 SMP Debian 4.18~rc4-1~exp1 (2018-07-12)
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-4.18.0-rc4-amd64 root=/dev/mapper/rhovanion--vg-root ro pcie_port_pm=off pcie_aspm=force bbswitch.dyndbg quiet
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
--
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 3075436
[    0.000000] Policy zone: Normal
[    0.000000] Kernel command line: BOOT_IMAGE=/vmlinuz-4.18.0-rc4-amd64 root=/dev/mapper/rhovanion--vg-root ro pcie_port_pm=off pcie_aspm=force bbswitch.dyndbg quiet
[    0.000000] PCIe ASPM is forcibly enabled
[    0.000000] Calgary: detecting Calgary via BIOS EBDA area
--
[   75.261628] usbcore: registered new interface driver rndis_host
[   75.361158] rndis_host 2-3:1.0 enxac5043ccbff8: renamed from eth1
[   75.722701] bbswitch: loading out-of-tree module taints kernel.
[   75.723660] bbswitch: version 0.8
[   75.723669] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   75.723680] bbswitch: Found discrete VGA device 0000:08:00.0: \_SB_.PCI0.RP05.PEGP
[   75.723696] ACPI Warning: \_SB.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20180531/nsarguments-66)
[   75.723878] bbswitch: detected an Optimus _DSM function
[   75.723901] pci 0000:08:00.0: enabling device (0006 -> 0007)
[   75.724117] bbswitch: Succesfully loaded. Discrete card 0000:08:00.0 is on
[   75.725930] bbswitch: disabling discrete graphics
[   75.726497] bbswitch: Result of Optimus _DSM call: 11000059
[   76.149216] usb 2-6: New USB device found, idVendor=0cf3, idProduct=e005, bcdDevice= 0.02
[   76.149219] usb 2-6: New USB device strings: Mfr=0, Product=0, SerialNumber=0
--
[  139.157964] PKCS#7 signature not signed with a trusted key
[  149.270434] PKCS#7 signature not signed with a trusted key
[  152.936326] bbswitch: enabling discrete graphics
[  156.492705] nvidia: module license 'NVIDIA' taints kernel.
[  156.492707] Disabling lock debugging due to kernel taint
--
[  156.521864] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  390.77  Tue Jul 10 18:28:52 PDT 2018 (using threaded interrupts)
[  170.590012] nvidia-nvlink: Unregistered the Nvlink Core, major device number 242
[  170.669996] bbswitch: disabling discrete graphics
[  170.670633] bbswitch: Result of Optimus _DSM call: 11000059
[  170.687699] pci 0000:08:00.0: Refused to change power state, currently in D0
```

Do you need any additional info like DSDT?

Lekensteyn, 2018-08-15:
Do you observe any issues other than the message? Like battery drain or increased heat?

When is your system produced? `cd /sys/class/dmi/id && grep . bios_*`
If it is a recent one, the current "stable" bbswitch might not be the best solution.

Lekensteyn, 2018-08-20:
Specifically for the XPS 9560 you can try to boot with the `acpi_rev_override=1` kernel option to *workaround* the hang issue. This makes it possible to transition to a lower power state without breaking all kinds of stuff.

On the long term, I have been trying (again) this weekend to find a proper fix (using a XPS 9560 from my brother as test target), but there is not much progress unfortunately.

glunardi, 2018-08-20:
Will try right away and report back. Thank you so much for working on this.

If we can help with any testing, do not hesitate to ask.

lumag, 2018-09-05:
Yes, that still gives an error. Strangely enough with latest nVidia driver 390.77 and Linux 4.17.17 I observe even less stable behaviour (often optirun won't work at all, with the following message in dmesg):
```
Sep  5 13:36:02 rhovanion kernel: [ 8515.376844] NVRM: RmInitAdapter failed! (0x25:0x40:1101)
Sep  5 13:36:02 rhovanion kernel: [ 8515.575817] NVRM: rm_init_adapter failed for device bearing minor number 0
```

GreyDireWolf, 2018-09-06:
Currently on my arch machine I can't reach to graphical target(it hardlocks the kernel). I have to manually start x from multi-user target. Also lspci hardlocks the kernel. I tested another kernel without bbswitch which works fine.

zaro, 2019-05-11:
Same here on my Xiaomi Notebook Pro with Fedora 30.

```
$ lspci |grep -E "VGA|3D"
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
01:00.0 3D controller: NVIDIA Corporation GP108M [GeForce MX150] (rev ff)
```
Everything installs correctly but when I try to run something with optirun I get:
```
$ optirun nvidia-settings
[  145.725551] [ERROR]Cannot access secondary GPU - error: Could not enable discrete graphics card

[  145.725590] [ERROR]Aborting because fallback start is disabled.
```

And in the syslog:

```
May 11 12:50:44 mi kernel: pci 0000:01:00.0: Refused to change power state, currently in D3
May 11 12:50:44 mi kernel: bbswitch: enabling discrete graphics
May 11 12:50:44 mi kernel: pci 0000:01:00.0: Refused to change power state, currently in D3
May 11 12:50:44 mi bumblebeed[957]: [  489.061622] [ERROR]Could not enable discrete graphics card
```

ukos-git, 2019-05-25:
Same here on DELL Precision 5530 Laptop with  Nvidia Quadro P1000 on debian (4.19.0-5-amd64)

<details>
<summary>
package versions
</summary>
<pre>
apt-cache policy bumblebee-nvidia primus libgl1-nvidia-glx         
bumblebee-nvidia:
  Installed: 3.2.1-20
  Candidate: 3.2.1-20
  Version table:
 *** 3.2.1-20 500
        500 http://deb.debian.org/debian buster/contrib amd64 Packages
        100 /var/lib/dpkg/status
primus:
  Installed: 0~20150328-7
  Candidate: 0~20150328-7
  Version table:
 *** 0~20150328-7 500
        500 http://deb.debian.org/debian buster/main amd64 Packages
        100 /var/lib/dpkg/status
libgl1-nvidia-glx:
  Installed: 418.74-1
  Candidate: 418.74-1
  Version table:
 *** 418.74-1 500
        500 http://deb.debian.org/debian buster/non-free amd64 Packages
        100 /var/lib/dpkg/status
</pre>
</details>

WhyKickAmooCow, 2019-05-31:
Similar to @zaro.

Fedora 30. XPS 9570 with 1050ti.

```
# optirun glxgears
[23566.444916] [ERROR]Cannot access secondary GPU - error: Could not enable discrete graphics card

[23566.445020] [ERROR]Aborting because fallback start is disabled.

```

and in dmesg:

```
[23694.694974] bbswitch: enabling discrete graphics
[23694.695016] pci 0000:01:00.0: Refused to change power state, currently in D3
```

ukos-git, 2019-06-01:
Power Management for these cards is not supported by nvidia firmware. As soon as bbswitch disables the discrete card, it will never be available again. 

My current approach is to disable power management completely for the nvidia card in bumblebee.conf. This leaves the card available for applications (constantly on) but e.g. standby will not work anymore and around 5W will be lost to graphics.  

Lets hope for a fix upstream at DELL and all the other notebook manufacturers or a decent implementation at nouveau which does PM related stuff better than the nvidia driver. 

Anyway. Unfortunately, I don‘t think, this is a bbswitch issue.

WhyKickAmooCow, 2019-06-02:
Thanks for the explanation. 

I have found that the [nvidia-xrun](https://github.com/Witko/nvidia-xrun) utility works almost flawlessly. After following the instructions in the [copr repo](https://copr.fedorainfracloud.org/coprs/ekultails/nvidia-xrun/), and setting `nvidia-drm.modeset=0` in `/etc/default/grub` I can power on the GPU as needed, and turn it off when I want power savings. Apparently the performance is also significantly better than with bumblebee, although some applications do not run very well by themselves and work much better inside a WM like openbox.

It might be worth a try as it is the best solution I have found so far.