Bumblebee-Project/bbswitch #178: bbswitch fails to start on Fedora 28

ivandeskov, 2018-08-18:
Hi,

I just installed bumblebee on my Dell XPS 15 with GeForce GTX 1050 Ti Mobile running Fedora 28.

Output from ``$ systemctl status bumblebeed.service``

>● bumblebeed.service - Bumblebee C Daemon
   Loaded: loaded (/usr/lib/systemd/system/bumblebeed.service; enabled; vendor preset: disabled)
   Active: active (running) since Sat 2018-08-18 19:47:53 BST; 14min ago
 Main PID: 964 (bumblebeed)
    Tasks: 1 (limit: 4915)
   Memory: 2.2M
   CGroup: /system.slice/bumblebeed.service
           └─964 /usr/sbin/bumblebeed
Aug 18 19:47:53 chocho systemd[1]: Started Bumblebee C Daemon.
Aug 18 19:47:53 chocho bumblebeed[964]: modprobe: ERROR: could not insert 'bbswitch': Operation not permitted
Aug 18 19:47:53 chocho bumblebeed[964]: [    5.844055] [ERROR]Module bbswitch could not be loaded (timeout?)
Aug 18 19:47:53 chocho bumblebeed[964]: [    5.844095] [WARN]No switching method available. The dedicated card will always be on.
Aug 18 19:47:53 chocho bumblebeed[964]: [    5.845788] [INFO]/usr/sbin/bumblebeed 3.2.1 started

I did try running
``$ sudo /usr/sbin/bumblebeed``

And the result was similar:
>modprobe: ERROR: could not insert 'bbswitch': Operation not permitted
[  628.862822] [ERROR]Module bbswitch could not be loaded (timeout?)
[  628.862833] [WARN]No switching method available. The dedicated card will always be on.
[  628.863988] [INFO]/usr/sbin/bumblebeed 3.2.1 started

Please let us know if you need more info.

Thanks

Lekensteyn, 2018-08-20:
Possibly a selinux policy issue, please seek help with Fedora help channels.

Also note that you might have to add `acpi_rev_override=1` to your kernel command line in order to avoid hangs with your laptop model.

ivandeskov, 2018-08-25:
That was to do with the fact that I had secure boot enabled and bbswitch was unsigned. Couldn't be bothered signing the module, so just disabled secure boot and bbswitch is now loading ok.

I've also added that property to my kernel command line. Thanks for your help

closed 2018-08-25 by ivandeskov