Bumblebee-Project/bbswitch #187: Integrated graphic detected as discrete graphic

neoacevedo, 2019-09-23:
```sh
nestor@neoacevedo:~> dmesg | grep bbswitch
[    5.542699] bbswitch: version 0.8
[    5.542714] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.GPP0.VGA_
[    5.542718] bbswitch: Found discrete VGA device 0000:04:00.0: \_SB_.PCI0.GP17.VGA_
[    5.542747] bbswitch: failed to evaluate \_SB_.PCI0.GP17.VGA_._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    5.542755] bbswitch: failed to evaluate \_SB_.PCI0.GP17.VGA_._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    5.542756] bbswitch: No suitable _DSM call found.
```

busID 01:00.0 Nvidia dGPU
busID 04:00.0 AMD Radeon Vega iGPU

Siorki, 2020-01-26:
Same issue with different hardware

bbswitch log : 
```
# dmesg | grep -C 10 bbswitch
[    3.109629] bbswitch: version 0.8
[    3.109643] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.GPP0.PEGP
[    3.109647] bbswitch: Found discrete VGA device 0000:05:00.0: \_SB_.PCI0.GP17.VGA_
[    3.109662] bbswitch: failed to evaluate \_SB_.PCI0.GP17.VGA_._DSM {0xF8,0xD8,0x86,0xA4,0xDA,0x0B,0x1B,0x47,0xA7,0x2B,0x60,0x42,0xA6,0xB5,0xBE,0xE0} 0x100 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    3.109666] bbswitch: failed to evaluate \_SB_.PCI0.GP17.VGA_._DSM {0xA0,0xA0,0x95,0x9D,0x60,0x00,0x48,0x4D,0xB3,0x4D,0x7E,0x5F,0xEA,0x12,0x9F,0xD4} 0x102 0x0 {0x00,0x00,0x00,0x00}: AE_NOT_FOUND
[    3.109667] bbswitch: No suitable _DSM call found.
[    3.115139] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    3.115141] Bluetooth: BNEP filters: protocol multicast
[    3.115148] Bluetooth: BNEP socket layer initialized
[    3.189794] EDAC amd64: Node 0: DRAM ECC disabled.
[    3.189796] EDAC amd64: ECC disabled in the BIOS or no ECC capability, module will not load.
                Either enable ECC checking or force module loading by setting 'ecc_enable_override'.
                (Note that use of the override may cause unknown side effects.)
[    3.290320] EDAC amd64: Node 0: DRAM ECC disabled.
[    3.290323] EDAC amd64: ECC disabled in the BIOS or no ECC capability, module will not load.
                Either enable ECC checking or force module loading by setting 'ecc_enable_override'.
``` 

Kernel version
```
uname -a
Linux cadmium 5.4.0-2-amd64 #1 SMP Debian 5.4.8-1 (2020-01-05) x86_64 GNU/Linux
```

Xorg and driver version (using driver nouveau. Proprietary nvidia driver not installed)
```
# dpkg-query -l "*xorg*:amd64"
||/ Nom                         Version                  Architecture Description
+++-===========================-========================-============-=================================================
ii  xorg                        1:7.7+20                 amd64        X.Org X Window System
ii  xserver-xorg                1:7.7+20                 amd64        X.Org X server
ii  xserver-xorg-core           2:1.20.7-2               amd64        Xorg X server - core server
ii  xserver-xorg-input-all      1:7.7+20                 amd64        X.Org X server -- input driver metapackage
ii  xserver-xorg-input-libinput 0.29.0-1                 amd64        X.Org X server -- libinput input driver
ii  xserver-xorg-input-mouse    1:1.9.3-1                amd64        X.Org X server -- mouse input driver
ii  xserver-xorg-input-wacom    0.34.99.1-1+b1           amd64        X.Org X server -- Wacom input driver
ii  xserver-xorg-legacy         2:1.20.7-2               amd64        setuid root Xorg server wrapper
ii  xserver-xorg-video-amdgpu   19.1.0-1                 amd64        X.Org X server -- AMDGPU display driver
ii  xserver-xorg-video-ati      1:19.1.0-1               amd64        X.Org X server -- AMD/ATI display driver wrapper
ii  xserver-xorg-video-fbdev    1:0.5.0-1                amd64        X.Org X server -- fbdev display driver
ii  xserver-xorg-video-intel    2:2.99.917+git20190815-1 amd64        X.Org X server -- Intel i8xx, i9xx display driver
ii  xserver-xorg-video-nouveau  1:1.0.16-1               amd64        X.Org X server -- Nouveau display driver
ii  xserver-xorg-video-qxl      0.1.5-2+b1               amd64        X.Org X server -- QXL display driver
ii  xserver-xorg-video-radeon   1:19.1.0-1               amd64        X.Org X server -- AMD/ATI Radeon display driver
ii  xserver-xorg-video-vesa     1:2.4.0-2                amd64        X.Org X server -- VESA display driver
ii  xserver-xorg-video-vmware   1:13.3.0-2               amd64        X.Org X server -- VMware display driver
```

Distribution : **Debian bullseye** (testing)

Machine information : [ASUS TUF Gaming FX505DD](https://bugs.launchpad.net/lpbugreporter/+bug/752542/comments/848)

Let me know if you need more details or there are any tests that I can run to help you debug. Thanks !

Siorki, 2020-01-27:
Issue spotted. The test to identify the integrated graphics device only triggers on Intel cards

In bbswitch.c : 421
``` C
if (pdev->vendor == PCI_VENDOR_ID_INTEL) {
    igd_handle = handle;
    pr_info("Found integrated VGA device %s: %s\n",
        dev_name(&pdev->dev), (char *)buf.pointer);
}
```

Thus it won't support hardware with both ATI (integrated) and NVidia (discrete) cards.
As a quick and dirty fix, I replaced `PCI_VENDOR_ID_INTEL` by  `PCI_VENDOR_ID_ATI` in the code above and got the module to work.

bomiyr, 2021-11-20:
I have pretty the same issue with two "discrete" GPUs. My hardware: Lenovo Legion 5, CPU AMD 4800H with AMD Renoir graphics + GeForce GTX 1660 Ti Mobile. OS openSUSE Tumbleweed.

I downloaded the code, applied the fix with adding condition || pdev->vendor == PCI_VENDOR_ID_ATI and compiled. After that I was able to load my version of bbswitch. And still I was not able to make it work, even more - it has really scary behavior. Right after I disable nvidia GPU all fans started spinning at 100%, and then after several dozens of seconds laptop just powered off. Maybe it has some hardware protection or something similar that treat disabled GPU as something wrong, I don't know

whenov, 2021-11-20:
> I have pretty the same issue with two "discrete" GPUs. My hardware: Lenovo Legion 5, CPU AMD 4800H with AMD Renoir graphics + GeForce GTX 1660 Ti Mobile. OS openSUSE Tumbleweed.
>
> I downloaded the [...]

If you want to disable the Nvidia GPU, try using `echo 1 > '/sys/bus/pci/devices/0000:01:00.0/remove'`. (You may need to modify the path)

I also have the same machine but with GTX 2060 and Arch Linux instead.

bomiyr, 2021-11-20:
@whenov  Thank you for reply.Before I'll try, how to revert it back? Is it just ’echo 0 > {same path}/remove’?
Also, if you have the same machine, what is your power consumtion in idle when discrete GPU is disabled? I have ~13w when both GPU enabled. I'm trying to understand if it's even worth to bother.

whenov, 2021-11-20:
@bomiyr
To revert the action, either do `echo 1 > /sys/bus/pci/rescan` or simply reboot.
With the discrete GPU disabled, the power consumption on my machine is about 6.8W when idling.

bomiyr, 2021-11-20:
@whenov 
Thank you! it is working, but I had to change the command to `echo 1 | sudo tee /sys/bus/pci/devices/0000\:01\:00.0/remove`. 
Right now I have idle power consumption ~8.5W when Nvidia GPU disabled, and ~11.5W in offload mode, when both GPU enabled.