Bumblebee-Project/bbswitch #188: bbswitch module fails to start at startup

surajrv6, 2019-09-30:
**bbswitch module** fails to load at **startup**. The module keeps **unloading** itself. Unable to use bbswitch with **bumblebee**. I am able to load the module manually using `# modprobe bbswitch` and also able to turn off and the graphics card. I am on Arch linux.
I had the same problem in Debian (testing and stable) as well. 
I am using ASUS TUF FX504GD laptop
I didnot try any boot options like acpi_osi...

`Linux suraj-laptop 5.3.1-arch1-1-ARCH #1 SMP PREEMPT Sat Sep 21 11:33:49 UTC 2019 x86_64 GNU/Linux`

dmesg log

```
[root@suraj-laptop suraj]# dmesg | grep -C 10 bbswitch:
[    2.383351] systemd[1]: Set hostname to <suraj-laptop>.
[    2.455952] usb 1-14: new full-speed USB device number 4 using xhci_hcd
[    2.548224] systemd[1]: Created slice system-getty.slice.
[    2.548696] systemd[1]: Created slice system-systemd\x2dfsck.slice.
[    2.549945] systemd[1]: Created slice User and Session Slice.
[    2.550812] systemd[1]: Started Dispatch Password Requests to Console Directory Watch.
[    2.551654] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[    2.552737] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[    2.553571] systemd[1]: Reached target Local Encrypted Volumes.
[    2.577949] EXT4-fs (sda3): re-mounted. Opts: (null)
[    2.578742] bbswitch: loading out-of-tree module taints kernel.
[    2.578765] bbswitch: module verification failed: signature and/or required key missing - tainting kernel
[    2.578975] bbswitch: version 0.8
[    2.578981] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[    2.578987] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[    2.578997] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20190703/nsarguments-59)
[    2.579149] bbswitch: detected an Optimus _DSM function
[    2.579158] pci 0000:01:00.0: enabling device (0006 -> 0007)
[    2.579297] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[    2.579456] random: lvm: uninitialized urandom read (4 bytes read)
[    2.596891] usb 1-14: New USB device found, idVendor=8087, idProduct=0aaa, bcdDevice= 0.02
[    2.596892] usb 1-14: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.756548] audit: type=1130 audit(1569826524.009:2): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-journald comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    2.761488] systemd-journald[319]: Received client request to flush runtime journal.
[    2.794184] audit: type=1130 audit(1569826524.045:3): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-udevd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    2.834605] input: Asus Wireless Radio Control as /devices/LNXSYSTM:00/LNXSYBUS:00/ATK4002:00/input/input6
[    2.836898] audit: type=1130 audit(1569826524.089:4): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-journal-flush comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    2.837570] ACPI: AC Adapter [ACAD] (on-line)
[    2.844149] acpi PNP0C14:03: duplicate WMI GUID 05901221-D566-11D1-B2F0-00A0C9062910 (first instance was on PNP0C14:02)
--
[    3.712164] input: HDA Intel PCH HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input20
[    3.754208] r8169 0000:02:00.0 enp2s0: Link is Down
[    3.778067] usbcore: registered new interface driver btusb
[    3.778770] Bluetooth: hci0: Firmware revision 0.1 build 216 week 27 2019
[    3.781988] iwlwifi 0000:00:14.3: Applying debug destination EXTERNAL_DRAM
[    3.897237] iwlwifi 0000:00:14.3: Applying debug destination EXTERNAL_DRAM
[    3.962197] iwlwifi 0000:00:14.3: FW already configured (0) - re-configuring
[    3.972079] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[    4.003852] iwlwifi 0000:00:14.3: Applying debug destination EXTERNAL_DRAM
[    4.119843] iwlwifi 0000:00:14.3: Applying debug destination EXTERNAL_DRAM
[    4.142757] bbswitch: Unloaded. Discrete card 0000:01:00.0 is on
[    4.185319] iwlwifi 0000:00:14.3: FW already configured (0) - re-configuring
[    4.197844] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[    4.856001] mc: Linux media interface: v0.10
[    4.876537] videodev: Linux video capture interface: v2.00
[    5.001396] uvcvideo: Found UVC 1.00 device USB2.0 HD UVC WebCam (13d3:5666)
[    5.010443] uvcvideo 1-7:1.0: Entity type for entity Realtek Extended Controls Unit was not initialized!
[    5.010447] uvcvideo 1-7:1.0: Entity type for entity Extension 4 was not initialized!
[    5.010450] uvcvideo 1-7:1.0: Entity type for entity Processing 2 was not initialized!
[    5.010452] uvcvideo 1-7:1.0: Entity type for entity Camera 1 was not initialized!
[    5.010564] input: USB2.0 HD UVC WebCam: USB2.0 HD as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/input/input21
--
[   10.076710] wlo1: send auth to (my mac address) (try 1/3)
[   10.120348] wlo1: authenticated
[   10.124572] wlo1: associate with (my mac address) (try 1/3)
[   10.145268] wlo1: RX AssocResp from (my mac address) (capab=0x431 status=0 aid=5)
[   10.147825] wlo1: associated
[   10.214129] IPv6: ADDRCONF(NETDEV_CHANGE): wlo1: link becomes ready
[   10.271921] audit: type=1130 audit(1569826531.522:35): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=rtkit-daemon comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[... 235 lines cut ...]
Sep 30 12:24:15 suraj-laptop dbus-daemon[975]: [session uid=1000 pid=975] Activating service name='ca.desrt.dconf' requested by ':1.58' (uid=1000 pid=1919 comm="mousepad /etc/modprobe.d/bbswitch.conf ")
Sep 30 12:25:06 suraj-laptop lightdm[788]: Available kernel modules : ['nouveau', 'bbswitch', 'nvidia', 'nvidia_drm', 'nvidia_modeset']
Sep 30 12:25:06 suraj-laptop lightdm[788]: Setting GPU power to ON via bbswitch
Sep 30 12:25:06 suraj-laptop lightdm[788]: Unloading modules ['bbswitch'] (if loaded)
Sep 30 12:25:06 suraj-laptop kernel: bbswitch: Unloaded. Discrete card 0000:01:00.0 is on
Sep 30 12:25:24 suraj-laptop kernel: bbswitch: loading out-of-tree module taints kernel.
Sep 30 12:25:24 suraj-laptop kernel: bbswitch: module verification failed: signature and/or required key missing - tainting kernel
Sep 30 12:25:24 suraj-laptop kernel: bbswitch: version 0.8
Sep 30 12:25:24 suraj-laptop kernel: bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
Sep 30 12:25:24 suraj-laptop kernel: bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
Sep 30 12:25:24 suraj-laptop kernel: bbswitch: detected an Optimus _DSM function
Sep 30 12:25:24 suraj-laptop kernel: bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
Sep 30 12:25:24 suraj-laptop systemd-modules-load[320]: Inserted module 'bbswitch'
Sep 30 12:25:25 suraj-laptop prime-switch-boot[481]: Available kernel modules : ['nouveau', 'bbswitch', 'nvidia', 'nvidia_drm', 'nvidia_modeset']
Sep 30 12:25:25 suraj-laptop prime-switch-boot[481]: Setting GPU power to ON via bbswitch
Sep 30 12:25:25 suraj-laptop prime-switch-boot[481]: Unloading modules ['bbswitch'] (if loaded)
Sep 30 12:25:25 suraj-laptop kernel: bbswitch: Unloaded. Discrete card 0000:01:00.0 is on
Sep 30 12:26:04 suraj-laptop kernel: bbswitch: version 0.8
Sep 30 12:26:04 suraj-laptop kernel: bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
Sep 30 12:26:04 suraj-laptop kernel: bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
Sep 30 12:26:04 suraj-laptop kernel: bbswitch: detected an Optimus _DSM function
Sep 30 12:26:04 suraj-laptop kernel: bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
Sep 30 12:26:13 suraj-laptop kernel: bbswitch: disabling discrete graphics

```

eviltak, 2019-10-02:
I'm facing the same problem. `bbswitch` unloads itself when started via `/etc/modules-load.d/`, but stays loaded (and is able to turn the card on/off) when loaded manually via `modprobe`.

surajrv6, 2019-10-08:
Hmm, I think maybe it is kernel related issue? Yesterday, I installed debian buster stable with kernel 4.19. Same problem except the nvidia card was not turning off in debian. I bypassed the laptop hanging issue by putting bbswitch load_state=0 in modprobe.d (instructions from bbswitch github main page) Didnt try Debian testing 5.0 kernel..