Bumblebee-Project/bbswitch #191: Is the card really off?

mindrunner, 2019-12-21:
Hi there
I have a Lenovo Yoga 730 15-IKB. BIOS setting is either `optimus` or `discrete`. I am running the optimus setting. I am using https://github.com/Askannz/optimus-manager to switch between `intel` and `nvidia` on demand. Most of the time, I do not need discrete graphics and prefer having a cooler laptop and more battery power. Everything seems to be fine, however, I would like to confirm that my card is really turned off and there is no way to save more power (extend battery life). In intel mode, I can see the following:

```
[le@w530]: ~>$ cat /proc/acpi/bbswitch 
0000:3b:00.0 OFF
```

```
[le@w530]: ~>$ lsmod | grep nvidia
[le@w530]: ~>$ 
```

```
[le@w530]: ~>$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
[le@w530]: ~>$ lspci | grep NVIDIA
3b:00.0 3D controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev ff)

```

[image]

Am I on maximum power saving, or do I need to tweak more?

Lekensteyn, 2019-12-22:
Usually watching the battery drain is a better indicator than any of the tunables.

When you see `rev ff` in lspci, it typically indicates that the GPU device is inaccessible. Hopefully that is due to lack of power, but I am not sure whether the power resource is always turned off in that case. When you do use bbswitch, then Runtime PM is not used and may in fact conflict with bbswitch since they use different mechanisms to achieve the same.

If you do not really need bbswitch, and are only interested in power saving, then I recommend removing bbswitch and rely on runtime PM instead. Note that using `lspci` will temporarily restore power while the PCI config space is read, so to effectively measure whether power is indeed off, watch the battery drain in the PowerTOP overview.

mindrunner, 2019-12-22:
> Usually watching the battery drain is a better indicator than any of the tunables.
Do you mean the power estimates and usage percentile in the device-stats tab?
The NVidia card does not show any power estimates and a 0% usage. When running `watch -n1 lspci`, the percentile slowly increases. 
I'd say, it's all good here, huh? Exactly what I can expect (assuming I can trust powertop)

>When you see rev ff in lspci, it typically indicates that the GPU device is inaccessible.
That's good information, thank you.

> If you do not really need bbswitch, and are only interested in power saving, then I recommend removing bbswitch and rely on runtime PM instead. 
Not sure if that is a supported option in optimus-manager. For me it seems like bbswitch is doing its job quite nicely, so no real reason to abandon it. Or what's my advantage with only relying on runtime PM?