Bumblebee-Project/bbswitch #197: bbswitch finds 2 discrete graphic cards on archlinux

QbieShay, 2020-04-04:
I have an asus TUF 505 laptop with a Ryzen7h and an nvidia 1660ti.

I followed all the instructions on using nvidia-xrun, but i see this with dmesg:
```
[    3.185160] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.GPP0.PEGP
[    3.185165] bbswitch: Found discrete VGA device 0000:05:00.0: \_SB_.PCI0.GP17.VGA_
```

lspci output:
```
01:00.0 VGA compatible controller: NVIDIA Corporation TU116M [GeForce GTX 1660 Ti Mobile] (rev a1)
05:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev c1)
```

this clicks perfectly with the output when i run bumblebee:
```
[  376.858693] [ERROR]No integrated video card found, quitting.
```

OS: `5.5.13-arch2-1`

This is as far as i came with my laptop. If anyone knows how to solve this, I'll be immensely grateful.

sukanka, 2020-04-13:
See this  issue, https://github.com/Bumblebee-Project/bbswitch/issues/187#issue-497288943
I was experiencing the same problem. And I make a package in aur named [bbswitch-ati-git](https://aur.archlinux.org/packages/bbswitch-ati-git/) to fix this and compilation error on Linux 5.6.

QbieShay, 2020-04-15:
Hi @sukanka. Thank you so much for sharing your fix!
Unfortunately it doesn't work for me, I have the same issue with bumblebeed saying I don't have an integrated card :(

sukanka, 2020-04-15:
Hi @QbieShay ,  I guess your problem is also because of
https://github.com/Bumblebee-Project/bbswitch/blob/0c38f97c3a647aebbc298b830ac7496bfcb86a7d/bbswitch.c#L421-L430

but your `vendor` is neither `PCI_VENDOR_ID_INTEL` nor `PCI_VENDOR_ID_ATI`, so my fix also fails.
maybe you can try to replace it with `PCI_VENDOR_ID_AMD`.

bomiyr, 2021-11-20:
I have pretty the same issue with two "discrete" GPUs. My hardware: Lenovo Legion 5, CPU AMD 4800H with AMD Renoir graphics + GeForce GTX 1660 Ti Mobile. OS openSUSE Tumbleweed.

I downloaded the code, applied the fix with adding condition ` || pdev->vendor == PCI_VENDOR_ID_ATI` and compiled. After that I was able to load my version of bbswitch. And still I was not able to make it work, even more - it has really scary behavior. Right after I disable nvidia GPU all fans started spinning at 100%, and then after several dozens of seconds laptop just powered off. Maybe it has some hardware protection or something similar that treat disabled GPU as something wrong, I don't know